libusb/hidapi #377: macOS: kIOMasterPortDefault is deprecated

szechyjs, 2022-02-07:
```
hidapi/mac/hid.c:795:39: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:123:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
In file included from ../go/pkg/mod/github.com/bearsh/hid@v1.4.0/hid_enabled.go:46:
hidapi/mac/hid.c:800:34: warning: 'kIOMasterPortDefault' is deprecated: first deprecated in macOS 12.0 [-Wdeprecated-declarations]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:123:19: note: 'kIOMasterPortDefault' has been explicitly marked deprecated here
```

Youw, 2022-02-07:
Has Apple provided an alternative?

szechyjs, 2022-02-08:
> Has Apple provided an alternative?

It looks like its `kIOMainPortDefault`

perhaps something like...
```c
#if (MAC_OS_X_VERSION_MAX_ALLOWED < 120000) // Before macOS 12 Monterey
  #define kIOMainPortDefault kIOMasterPortDefault
#endif
```

mcuee, 2022-02-09:
Interestingly libusb uses a simple way to do this. I am not saying it is a better way though.
https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L57
```
/* Both kIOMasterPortDefault or kIOMainPortDefault are synonyms for 0. */
static const mach_port_t darwin_default_master_port = 0;
```
Ref: 
https://developer.apple.com/documentation/iokit/kiomasterportdefault
https://developer.apple.com/documentation/iokit/kiomainportdefault
This is a synonym for NULL, if you'd rather use a named constant.

PR #406 (2022-05-02): Don't use deprecated kIOMasterPortDefault
`kIOMasterPortDefault` is deprecated since macOS 12.0.
One alternative is to use named constant `kIOMainPortDefault` which is not available before macOS 12.0.

Both named constants are just an alias for `NULL`, so it is simpler to use it directly instead.

Fixes: #377
commit 2aeb0688 (2022-05-02): Don't use deprecated kIOMasterPortDefault (#406)

`kIOMasterPortDefault` is deprecated since macOS 12.0.
One alternative is to use named constant `kIOMainPortDefault` which is not available before macOS 12.0.

Both named constants are just an alias for `NULL`, so it is simpler to use it directly instead.

Fixes: #377
closed 2022-05-02 by Youw