libusb/hidapi #380: hid_write always failed!

arilink-tech, 2022-02-10:
I used Hidapi to create a project for two-way communication with HID, now I can read the messages sent from the device to the computer normally and can use Bughound to send data to the device, but when I try to call hid_write to send a message to the device it returns the value Always -1 Failed to send my device's report ID is 1. I also set the first bit of the data packet to 1, but I still can't send data successfully. Why?

arilink-tech, 2022-02-11:
Bus Hound is a USB debugging tool link is https://perisoft.net/bushound/ I use it for USB debugging I am using USB Out bluk transmission wMaxPacketSize is 64bytes so I should set up an array of 64+1 size Do you want to communicate? I will upload the USB Properties exported by the USB Lyzer

arilink-tech, 2022-02-11:
<HTML><HEAD><TITLE></TITLE>
<STYLE type=text/css>
body {padding:0px 0px 0px 0px;background:#fff;color:#000;font-family:Arial,Verdana,Helvetica,sans-serif;font-size:8pt;}
th {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#c9c9c9;color:#fff;text-align:left;font-size:8pt;font-weight:normal;vertical-align:bottom;border:#fff 1px solid;}
td {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#fff;vertical-align:top;font-size:8pt;}
th.ud {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#c9c9c9;color:#fff;text-align:left;font-size:8pt;font-weight:normal;vertical-align:bottom;border:#fff 1px solid;}
td.ud {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#fff;vertical-align:top;font-size:8pt;}
td.udofs {padding-top:0px;padding-right:8px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;background:#fff;vertical-align:top;font-size:8pt;text-align:right;}
td.udsiz {padding-top:0px;padding-right:8px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;background:#fff;vertical-align:top;font-size:8pt;text-align:right;}
td.udval {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;background:#fff;vertical-align:top;font-size:8pt;text-align:right;}
td.udfld {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#fff;vertical-align:top;font-size:8pt;}
td.udbmfld {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:20px;background:#fff;vertical-align:top;font-size:8pt;white-space:nowrap;}
td.udhex {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;background:#fff;vertical-align:top;font-size:8pt;text-align:left;white-space:nowrap;}
td.uddes {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background:#fff;vertical-align:top;font-size:8pt;}
div.uddiv1 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:12px;margin-right:0px;margin-bottom:4px;margin-left:1px;text-align:left;}
div.uddiv2 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:4px;margin-right:0px;margin-bottom:4px;margin-left:12px;text-align:left;}
div.uddiv3 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:2px;margin-right:0px;margin-bottom:2px;margin-left:0px;text-align:left;}
div.uddiv4 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:2px;margin-right:0px;margin-bottom:2px;margin-left:20px;text-align:left;}
div.uddiv5 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:6px;margin-right:0px;margin-bottom:6px;margin-left:20px;text-align:left;}
div.uddiv6 {font-size:8pt;font-style:normal;font-weight:bold;text-decoration:none;margin-top:12px;margin-right:0;margin-bottom:4px;margin-left:0;text-align:left;}
div.uddiv7 {font-size:8pt;font-style:normal;font-weight:normal;text-decoration:none;margin-top:2px;margin-right:0;margin-bottom:4px;margin-left:0;text-align:left;}
a {color:#000;text-decoration:none;white-space:nowrap;}
a:link {color:#000;text-decoration:none;white-space:nowrap;}
a:hover {color:#000;text-decoration:none;white-space:nowrap;}
a:visited {color:#000;text-decoration:none;white-space:nowrap;}
a:active {color:#000;text-decoration:none;white-space:nowrap;}
a.totop {color:#ccc;font-family:"Courier New",Courier,mono;text-decoration:none;white-space:nowrap;margin-left:8px;}
.error {color:#f00;}
div.hrddiv1 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:12px;margin-right:0px;margin-bottom:4px;margin-left:1px;text-align:left;}
div.hrddiv2 {font-size:8pt;font-style:normal;text-decoration:none;margin-top:2px;margin-right:0px;margin-bottom:2px;margin-left:0px;text-align:left;}
th.hrd {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;background-color:#c9c9c9;color:#fff;text-align:left;font-size:8pt;white-space:nowrap;vertical-align:bottom;border:#fff 1px solid;font-weight:normal;}
td.hrdn {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;vertical-align:top;font-size:8pt;white-space:nowrap;background-color:#fff;}
td.hrdm {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;vertical-align:top;font-size:8pt;text-align:left;white-space:nowrap;background-color:#fff;}
td.hrdnin {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;vertical-align:top;font-size:8pt;white-space:nowrap;background-color:#e0ffe0;border:#e0ffe0 1px solid;}
td.hrdmin {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;vertical-align:top;font-size:8pt;text-align:left;white-space:nowrap;background-color:#e0ffe0;border:#e0ffe0 1px solid;}
td.hrdnout {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;vertical-align:top;font-size:8pt;white-space:nowrap;background-color:#ffe4e4;border:#ffe4e4 1px solid;}
td.hrdmout {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;font-family:"Courier New",Courier,mono;vertical-align:top;font-size:8pt;text-align:left;white-space:nowrap;background-color:#ffe4e4;border:#ffe4e4 1px solid;}
td.hrdnfea {padding-top:0px;padding-right:3px;padding-bottom:0px;padding-left:3px;vertical-align:top;font-size:8pt;white-space:nowrap;background-color:#ffffd4;border:#ffffd4 1px solid;}
[... cut ...]
<TD class=ud></TD>
<TD class=udbmfld><NOBR>7..2: Reserved</NOBR></TD>
<TD class=ud></TD>
<TD class=udval><NOBR>000000.. </NOBR></TD>
<TD class=ud></TD></TR>
<TR vAlign=top>
<TD class=udofs>4</TD>
<TD class=udfld><NOBR>wMaxPacketSize</NOBR></TD>
<TD class=udsiz>2</TD>
<TD class=udval>0040h</TD>
<TD class=uddes>64 bytes</TD></TR>
<TR vAlign=top>
<TD class=udofs>6</TD>
<TD class=udfld><NOBR>bInterval</NOBR></TD>
<TD class=udsiz>1</TD>
<TD class=udval>05h</TD></TR></TBODY></TABLE></DIV></DIV></BODY></HTML>

arilink-tech, 2022-02-11:
[image]

arilink-tech, 2022-02-11:
I am operating on windows. I changed the sent data to 64+1 but still failed to send. The code of the sending part is as follows:
int res;
     unsigned char tx_buf[65]={0};
     if(hubDev==nullptr)
     {
         return;
     }

     tx_buf[0] = 0x01;
     tx_buf[1] = 0xfe;
     tx_buf[2] = 0x01;
     tx_buf[3] = channel;
     tx_buf[4] = 0xef;
     if(hubDev!=nullptr)
     {
         res = hid_write(hubDev,tx_buf,65);
         if(res==-1)
         {
             qDebug()<<"send data failed!!!";
         }
     }

arilink-tech, 2022-02-11:
I upload the full report descriptor for you to see

arilink-tech, 2022-02-11:
[CH32-USBHID.pdf](https://github.com/libusb/hidapi/files/8045580/CH32-USBHID.pdf)

arilink-tech, 2022-02-11:
#359 Could it be that this problem occurs only with 64 bytes?

arilink-tech, 2022-02-11:
I modified the descriptor Endpoint Descriptor 01 1 Out, Bulk, 8 bytes and Report Descriptor to 8byte hid_writei and wrote 9byte, still can't work

mcuee, 2022-02-11:
This device is not compliant to USB HID spec. The output endpoint can not be Bulk type for HID device. You have to change it to Interrupt type.

mcuee, 2022-02-11:
USB HID Spec
https://www.usb.org/sites/default/files/hid1_11.pdf

>4.4 Interfaces
> A HID class device communicates with the HID class driver using either the Control (default) pipe or an Interrupt pipe.

closed 2022-02-11 by mcuee

arilink-tech, 2022-02-11:
The FW of the device was made by my colleague, I asked him to change the FW of the device to interrupt transmission and try

todbot, 2022-02-11:
The HID Report Descriptor in [CH32-USBHID.pdf](https://github.com/libusb/hidapi/files/8045580/CH32-USBHID.pdf) says it's a gamepad. You may have an OS-level gamepad driver that is preventing `hidapi` from communicating with the device.

arilink-tech, 2022-02-11:
After modifying the FW of the device, communication is possible, but the transmission result is not normal. It is wrong to remove the first byte of reportID

arilink-tech, 2022-02-17:
@todbot Thank you very much for your help, the problem has been solved