libusb/hidapi #381: HIDAPI MacOS displaying duplicated information about a device connected through USC-C -> MicroUSB cable

itamarcps, 2022-02-12:
## Describe the bug

When connecting to a device using product_id, vendor_id and even serial_number, there is a chance that the device_handle will not open.

Apparently, this bug happens when using a USB-C -> Micro USB cable to connect to the HID Device, and hidapi is unable to differenciate the cable itself (Or something similar? See expected behavior.) and the device.

## How to reproduce

Run the following code using a USB-C > Micro USB Cable to connect to your device.

```
  struct hid_device_info *devs;
    devs = hid_enumerate(0x0, 0x0);
    while (devs) {
       // Printing information about the device itself...
       ...
       devs = devs->next;
    }
```
  
## Expected behavior

I should be able to at least see different serial_numbers so select which device to connect to.
## Actual behavior
In MacOS, the previous code will retrieve the following information:

```
Device Found
  type: 2c97 1015
  path: DevSrvsID:4294972422
  serial_number: 0001
  Manufacturer: Ledger
  Product:      Nano S
  Release:      201
  Interface:    1
  Usage (page): 0x1 (0xf1d0)

Device Found
  type: 2c97 1015
  path: DevSrvsID:4294972420
  serial_number: 0001
  Manufacturer: Ledger
  Product:      Nano S
  Release:      201
  Interface:    0
  Usage (page): 0x1 (0xffa0)
 
```

Be aware that these are completely different devices, but under hidapi they appear under the same proprieties (even serial_number)

when running the command ˜lsusb˜, I'm able to retrieve the following information:

```
Bus 000 Device 002: ID 2c97:1015 2c97 Nano S  Serial: 0001
Bus 000 Device 000: ID 2c97:1015 2c97 USB 3.1 Bus 
```

## Environment
 - MacOS Montery - 12.1

itamarcps, 2022-02-12:
Using the following code to always open a device handle seems to work, for the connection.

When I try to hid_read on this device, it simply freezes.

```
    ....
    devs = hid_enumerate(this->ledgerVID, this->ledgerPID);
    while (devs) {
      this->device_handle = hid_open_path(devs->path);
      if (!this->device_handle) {
        // Did not open... yet
        hid_close(this->device_handle);
      } else {
        // Handler opened!
        break;
      }
      devs = devs->next;
    }
    ...
    // Freezes here
    unsigned char receiveBuffer[64];
    res = hid_read(this->device_handle, receiveBuffer, sizeof(receiveBuffer));
```

todbot, 2022-02-13:
The full unique combination of ids for a device in hidapi is: vendor_id, product_id, usage_page, usage, serial_number

In your example, the two devices are on different usage_pages. 

When searching through devices during enumeration, be sure to filter on usage_page and usage as well.

itamarcps, 2022-02-13:
> The full unique combination of ids for a device in hidapi is: vendor_id, product_id, usage_page, usage, serial_number
> 
> In your example, the two devices are on different usage_pages.
> 
> When searching through devices during enumeration, be sure to filter on usage_page and usage as well.

As far as my understanding from the hidapi, there isn't a function that allows us to select which device to open the handle based on usage_page. Still, regardless of that, I have (as demonstrated in the previous comment) said that one of the devices always returns normally using hid_open_path. So I iterate through my two devices and connect to the one that works. I successfully write to it but reading from it only works sometimes when running my binary.

My code (without the while(devs) loop) is located here for further consultation:

https://github.com/avme/avme-wallet/blob/main/src/lib/ledger/comms.cpp#L71

JoergAtGithub, 2022-02-13:
1.) Call hid_enumerate, than you get a list of devices with all details
2.) Select the device you want to access and use the path of this device to call hid_open_path

todbot, 2022-02-13:
I guess I am confused as to what your issue is. 
It looks like from your code that you are not checking which usage_page and usage of the device, but only the vendor_id and product_id. hidapi returns multiple device entries for the same physical device if it has multiple usage_page definitions.

So a more complete enumeration and opening strategy might be:
1. call `hid_enumerate()` 
2. foreach device, if device's vid, pid, usage_page, usage all match: save device path
3. if saved device path, we found matching device, so open it

You can see this strategy [implemented in `hidapitester` here](https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L299), with the important code being:

```c
                    devs = hid_enumerate(vid, pid); // 0,0 = find all devices
                    cur_dev = devs;
                    while (cur_dev) {
                        if( (!vid || cur_dev->vendor_id == vid) &&
                            (!pid || cur_dev->product_id == pid) &&
                            (!usage_page || cur_dev->usage_page == usage_page) &&
                            (!usage || cur_dev->usage == usage) ) {
                            strncpy(devpath, cur_dev->path, MAX_STR); // save it!
                        }
                        cur_dev = cur_dev->next;
                    }
                    hid_free_enumeration(devs);
```

itamarcps, 2022-02-14:
> I guess I am confused as to what your issue is. It looks like from your code that you are not checking which usage_page and usage of the device, but only the vendor_id and product_id. hidapi returns multiple device entries for the same physical device if it has multiple usage_page definitions. And it looks like your device is showing up as two devices in hidapi because of that.
> 
> So a more complete enumeration and opening strategy might be:
> 
> 1. call `hid_enumerate()`
> 2. foreach device, if device's vid, pid, usage_page, usage all match: save device path
> 3. if saved device path, we found matching device, so open it
> 
> You can see this strategy [implemented in `hidapitester` here](https://github.com/todbot/hidapitester/blob/master/hidapitester.c#L299), with the important code being:
> 
> ```c
>                     devs = hid_enumerate(vid, pid); // 0,0 = find all devices
>                     cur_dev = devs;
>                     while (cur_dev) {
>                         if( (!vid || cur_dev->vendor_id == vid) &&
>                             (!pid || cur_dev->product_id == pid) &&
>                             (!usage_page || cur_dev->usage_page == usage_page) &&
>                             (!usage || cur_dev->usage == usage) ) {
>                             strncpy(devpath, cur_dev->path, MAX_STR); // save it!
>                         }
>                         cur_dev = cur_dev->next;
>                     }
>                     hid_free_enumeration(devs);
> ```

After researching about my USB device, I have found (accordingly to the documentation provided by the USB https://www.usb.org/sites/default/files/hut1_22.pdf) that the usage_page of my device is "0xF1D0", with that, I have used the following code to connect to my device.

```

    devs = hid_enumerate(this->ledgerVID, this->ledgerPID);
    while (devs) {
      if (devs->usage_page == 0xF1D0) {
        std::cout << "Connected!" << std::endl;
        this->device_handle = hid_open_path(devs->path);
        break;
      }
      devs = devs->next;
    }

```

After doing so, I was successfully able to open the device_handle to my device and write to it, but as soon as my code reaches the part where I try to hid_read from my device, using a 64 bytes array as a buffer, it simply freezes in this part.

But, contrary to the documentation I'm able to connect to the other device using devs->usage_page != 0xF1D0, write and receive from it...

I'm currently confused about how to properly implement this as the documentation itself goes against my own testings.

todbot, 2022-02-14:
The patch you have that's just for macOS should be applied to all OSes.  hidapi doesn't guarantee the return order in which devices (or sub-devices for those with multiple usage_pages) are presented in `hid_enumerate()`. It's likely you're getting lucky on other OSes that it's returning the usage_page you want first but on MacOS it only sometimes returns it first.

todbot, 2022-02-15:
Which is the single device that appears on Windows & Linux? I would expect it to be the one with usage_page `0xffa0` as I expect on Windows & Linux you might have OS-level U2F drivers installed (e.g. `libu2f-udev` on Linux)

itamarcps, 2022-02-16:
> Which is the single device that appears on Windows & Linux? I would expect it to be the one with usage_page `0xffa0` as I expect on Windows & Linux you might have OS-level U2F drivers installed (e.g. `libu2f-udev` on Linux)

The only device which appears in Linux and Windows is the `0xffa0` device, any reason for that happening? besides that, hidapi is displaying a different name compared to the command `lsusb`, any reason for that?

todbot, 2022-02-16:
I believe it's happening because of what I said: there are U2F drivers in Linux and Windows that are claiming the device.

itamarcps, 2022-02-16:
> I believe it's happening because of what I said: there are U2F drivers in Linux and Windows that are claiming the device.
> 
> What is the name difference between `lsusb` and `hidapi`?

Thank you for the information, I did not know about U2F drivers being able to claim the device and """hide""" it from lsusb and hidapi, but still, why the naming differs between lsusb and hidapi on macos?

todbot, 2022-02-16:
What are the naming differences?  hidapi is just printing out the raw "vendor" and "product" strings the device is providing.

mcuee, 2022-04-25:
It seems there is no hidapi issues here. Probably we can close this.

mcuee, 2022-05-04:
Close this for now. Please re-open when there are new info to show it is an issue with HIDAPI.

closed 2022-05-04 by mcuee