libusb/hidapi #387: HID Interface string descriptor product

adriencarbonaro, 2022-02-23:
I've seen a strange behavior with HID product string.

## Context

- I use `iProduct` string descriptor to set my product name.
- I use HID Interface `iInterface` to describe HID interface with a name.

## Current behavior

- When I set the `iInterface` string descriptor in HID Interface descriptor, the function `hid_get_product_string` returns the `iInterface` string descriptor rather than `iProduct`. But the actual `iProduct` string descriptor is also defined.
- Windows resolves the `iProduct` without issue and also resolves the HID `iInterface` string descriptor.

## Descriptors

```
Device Descriptor:
------------------------------
0x12      bLength
0x01      bDescriptorType
0x0200    bcdUSB
0xEF      bDeviceClass
0x02      bDeviceSubClass   
0x01      bDeviceProtocol   
0x40      bMaxPacketSize0
0x0001    idVendor
0x0002    idProduct
0x0200    bcdDevice
0x01      iManufacturer       "Me"
0x02      iProduct            "ProductName"
0x03      iSerialNumber       "1234567890"
0x01      bNumConfigurations

[...]

Interface Descriptor:
------------------------------
0x09      bLength
0x04      bDescriptorType
0x03      bInterfaceNumber
0x00      bAlternateSetting
0x02      bNumEndPoints
0x03      bInterfaceClass     (Human Interface Device Class)
0x00      bInterfaceSubClass   
0x00      bInterfaceProtocol   
0x07      iInterface          "HIDName"

[...]

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Me"
0x02   0x0409  "ProductName"
0x03   0x0409  "1234567890"
0x06   0x0409  "Other Interface string desc"
0x07   0x0409  "HIDName"
```

DJm00n, 2022-03-17:
Hi!
`hidapi` uses `HidD_GetProductString` API under the hood for `hid_get_product_string` on Windows.

See the following remark in [its docs](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/hidsdi/nf-hidsdi-hidd_getproductstring):

>The iProduct member of a [USB_DEVICE_DESCRIPTOR](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbspec/ns-usbspec-_usb_device_descriptor) structure for a particular interface is set by the [USB common class generic parent driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/) based on the following rules:
> - If the iInterface member of the [USB_INTERFACE_DESCRIPTOR](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbspec/ns-usbspec-_usb_interface_descriptor) structure for the interface is nonzero, the iProduct member of the USB_DEVICE_DESCRIPTOR structure for the interface is set to the iInterface member of the USB_INTERFACE_DESCRIPTOR structure.

Youw, 2022-03-30:
Don't see any additional additional responce from the author.
Don't see an issue here.

closed 2022-03-30 by Youw

adriencarbonaro, 2022-03-30:
Is there another way to get this string, and how Windows is able to get it (display it in `tdd` tool for example) and why does it replaces the product string set by device in the backend ?

DJm00n, 2022-03-30:
@adriencarbonaro 

>The point of iInterface is to give a name to the interface, not the product itself, as a product could have multiple interfaces.

USB devices with multiple interface are called "USB Composite Device" and driven [by special driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-common-class-generic-parent-driver).
So each interface is represented as a separate subdevice in the device tree (with separate driver!). Therefore, the name of this "interface device" is could be overridden by `iInterface` (if it is nonzero).

So for example one physical usb device could implement keyboard, mouse, keypad interfaces (seen as separate devices in WIndows device manager, with separate names). This is how typical Logitech USB receiver works.

If you really need "whole" device name you can fetch it with a bunch of `cfgmgr32` API calls.

adriencarbonaro, 2022-03-30:
My device is a composite, hence the whole interrogation.

Thanks for the support ! :)