libusb/hidapi #392: Building with autotools and cmake produce different library names

swt2c, 2022-02-28:
When building on Linux, building with autotools and cmake produce different library names.  autotools has always used (probably wrongly) an soname version of 0.0.0 whereas cmake has 0.11.2.

autotools:
```
/usr/lib64/libhidapi-hidraw.so.0
/usr/lib64/libhidapi-hidraw.so.0.0.0
/usr/lib64/libhidapi-libusb.so.0
/usr/lib64/libhidapi-libusb.so.0.0.0
```

cmake:
```
/usr/lib64/libhidapi-hidraw.so.0
/usr/lib64/libhidapi-hidraw.so.0.11.2
/usr/lib64/libhidapi-libusb.so.0
/usr/lib64/libhidapi-libusb.so.0.11.2
```

Youw, 2022-02-28:
> (probably wrongly)

Yeap, that's the point. And it is not going to be fixed.

Youw, 2022-03-05:
Don't see any more questions/issues here, closing.

closed 2022-03-05 by Youw