libusb/hidapi #399: PS3 Controller doesn't show any data

pknessness, 2022-03-25:
I tried this with my PS3 Controller, there's no data, even when I press some buttons and move the joysticks around.
I tried with a LUNA controller and it worked fine, is there something I need to do differently with a PS3 Controller?

Youw, 2022-03-25:
Which platform?

todbot, 2022-03-25:
I'm pretty sure PS3 controller is not a HID device.  At least judging from all the HID drivers for it: https://github.com/ViGEm/DsHidMini#related-projects

DJm00n, 2022-04-16:
@todbot Sixaxis (aka PS3 controller) is HID controller. But it has broken report descriptor.
https://vigem.org/research/SIXAXIS-native-HID-Report-Descriptor/

mcuee, 2022-05-04:
Close this for now. Please re-open if you have followed the suggestion from DJm00n and still have issues.

closed 2022-05-04 by mcuee