libusb/hidapi #401: [HelpRequest] Cmake hidtest.exe cross-compilation from WSL2 (Ubuntu 20.04) for Windows

xcme, 2022-03-29:
Hello,

This is not an issue, actually, just looking for help. Could, please, anyone provide an example how to compile **hidtest** for Windows under Linux? At the moment I installed mingw-w64 and other dependencies, but not sure how to proceed. I tried with _cmake_ with options:
```
-DHIDAPI_BUILD_HIDTEST=ON -DWIN32=1
```
A then just **make -C \<path\>** but it doesn't see files from _/usr/share/mingw-w64/include_, like **windows.h** and so on. Googled a lot, but most examples are for Windows and without experience with these tools it's to hard to go further. I'd appreciate your help.

P.S. Compilation for Linux seems to work, but I want to get hidtest.exe for Windows.

Youw, 2022-03-29:
You need to google general CMake cross-compilation (there is nothing HIDAPI-specific).
Brief googling lead me to here: https://gist.github.com/peterspackman/8cf73f7f12ba270aa8192d6911972fe8

Haven't actually tried, but looks about right AFAIK.

xcme, 2022-03-29:
Thanks a lot! I've seen that file but didn't try. Now just copied, used and it worked. Not sure if I understand what I did (I'm even not a programmer at all), but it helped :)

Do you think there could be the **official** toolchain file for hidapi? Not sure if the current one is optimal... Anyway, it works, thank you for help!

Youw, 2022-03-29:
> Not sure if I understand what I did

Manual: https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html
Book: https://cmake.org/cmake/help/book/mastering-cmake/chapter/Cross%20Compiling%20With%20CMake.html

> the official toolchain file for hidapi

That makes no sense. There is countless possible toolchains for countless platforms/cross-compilations/etc.
Again - there is nothing HIDAPI-specific about it. Everything is _generic_ CMake.

closed 2022-03-29 by Youw