libusb/hidapi #407: hidapi-0.12.0 Test matrix

Youw, 2022-05-03:
Latest release candidate: [hidapi-0.12.0-rc1](https://github.com/libusb/hidapi/releases/tag/hidapi-0.12.0-rc1)

mcuee, 2022-05-03:
I should be able to test under the following system. I will use CMake on most of the systems. Then I will also test the VS solution using VS2019.

### macOS 12 ARM64 (mac Mini M1)
  - [ ] build: Apple clang
  - [ ] run:  hidtest and hidapitester

### macOS 12 x86_64 using Rosetta Translation, not native (mac Mini M1)
  - [ ] build: Apple clang
  - [ ] run: hidtest and hidapitester
  
### Linux x86_64 (Ubuntu 20.04)
  - [ ] build: gcc 9.4
  - [ ] run: hidtest and hidaptester

### Linux ARM32 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 32 bit)
  - [ ] build: gcc
  - [ ] run: hidtest and hidaptester

### Linux ARM64 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 64 bit)
  - [ ] build: gcc
  - [ ] run: hidtest and hidaptester
 
### Windows VS2019 64 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: hidtest and hidaptester
  
### Windows VS2019 32 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: hidtest and hidaptester
 
### Windows MinGW 64 bit (Windows 11 x64 laptop)
  - [ ] build: MSYS2 MinGW-w64 64bit GCC
  - [ ] run: hidtest and hidaptester
  
### Windows MinGW 32 bit (Windows 11 x64 laptop)
  - [ ] build: MinGW-w64 32 bit GCC
  - [ ] run: hidtest and hidaptester

mcuee, 2022-05-03:
I can not test macOS Intel but I am sure there are people who can test it. Same for FreeBSD. Then I am probably not good at testing cython-hidapi and node-hid but again other people can test.

Hopefully someone can test macOS Intel and older macOS versions, old Windows XP/Vista/7 32bit/64bit versions. It will be good too that the users of the language bindings can test as well.

Then it will be helpful if Linux distro maintainers of hidapi can check out this new release as well.

The hope is to have the release in two weeks' time.

mcuee, 2022-05-03:
BTW, github actions have already tested the build of hidapi 0.12.0-rc1 under the following platform 
https://github.com/libusb/hidapi/actions/runs/2263027445
1) macOS 11 automake and cmake using Apple clang
2) Ubuntu 20.04 cmake 
3) Windows Server 2022 cmake: msvc (VS2022), nmake (VS2022) and MinGW gcc 8.1.0

mcuee, 2022-05-03:
hidapitest log under Ubuntu 20.04 (x86_64).
```
~/build/hidapitester$ ./hidapitester --vidpid 0925:7001  -l 3 --open --send-output 2,2,3 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 02 02 03
Reading 3-byte input report 0, 250 msec timeout...read 3 bytes:
 01 02 03
Closing device

./hidapitester --vidpid 0925:7001  -l 4 --open --send-feature 3,4,5,6 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 4-byte feature report...wrote 4 bytes:
 03 04 05 06
Reading 4-byte feature report, report_id 4...read 3 bytes:
 04 04 05 00
Closing device

```

mcuee, 2022-05-03:
@todbot If possible, please help to carry out some tests using node-hid and hidapitester as you seem to have many working device FW to test hidapi.

mcuee, 2022-05-03:
@Youw 
The Darwin CMake build failed (for hidtest) under my Mac Mini M1 (Apple clang), either for the native build or the x86_64 build under Rosetta.

github action does not detect this since it does not build hidtest.

```
mcuee@mcuees-Mac-mini build % make
[ 25%] Building C object src/mac/CMakeFiles/hidapi_darwin.dir/hid.c.o
/Users/mcuee/build/hidapi/hidapi/mac/hid.c:793:39: warning: incompatible pointer to integer conversion passing 'void *' to parameter of type 'mach_port_t' (aka 'unsigned int') [-Wint-conversion]
                        return IOServiceGetMatchingService(NULL, IORegistryEntryIDMatching(entry_id));
                                                           ^~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/stddef.h:89:16: note: expanded from macro 'NULL'
#  define NULL ((void*)0)
               ^~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:419:14: note: passing argument to parameter 'mainPort' here
        mach_port_t     mainPort,
                        ^
/Users/mcuee/build/hidapi/hidapi/mac/hid.c:798:34: warning: incompatible pointer to integer conversion passing 'void *' to parameter of type 'mach_port_t' (aka 'unsigned int') [-Wint-conversion]
                return IORegistryEntryFromPath(NULL, path);
                                               ^~~~
/Library/Developer/CommandLineTools/usr/lib/clang/13.1.6/include/stddef.h:89:16: note: expanded from macro 'NULL'
#  define NULL ((void*)0)
               ^~~~~~~~~~
/Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/IOKit.framework/Headers/IOKitLib.h:964:15: note: passing argument to parameter 'mainPort' here
        mach_port_t             mainPort,
                                ^
2 warnings generated.
[ 50%] Linking C shared library libhidapi.dylib
[ 50%] Built target hidapi_darwin
[ 75%] Building C object hidtest/CMakeFiles/hidtest.dir/test.c.o
/Users/mcuee/build/hidapi/hidapi/hidtest/test.c:40:10: fatal error: 'hidapi_darwin.h' file not found
#include <hidapi_darwin.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [hidtest/CMakeFiles/hidtest.dir/test.c.o] Error 1
make[1]: *** [hidtest/CMakeFiles/hidtest.dir/all] Error 2
make: *** [all] Error 2
```

Youw, 2022-05-03:
```/Users/mcuee/build/hidapi/hidapi-hidapi-0.12.0-rc1/hidtest/test.c:40:10: fatal error: 'hidapi_darwin.h' file not found
#include <hidapi_darwin.h>
         ^~~~~~~~~~~~~~~~~
```
I remember I've been fixing it. Maybe the fix got lost in a merge conflict or smth.
Thanks for finding it. I'll update the CI script to test this as well.

JoergAtGithub, 2022-05-03:
I compiled it on Windows 7 VisualStudio 2019 CMake. All I tried, works as before!

Youw, 2022-05-04:
#409 with the fix

mcuee, 2022-05-06:
#409 is good and it has fixed the warnings and hidtest build issues under macOS.

mcuee, 2022-08-01:
hidapi 0.12.0 has been released on 25 May 2022. So this can be closed.

closed 2022-08-01 by mcuee