libusb/hidapi #411: Recommended udev rules for hidraw stop working after device reconnection

PAI5REECHO, 2022-05-05:
I'm using https://github.com/libusb/hidapi/blob/2aeb0688fa8c7b14cc767f7f13088f703487f754/udev/69-hid.rules#L15-L16 which works initially at boot, but when the device is unplugged and plugged in again it stops working until I run `sudo udevadm control --reload-rules && sudo udevadm trigger`. Is there a way I can get this rule to keep working with hot-plugged devices?

Youw, 2022-05-05:
What order (number prefix) your udev.rules has?

Youw, 2022-05-05:
Try: `# udevadm monitor` to see udev log

PAI5REECHO, 2022-05-05:
> What order (number prefix) your udev.rules has?

```console
$ rg uinput /etc/udev/rules.d/99-local.rules
9:KERNEL=="uinput", MODE="0660", GROUP="input", OPTIONS+="static_node=uinput"
$ ls /etc/udev/rules.d/
00-path.rules              13-dm-disk.rules           69-bcache.rules                       80-udisks2.rules                 98-ipv6-privacy-extensions.rules
01-md-raid-creating.rules  51-android.rules           69-dm-lvm.rules                       90-libinput-fuzz-override.rules  99-ipv6-privacy-extensions.rules
10-dm.rules                63-md-raid-arrays.rules    69-md-clustered-confirm-device.rules  90-pipewire-alsa.rules           99-local.rules
11-dm-lvm.rules            64-md-raid-assembly.rules  80-libinput-device-groups.rules       95-dm-notify.rules               99-razer.rules
# udevadm monitor # plugs in device now:
KERNEL[37380.846057] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10 (usb)
KERNEL[37380.850205] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.0 (usb)
KERNEL[37380.850531] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.0/tty/ttyACM0 (tty)
KERNEL[37380.850554] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.0 (usb)
KERNEL[37380.850909] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.1 (usb)
KERNEL[37380.850925] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.1 (usb)
KERNEL[37380.851156] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2 (usb)
KERNEL[37380.851257] add      /devices/virtual/workqueue/scsi_tmf_7 (workqueue)
KERNEL[37380.851316] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7 (scsi)
KERNEL[37380.851332] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/scsi_host/host7 (scsi_host)
KERNEL[37380.851415] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2 (usb)
KERNEL[37380.851716] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3 (usb)
UDEV  [37380.852341] add      /devices/virtual/workqueue/scsi_tmf_7 (workqueue)
KERNEL[37380.852434] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/0003:239A:80F4.000D (hid)
KERNEL[37380.852494] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/usbmisc/hiddev0 (usbmisc)
KERNEL[37380.852547] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/0003:239A:80F4.000D/hidraw/hidraw6 (hidraw)
KERNEL[37380.852648] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/0003:239A:80F4.000D (hid)
KERNEL[37380.852662] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3 (usb)
KERNEL[37380.853021] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4 (usb)
KERNEL[37380.854312] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4 (sound)
KERNEL[37380.854359] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/midiC4D0 (sound)
KERNEL[37380.854370] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/midi4 (sound)
KERNEL[37380.854397] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/dmmidi4 (sound)
KERNEL[37380.854405] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/seq-midi-4-0 (snd_seq)
KERNEL[37380.854517] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/seq-midi-4-0 (snd_seq)
KERNEL[37380.854531] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/controlC4 (sound)
KERNEL[37380.854544] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4 (usb)
KERNEL[37380.854741] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.5 (usb)
KERNEL[37380.854756] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.5 (usb)
KERNEL[37380.854785] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10 (usb)
UDEV  [37380.855653] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10 (usb)
UDEV  [37380.863332] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.0 (usb)
UDEV  [37380.863362] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2 (usb)
UDEV  [37380.863381] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.5 (usb)
UDEV  [37380.863641] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.1 (usb)
UDEV  [37380.865123] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4 (usb)
UDEV  [37380.865143] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7 (scsi)
UDEV  [37380.865160] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.5 (usb)
UDEV  [37380.865177] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3 (usb)
UDEV  [37380.865194] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.1 (usb)
UDEV  [37380.865208] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/usbmisc/hiddev0 (usbmisc)
UDEV  [37380.865889] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/scsi_host/host7 (scsi_host)
UDEV  [37380.865922] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4 (sound)
UDEV  [37380.865945] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.3/0003:239A:80F4.000D (hid)
UDEV  [37380.865966] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2 (usb)
UDEV  [37380.868855] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/dmmidi4 (sound)
UDEV  [37380.868909] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/midiC4D0 (sound)
UDEV  [37380.868929] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.4/sound/card4/seq-midi-4-0 (snd_seq)
[... cut ...]
UDEV  [37382.035017] add      /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/target7:0:0/7:0:0:0/block/sde/sde1 (block)
UDEV  [37382.036207] bind     /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/target7:0:0/7:0:0:0 (scsi)
KERNEL[37382.086734] change   /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/target7:0:0/7:0:0:0/block/sde/sde1 (block)
UDEV  [37382.133158] change   /devices/pci0000:00/0000:00:14.0/usb2/2-10/2-10:1.2/host7/target7:0:0/7:0:0:0/block/sde/sde1 (block)
```

Youw, 2022-05-05:
> 99-local.rules

> TAG+="uaccess"

https://github.com/libusb/hidapi/blob/c95795ab5c9409994be7f51e2c6c97a6836e9616/udev/69-hid.rules#L20

PAI5REECHO, 2022-05-05:
I did read that before but my distro doesn't have any such `73-seat-late.rules` (see output of `ls /etc/udev/rules.d/` above) so I'm confused :(

Youw, 2022-05-05:
It is a part of [systemd](https://cgit.freedesktop.org/systemd/systemd/tree/src/login/73-seat-late.rules.in), not just some distro.

On Debian systems it is located at [/lib/udev/rules.d/73-seat-late.rules](https://www.apt-browse.org/browse/debian/wheezy/main/amd64/systemd/44-11+deb7u4/file/lib/udev/rules.d/73-seat-late.rules).

> Does this literally just mean

Documentation is pretty clear about what to do with reference about why it has to be done that way.

closed 2022-05-05 by Youw

PAI5REECHO, 2022-05-06:
Works, thanks :)