libusb/hidapi #417: Physical device ID for Linux/hidraw

JoergAtGithub, 2022-05-18:
With hid_darwin_get_location_id (#378) and hid_winapi_get_container_id(#379) we have physical device identifiers for Windows and MacOS, but not for Linux.
The Chromium HID implementation uses the same identifiers for Windows and MacOS, but has also an implementation for hidraw. They use the sysfs path of the device node that represents the physical device as physical identifier. They implemented a recursive search to gather this information:
https://chromium.googlesource.com/chromium/src/+/refs/changes/62/3314562/3/services/device/hid/hid_service_linux.cc#138

Youw, 2022-05-18:
- hidraw:
once we have [`hid_get_path`](https://github.com/libusb/hidapi/pull/164) I believe it is enough to identify the physysical device, since paths like `/dev/hidraw*` are [easily mapped](https://arvchristos.github.io/post/matching-dev-hidraw-devices-with-physical-devices/) to `/sys/class/hidraw/hidraw*`; that information should is more than enough for the application to get what it needs;
- libusb:
having `hid_get_path` we may say the very similar thing, as the path we generate is the same (https://github.com/libusb/hidapi/pull/291) as in `/sys/bus/usb/devices/`

Youw, 2022-08-21:
Resolved by #432

closed 2022-08-21 by Youw