libusb/hidapi #420: Commit 8b26e4962175a82205ebc499fdc2682e581e5fbe has gone missing

russell-taylor, 2022-05-25:
The https://github.com/vrpn/vrpn.git project uses HIDAPI as a submodule.  We have been checking out commit 8b26e4962175a82205ebc499fdc2682e581e5fbe for awhile now to make sure we're building using something that has been tested across a number of devices.

We just noticed today that we can't check out that commit anymore.

Has there been some pruning of old commits?  Can they be restored?

Thanks,
Russ

Youw, 2022-05-25:
This commit was never a part of `master`, and we never touch the history of `master` commits.
Maybe it was a part of one of the forks (that has been deleted) with a PR agains libusb/hidapi.

Or it might be a part of a development branch that has been deleted.

---
Can it be restored - NO, unless someone has a local copy of the branch that contains that exact commit.

---
I strongly recomment switching to one of the release tags/commits (e.g. to the latest one) instead.

Youw, 2022-05-25:
As far as I can tell from vrpn [commit history](https://github.com/vrpn/vrpn/commit/c0feef639cfc50e40b83f6eece6ea04fd02a11e1) - there is no rationale/explanation why 8b26e4962175a82205ebc499fdc2682e581e5fbe was used and where it came from.

russell-taylor, 2022-05-25:
We were trying to find a version that worked with the largest possible set of devices and operating systems and we were testing as a group at multiple institutions, each on some devices.

This was the one we settled on that worked the best.

I'll push from a local copy to a fork of the repository.

We do want to keep up to date with recent tags and releases, but we no longer have a number of the devices in house that we used to test on and previous updates have broken older device compatibilities before so we pretty much only update when we find someone who has a problem.

russell-taylor, 2022-05-25:
Pushed branch to a fork and redirected our build to use it.

Thanks for maintaining hidapi!

closed 2022-05-25 by russell-taylor