libusb/hidapi #429: find_package(Iconv) fails when both external and internal iconv implementations exist

Youw, 2022-06-13:
Reported at: https://github.com/libusb/hidapi/pull/426
Failed sample: https://github.com/openwrt/packages/actions/runs/2477509891

Youw, 2022-06-13:
> https://github.com/openwrt/packages/runs/6838046113?check_suite_focus=true

@neheb would it be possible to see the content of:
> "/home/build/openwrt/build_dir/target-x86_64_musl/hidapi-hidapi-0.12.0/CMakeFiles/CMakeOutput.log".
> "/home/build/openwrt/build_dir/target-x86_64_musl/hidapi-hidapi-0.12.0/CMakeFiles/CMakeError.log".

Maybe you can run the build locally or so.

---
And the content of:
> "/home/build/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-stub/lib"

I woulder what CMake tries (and fails) to find there.

Youw, 2022-06-13:
Looks like this is a very specific case, during the cross-compilation, and when iconv is cross-compiled as well, and passed to CMake build externally.
This might be an issue specific to openwrt build infrastructure.

Youw, 2022-06-14:
Problem analysis:

From the logs I can tell that openwrt doesn't follow minimal guidelines for CMake [cross-compilation](https://cmake.org/cmake/help/book/mastering-cmake/chapter/Cross%20Compiling%20With%20CMake.html) - no propper toolchain or specifying paths/[environment](https://cmake.org/cmake/help/latest/variable/CMAKE_PREFIX_PATH.html#variable:CMAKE_PREFIX_PATH) to the libraries, the way CMake can recognize it.

One way of fixing it on the environment side is to set the `CMAKE_PREFIX_PATH` to:
`/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr;/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/libiconv-full;/home/mangix/devstuff/openwrt/staging_dir/target-mips_24kc_musl/usr/lib/libintl-full`.

But that way, obviously, requres changes to `cmake.mk` (maybe even more), which I don't expect to be done any time soon.
Making a library-specific (Iconv) fix for HIDAPI only at `hidapi/Makefile` globally doesn't make much sense, because it is not the first cmake build/package that is broken that way [as far as I can tell](https://github.com/openwrt/packages/pull/18738#issuecomment-1152741609).

---
On the bright side:

I can make a workaround on the HIDAPI side, so the Iconv will be found by HIDAPI/CMake even if CMake fails to do so by default.

neheb, 2022-06-15:
That'd be great.

PR #430 (2022-06-21): Ensure Iconv is found when provided via CFLAGS/LDFLAGS
- by default find_file/find_library doesn't respect CFLAGS/LDFLAGS, and FindIconv fails to find Iconv;
- by explicitly trying to link against `-liconv` - we're checking if library is available in such way;
- additionally: if Iconv is detected as BUILT_IN, no need to explicitly depend on `Iconv::Iconv`;

Fixes: #429
closed 2022-06-21 by Youw