libusb/hidapi #436: Different output using hidapitester between 0.10.11 and 0.12.0

nicoduj, 2022-07-26:
Hi, 

I have different output using those 2 version, with hidapitester. Moreover, on MAcOS Monterey, when opening a bluetooth device, it doesn't work without sudo (and after authorizing terminal to read inputs in system preference) 

I have also posted the issue there, don't know if it is the lib or the tool : https://github.com/todbot/hidapitester/issues/16

Results for with 0.10.11
```
imac-de-nico:hidapitester nico$ ./hidapitester --list
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
0000/0000:  - Apple Mikey HID Driver
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
imac-de-nico:hidapitester nico$ ./hidapitester -vidpid 046D/C52B --list-detail
hidapitester: invalid option -- i
hidapitester: invalid option -- d
hidapitester: invalid option -- p
hidapitester: invalid option -- i
hidapitester: invalid option -- d
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
  vendorId:      0x05AC
  productId:     0x8294
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968342

05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
  vendorId:      0x05AC
  productId:     0x8294
  usagePage:     0x0001
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968342

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0006
  serial_number:  
  interface:     0 
  path: DevSrvsID:4294968380

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968387

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968387

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x000C
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968387

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0080
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968387

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFFBC
  usage:         0x0088
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294968387

0000/0000:  - Apple Mikey HID Driver
  vendorId:      0x0000
  productId:     0x0000
  usagePage:     0x000C
  usage:         0x0001
  serial_number:  
  interface:     -1 
  path: DevSrvsID:4294968463

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0001
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294968403

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0002
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294968403

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0004
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294968403

05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
  vendorId:      0x05AC
  productId:     0x8294
  usagePage:     0x0001
  usage:         0x0006
  serial_number:  
  interface:     0 
  path: DevSrvsID:4294968390
```

Results with 0.12.0 : 
```
mac-de-nico:logiSwitch nico$ ./hidapitester --list
046D/C52B: Logitech - USB Receiver
0000/0000:  - Apple Mikey HID Driver
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
imac-de-nico:logiSwitch nico$ ./hidapitester -vidpid 046D/C52B --list-detail
hidapitester: invalid option -- i
hidapitester: invalid option -- d
hidapitester: invalid option -- p
hidapitester: invalid option -- i
hidapitester: invalid option -- d
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
  vendorId:      0x05AC
  productId:     0x8294
  usagePage:     0x0001
  usage:         0x0006
  serial_number:  
  interface:     0 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS06@14500000/Bluetooth USB Host Controller@14500000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0006
  serial_number:  
  interface:     0 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS04@14300000/USB Receiver@14300000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS04@14300000/USB Receiver@14300000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

0000/0000:  - Apple Mikey HID Driver
  vendorId:      0x0000
  productId:     0x0000
  usagePage:     0x000C
  usage:         0x0001
  serial_number:  
  interface:     -1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SBUS@1F,4/AppleSMBusPCI/BUS0@0/AppleSMBusControllerICH/MKY0/AppleMikeyDriver/AppleMikeyHIDDriver

05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
  vendorId:      0x05AC
  productId:     0x8294
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS06@14500000/Bluetooth USB Host Controller@14500000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0001
  serial_number:  
  interface:     2 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS04@14300000/USB Receiver@14300000/IOUSBHostInterface@2/AppleUserUSBHostHIDDevice
```

Youw, 2022-07-26:
> Results for with 0.10.11

I assume that is 0.10.**1**.

Could you chech the behavior with `0.11.0` for comparison?

> don't know if it is the lib or the tool

Could you try using [hidtest](https://github.com/libusb/hidapi/tree/master/hidtest)?

mcuee, 2022-07-31:
@nicoduj 

Are you testing on the same computer and same mac OS version?

BTW, you need to fix your command line. It is `--vidpid` and not `-vidpid`. And if you following the help you should use `--vidpid 046d:c52b`, even though --vidpid 046d/c52b` seems to work fine as well.

mcuee, 2022-07-31:
This is what I get from latest hidapitester git and hidapi git under my macOS M1 (macOS 12.4), similar to your first one.

```

mcuee@mcuees-Mac-mini hidapitester % ./hidapitester --list
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
0000/0000: APPL - BTM
046D/1000: Logitech - LogiVirtualHIDDevice
046D/1000: Logitech - LogiVirtualHIDDevice
0000/0000: Apple - 
046D/B33D: Unknown - Keyboard K480
046D/B33D: Unknown - Keyboard K480
046D/B33D: Unknown - Keyboard K480
046D/B33D: Unknown - Keyboard K480
046D/B33D: Unknown - Keyboard K480
1915/1025: ZY.Ltd - ZY Control Mic
1915/1025: ZY.Ltd - ZY Control Mic
1915/1025: ZY.Ltd - ZY Control Mic
1915/1025: ZY.Ltd - ZY Control Mic
1915/1025: ZY.Ltd - ZY Control Mic
045E/082F: Microsoft - Microsoft Bluetooth Mouse
045E/082F: Microsoft - Microsoft Bluetooth Mouse
045E/082F: Microsoft - Microsoft Bluetooth Mouse
045E/082F: Microsoft - Microsoft Bluetooth Mouse
1915/1025: ZY.Ltd - ZY Control Mic
0000/0000: Apple - Headset
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver

mcuee@mcuees-Mac-mini hidapitester % ./hidapitester --vidpid 046d:c52B --list-detail 
046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0006
  serial_number:  
  interface:     0 
  path: DevSrvsID:4294969890

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294969889

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294969889

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x000C
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294969889

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0080
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294969889

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFFBC
  usage:         0x0088
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294969889

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0001
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294969891

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0002
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294969891

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0004
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294969891

```

nicoduj, 2022-07-31:
Hi,

@mcuee : Sorry for the command line, indeed it was wrong but results is the same. I am testing exactly on the same Mac (Imac 5K intel, Monterey 12,5) . The thing is that the number of devices listed is not the same.

@Youw : yes, sorry, 0.10.1 . I have just tested by recompiling with hidapi 0.11.0 and have the same output (which seems wrong ?) : 

```
iMac-de-Nico:hidapitester nico$ ./hidapitester --list
0000/0000:  - Apple Mikey HID Driver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
046D/C52B: Logitech - USB Receiver
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
05AC/8294: Broadcom Corp. - Bluetooth USB Host Controller
iMac-de-Nico:hidapitester nico$ ./hidapitester --vidpid 046D:C52B --list-detail
046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0001
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294979353

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0002
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294979353

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFF00
  usage:         0x0004
  serial_number:  
  interface:     2 
  path: DevSrvsID:4294979353

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0002
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294979350

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0001
  serial_number:  
  interface:     1 
  path: DevSrvsID:4294979350

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
[... 114 lines cut ...]
  interface:     1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14100000/IOUSBHostDevice@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14140000/USB Receiver@14140000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0x0001
  usage:         0x0080
  serial_number:  
  interface:     1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14100000/IOUSBHostDevice@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14140000/USB Receiver@14140000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

046D/C52B: Logitech - USB Receiver
  vendorId:      0x046D
  productId:     0xC52B
  usagePage:     0xFFBC
  usage:         0x0088
  serial_number:  
  interface:     1 
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14100000/IOUSBHostDevice@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14140000/USB Receiver@14140000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice

iMac-de-Nico:hidapitester nico$ 
```

So I think it is the recompile process in my env that does "something" wrong.

mcuee, 2022-08-01:
@nicoduj 

So basically hidapi itself works fine. But the hidapitester 0.1 release here (https://github.com/todbot/hidapitester/releases) seems to have issues under macOS.

In that case, we can close the issue here.

closed 2022-08-01 by mcuee