libusb/hidapi #437: cannot show some devices

jiangzhouq, 2022-08-08:
I don't know where I'm going wrong, my mac can't display some devices that "system_profiler SPUSBDataType" and "ls usb" can.
here is what they output.
**I want to find "SKY32 Vi" device and connect it , read from it , write to it**

"lsusb"
```
Bus 001 Device 002: ID 0424:7206 SMSC USB7206 Smart Hub 
Bus 001 Device 003: ID 0424:7216 SMSC USB7216 Smart Hub 
Bus 001 Device 007: ID 0bda:8153 Realtek Semiconductor Corp. USB 10/100/1000 LAN  Serial: 0033000001
Bus 001 Device 001: ID 0424:4206 SMSC USB4206 Smart Hub 
Bus 001 Device 006: ID 046d:c52b Logitech Inc. USB Receiver 
Bus 001 Device 005: ID 0424:7260 SMSC USB2 Controller Hub 
Bus 001 Device 004: ID 0424:4216 SMSC USB4216 Smart Hub 
Bus 001 Device 009: ID 20d3:0015 20d3 Cricut Maker.  Serial: 4366121071
Bus 001 Device 008: ID 0424:7240 SMSC USB2 Controller Hub 
Bus 000 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub 
Bus 000 Device 001: ID 05e3:0610 Genesys Logic, Inc. USB2.1 Hub 
Bus 000 Device 003: ID 0483:5750 STMicroelectronics SKY32 Vi  Serial: SKYCUT_F385
Bus 000 Device 000: ID 0424:7206 SMSC USB 3.1 Bus 
Bus 000 Device 000: ID 05e3:0626 Genesys Logic, Inc. USB 3.1 Bus
```

"system_profiler SPUSBDataType"
```
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT6000USBXHCI

    USB 3.1 Bus:

      Host Controller Driver: AppleT6000USBXHCI

        USB7206 Smart Hub:

          Product ID: 0x7206
          Vendor ID: 0x0424  (SMSC)
          Version: 6.25
          Speed: Up to 10 Gb/s
          Manufacturer: Microchip
          Location ID: 0x01200000 / 2
          Current Available (mA): 900
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            USB7216 Smart Hub:

              Product ID: 0x7216
              Vendor ID: 0x0424  (SMSC)
              Version: 1.91
              Speed: Up to 10 Gb/s
              Manufacturer: Microchip
              Location ID: 0x01240000 / 3
              Current Available (mA): 900
              Current Required (mA): 0
              Extra Operating Current (mA): 0

                USB 10/100/1000 LAN:

                  Product ID: 0x8153
                  Vendor ID: 0x0bda  (Realtek Semiconductor Corp.)
                  Version: 33.00
                  Serial Number: 0033000001
                  Speed: Up to 5 Gb/s
                  Manufacturer: Realtek
                  Location ID: 0x01243000 / 7
                  Current Available (mA): 900
                  Current Required (mA): 256
                  Extra Operating Current (mA): 0
[... 490 lines cut ...]
usage_page : 1
vendor_id : 1133

interface_number : -1
manufacturer_string : Apple
path : b'DevSrvsID:4294968506'
product_id : 33028
product_string : 
release_number : 0
serial_number : 
usage : 138
usage_page : 32
vendor_id : 1452

interface_number : -1
manufacturer_string : Apple
path : b'DevSrvsID:4294968507'
product_id : 33028
product_string : 
release_number : 0
serial_number : 
usage : 5
usage_page : 65280
vendor_id : 1452
```

Youw, 2022-08-08:
What makes you think that your "SKY32 Vi" device is a HID device?

Youw, 2022-08-08:
Try `lsusb -v -d 0483:5750`

jiangzhouq, 2022-08-08:
> Try `lsusb --verbose -d 0483:5750`

```
SKY32 Vi:

              Product ID: 0x5750
              Vendor ID: 0x0483  (STMicroelectronics)
              Version: 2.00
              Serial Number: SKYCUT_F385
              Speed: Up to 12 Mb/s
              Manufacturer: SKYicroe
              Location ID: 0x00140000 / 3
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0
```

In windows, it show as a "print support" device.
You're right, it may not be a HID device, I was stuck in there.

Youw, 2022-08-08:
> it may not be a HID device

In which case HIDAPI is not that you need.

closed 2022-08-08 by Youw