libusb/hidapi #438: libusb: avoid sending LIBUSB_DT_REPORT for HID Report Descriptor

Youw, 2022-08-13:
Sometime it is required to get/parse the HID Report Descriptor, e.g.:
- to get [usage/usage_pair](https://github.com/libusb/hidapi/blob/eaa5c7c6f13e212b1936c2d4a7dc372b289aa9e0/libusb/hid.c#L669);
- to return the [descriptor itself](https://github.com/libusb/hidapi/issues/249);

Unfortunately not all devices have `LIBUSB_DT_REPORT` request implemented, even though raw USB descriptors contain the HID Report Descriptor.

Instead of using `LIBUSB_DT_REPORT` there are several possible alternatives:
- read/parse `descriptors` from sysfs - linux-specific solution;
- get/parse HID interface descriptor from libusb (need to check if this is possible);

Youw, 2022-08-13:
> get/parse HID interface descriptor from libusb (need to check if this is possible);

From HID USB specification:
```
The HID class uses the standard request Get_Descriptor as described in the USB
Specification. When a Get_Descriptor(Configuration) request is issued, it
returns the Configuration descriptor, all Interface descriptors, all Endpoint
descriptors, and the HID descriptor for each interface. It shall not return the
String descriptor, HID Report descriptor or any of the optional HID class
descriptors.
```
It shall not return ... **HID Report descriptor** ... .

---
Tried this with libusb: true - with my devices the interface descriptor contains the HID descriptor, but it doesn't include the HID Report descriptor.

---

Looks like it is not an option(

JoergAtGithub, 2022-08-14:
I wonder, what 
> not all devices have `LIBUSB_DT_REPORT` request implemented

means. What type of error do you get, if you try to gather the Report Descriptor?

Do you know, how operating system drivers of Windows or hidraw request the Report Descriptor from these devices?

I guess that usbutils (lsusb & usbhid-dump) fails for these devices too?

Youw, 2022-08-14:
> What type of error do you get, if you try to gather the Report Descriptor?

`libusb_control_transfer() for getting the HID report descriptor failed with -1: LIBUSB_ERROR_IO`

---
Maybe my assumption about `LIBUSB_DT_REPORT` not being supported by the device is not correct - that was my itinial thought.

---
> Do you know, how operating system drivers of Windows or hidraw request the Report Descriptor from these devices?

That's what I intent to find out. Not the first time I'll be going thru hidraw sources.

Youw, 2022-08-14:
> I guess that usbutils (lsusb & usbhid-dump) fails for these devices too?

`sudo lsusb -v` gives me a similar thing I got with libusb:
```
Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
```
Interestingly, `sudo usbhid-dump` actually gives me what I need:
```
001:018:002:DESCRIPTOR         1660470384.085296
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0 06 00 FF 09 04 A1 01 85 20 95
 0E 75 08 15 00 26 FF 00 09 41 81 00 09 41 91 00
 85 21 95 1F 09 42 81 00 09 42 91 00 C0

001:018:001:DESCRIPTOR         1660470384.087046
 05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
 15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 0C
 16 01 F8 26 FF 07 05 01 09 30 09 31 81 06 95 01
 75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
 02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 06 81 03 C0 06 BC FF 09 88
 A1 01 85 08 95 01 75 08 15 01 26 FF 00 19 01 29
 FF 81 00 C0

001:018:000:DESCRIPTOR         1660470384.088021
 05 01 09 06 A1 01 95 08 75 01 15 00 25 01 05 07
 19 E0 29 E7 81 02 81 03 95 05 05 08 19 01 29 05
 91 02 95 01 75 03 91 01 95 06 75 08 15 00 26 FF
 00 05 07 19 00 2A FF 00 81 00 C0
```

Maybe I just need to poke around its sources.

JoergAtGithub, 2022-08-14:
Both are based on libusb. I noticed, that usbhid-dump determines the size of the Report-Descriptor before reading it.

Youw, 2022-08-14:
Hm, that information is available from Interface descriptor. Maybe that's what is missing to make it wokr.
I'll try it (passing the actual Report descriptors size) later today.

Youw, 2022-08-14:
Found [this](https://elixir.bootlin.com/linux/v5.19.1/source/drivers/hid/usbhid/hid-core.c#L1038) in kernel/hidusb driver.
Looks like it uses the `LIBUSB_DT_REPORT` but it specifically requests the size as per HID descriptor.

JoergAtGithub, 2022-08-14:
I saw this code of usbhid-dump: https://github.com/gregkh/usbutils/blame/master/usbhid-dump/src/iface_list.c#L94

Youw, 2022-08-14:
Yeah, that's basically the same thing.

Youw, 2022-08-14:
I made an attempt to fix it here: https://github.com/libusb/hidapi/tree/libusb_report_descriptor_read
But it still doesn't wokr as expected.

I'll be able to check it some time later.

mcuee, 2022-08-15:
@Youw 

libusb xusb may not handle atill the situations but it works to list the HID report descriptors as well for simple cases.
https://github.com/libusb/libusb/blob/master/examples/xusb.c#L609

Example run against teensy 2.0 RawHID device.
```

mcuee@UbuntuSwift3:~/build/libusb$ ./examples/xusb -s 16c0:0486
Using libusb v1.0.26.11742

Opening device 16C0:0486...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 16C0:0486
         bcdDevice: 0102
   iMan:iProd:iSer: 0:1:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 73
         descriptor length: 9
             nb interfaces: 2
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 04
           max packet size: 0040
          polling interval: 01
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 02
           max packet size: 0020
          polling interval: 02

Kernel driver attached for interface 0: 1

Claiming interface 0...

Kernel driver attached for interface 1: 1

Claiming interface 1...

Reading string descriptors:
   String (0x01): "Teensyduino RawHID Device"

Reading OS string descriptor: no descriptor

Reading HID Report Descriptors:

  00000000  06 ab ff 0a 00 02 a1 01 75 08 15 00 26 ff 00 95  ........u...&...
  00000010  40 09 01 81 02 95 40 09 02 91 02 c0              @.....@.....

Skipping Feature Report readout (None detected)

Skipping Input Report readout (None detected)

Releasing interface 0...
Releasing interface 1...
Closing device...

```

Youw, 2022-08-15:
Ah, sorry I didn't explicitly mention it here regarding:
> But it still doesn't wokr as expected.
> I'll be able to check it some time later.

#444 completely resolves the issue now.
The problem was about not sending the correct libusb request to get the HID Report descriptor.

Youw, 2022-08-15:
BTW: in the xusb example [here](https://github.com/libusb/libusb/blob/b88f1441856a9ba481dc40c8c6d0415d8556514a/examples/xusb.c#L617) - there is a bug: `wIndex` parameter is set to `0`, which means the request will work only if the USB Interface №0 is a HID Interface.
`wIndex` has to be set to the HID Interface index.

mcuee, 2022-08-16:
> BTW: in the xusb example [here](https://github.com/libusb/libusb/blob/b88f1441856a9ba481dc40c8c6d0415d8556514a/examples/xusb.c#L617) - there is a bug: `wIndex` parameter is set to `0`, which means [...]

Yes that example is only meant for the paticular HID device in reality :
 `// Microsoft Sidewinder Precision Pro Joystick - 1 HID interface`
(https://github.com/libusb/libusb/blob/b88f1441856a9ba481dc40c8c6d0415d8556514a/examples/xusb.c#L1054)

Youw, 2022-08-16:
Explicitly naming the parameter value as a variable would make a better sample. Just a thought.

PR #444 (2022-08-16): libusb: fix HID Report descriptor request
- pass the actual HID Report descriptor size (from HID descriptor);
- interface_num has to be wIndex, and not as part of wValue for LIBUSB_DT_REPORT request;

TODO:
- [x] Iterate over all Interface descriptors, to find the HID Descriptor (currently assuming the first one);

Resolves: #438
closed 2022-08-16 by Youw