libusb/hidapi #443: libusb backend issue with Teensy HID device

mcuee, 2022-08-14:
From here.
https://github.com/avrdudes/avrdude/blob/main/src/teensy.c#L126-L131
```
        if (pdata->hid_usage == 0)
        {
            // On Linux, libhidapi does not seem to return the HID usage from the report descriptor.
            // We try to infer the board from the part information, until somebody fixes libhidapi.
            // To use this workaround, the -F option is required.
            avrdude_message(MSG_INFO, "%s: WARNING: Cannot detect board type (HID usage is 0)\n", progname);
```

This seems to happen to the libusb backend but not hidraw backend.

mcuee, 2022-08-14:
You can see from the following results that HID usage reporting is not good for the libusb backend.

```
mcuee@UbuntuSwift3:~/build/hidapi/build$ ./hidtest/hidtest_libusb 
hidapi test/example tool. Compiled with hidapi version 0.13.0, runtime version 0.13.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 16c0 0486
  path: 3-1:1.0
  serial_number: (null)
  Manufacturer: (null)
  Product:      Teensyduino RawHID Device
  Release:      102
  Interface:    0
  Usage (page): 0x0 (0x0)

Device Found
  type: 16c0 0486
  path: 3-1:1.1
  serial_number: (null)
  Manufacturer: (null)
  Product:      Teensyduino RawHID Device
  Release:      102
  Interface:    1
  Usage (page): 0x0 (0x0)

unable to open device

mcuee@UbuntuSwift3:~/build/hidapi/build$ ./hidtest/hidtest_hidraw 
hidapi test/example tool. Compiled with hidapi version 0.13.0, runtime version 0.13.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 16c0 0486
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: (null)
  Product:      Teensyduino RawHID Device
  Release:      102
  Interface:    0
  Usage (page): 0x200 (0xffab)

Device Found
  type: 16c0 0486
  path: /dev/hidraw4
  serial_number: 
  Manufacturer: (null)
  Product:      Teensyduino RawHID Device
  Release:      102
  Interface:    1
  Usage (page): 0x4 (0xffc9)

   ... skipped ...

unable to open device

mcuee@UbuntuSwift3:~/build/hidapi/build$ lsusb -vvv -d 16c0:0486

Bus 003 Device 025: ID 16c0:0486 Van Ooijen Technische Informatica Teensyduino RawHID
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x16c0 Van Ooijen Technische Informatica
  idProduct          0x0486 Teensyduino RawHID
  bcdDevice            1.02
  iManufacturer           0 
  iProduct                1 Teensyduino RawHID Device
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0049
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              2 Teensyduino RawHID
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      28
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              3 Emulated Arduino Serial
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      33
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

JoergAtGithub, 2022-08-14:
@mcuee Could you try, what usbhid-dump returns for this device?

Youw, 2022-08-14:
This is a known and documented issue: https://github.com/libusb/hidapi/blob/c3ae3d0b582ba5e6b26af603aea37d988e7eda7b/libusb/hid.c#L729

mcuee, 2022-08-15:
@Youw 
In this case, should I close this issue as this is a known limitation and there is a workaround now? 

For avrdude, I am suggesting to use the hidraw backend.
https://github.com/avrdudes/avrdude/issues/1065#issuecomment-1214373729

Youw, 2022-08-15:
Lets close it after #438 is resolved - that's enother related improvement.

Youw, 2022-08-16:
A [workaround](https://github.com/libusb/hidapi/issues/443#issuecomment-1214433164) is possible, based on current master.

closed 2022-08-16 by Youw

mcuee, 2022-08-16:
> A [workaround](https://github.com/libusb/hidapi/issues/443#issuecomment-1214433164) is possible, based on current master.

Oh yes. Thanks for pointing that out.

avrdude probably will only use the workaround after 0.13 is released.