libusb/hidapi #447: pkg-config files needed for CMake are not correctly installed by `libhidapi-dev`

raleighlittles, 2022-08-26:
# Problem description

I'm trying to use hidapi in a C++ project, and I'm getting this error when trying to compile via CMake:

```
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "Findhidapi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "hidapi", but
  CMake did not find one.

  Could not find a package configuration file provided by "hidapi" with any
  of the following names:

    hidapiConfig.cmake
    hidapi-config.cmake

  Add the installation prefix of "hidapi" to CMAKE_PREFIX_PATH or set
  "hidapi_DIR" to a directory containing one of the above files.  If "hidapi"
  provides a separate development package or SDK, be sure it has been
  installed.

```

# Troubleshooting

I've installed the library headers by running:

```bash
$ sudo apt install libhidapi-dev
```

Verified my library headers are installed:

```
➜ apt list --installed | grep hidapi                               

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libhidapi-dev/jammy,now 0.11.2-1 amd64 [installed]
libhidapi-hidraw0/jammy,now 0.11.2-1 amd64 [installed,automatic]
libhidapi-libusb0/jammy,now 0.11.2-1 amd64 [installed,automatic]

```

I've followed the instructions here on using hidapi in a CMake project: https://github.com/libusb/hidapi/blob/master/BUILD.cmake.md#using-hidapi-in-a-cmake-project

Here's the last few lines of my CMakeLists.txt file, showing that I've included the aforementioned instructions:

```
➜ cat cpp/CMakeLists.txt | tail -n 4
add_executable(${PROJECT_NAME} main.cpp)

find_package(hidapi REQUIRED)
target_link_libraries(${PROJECT_NAME} hidapi::hidapi)% 
```

And my main.cpp:

```
➜ cat cpp/main.cpp | head -n 4
#include <hidapi.h>
#include <wchar.h>

int main(int argc, char* argv[])
```
(The example in the README uses only hidapi.h (no prefix directory), but changing it to that made no difference either

What am I doing wrong?

# System info

* Ubuntu 22.04.1 LTS x86_64
* Kernel 5.15.0-46-generic
* cmake version 3.22.1

mcuee, 2022-08-26:
Latest version is 0.12, can you try that version or even git main?

Youw, 2022-08-26:
> What am I doing wrong?

You're probably doing it right. The problem with the version of the build provided by Ubuntu maintainers.

This repo/org is not responsible for delivering HIDAPI to end users.
Apparently, Ubuntu is not yet using the CMake builds scripts to build/install HIDAPI, at least on version 22.04.

If you want `find_package` to work - you need to build/install HIDAPI yourself (and I don't really advice you to install your own build of HIDAPI into the system).
The alternative is to use [pkg_check_modules](https://cmake.org/cmake/help/latest/module/FindPkgConfig.html#command:pkg_check_modules) until Ubuntu switches to using CMake build for HIDAPI.

Another alternative is to use HIDAPI as a git submodule (or something similar) and use it as CMake [subdirectory](https://github.com/libusb/hidapi/blob/master/BUILD.cmake.md#hidapi-as-a-subdirectory).

closed 2022-08-26 by Youw

raleighlittles, 2022-08-26:
@Youw Can you elaborate on this part:

>  I don't really advice you to install your own build of HIDAPI into the system

Why is this not recommended?

Youw, 2022-08-27:
If you build HIDAPI on your own and install somewhere locally, e.g.:
```sh
cmake -S <HIDAPI source> -B hidapi_build_dir -DCMAKE_INSTALL_PREFIX=$HOME/hidapi_local ...
cd hidapi_build_dir && cmake --build . --target install
```
\- that's fine. Only you know about that build/package, and you can use it correctly.

But if you install in into your _system_, e.g. by using `-DCMAKE_INSTALL_PREFIX=/usr/local` - that will work for some time...
But your _package manager_ (i.e. `apt` on Ubuntu) won't know anything about that installation.
If you happen to install any other package with `apt install` later, that happens to depend on HIDAPI - you'll have either two copies of HIDAPI installed in several different "system" places, or `apt` will overwrite your version of the build with the one that it knows about.
Not to mention possible issue that may happen during the `dist-upgrade` if you happen to use it later.

raleighlittles, 2022-08-29:
@Youw Ahhh okay, I see, my apologies. I didn't realize that your earlier quote was referring to installing hidapi _into any directory besides under`home`_.