libusb/hidapi #448: 64 bytes IN packet cost 2ms on Windows.

cy18, 2022-09-06:
I'm working on a project using USB-FS HID to communication between an MCU and PC. Everything works as expect except the 64 bytes IN packet cost 2ms to transfer.

For testing purpose, the program is quite simple. On MCU, it echos the received packet. On PC, it sends a 64 bytes packet, read the echo packet and check if they matches. Every 1 second, the number of transfer is printed. I have attached the the testing code at last.

On Linux, it works as expected. It cost 1ms to send the packet to MCU, and another 1 ms for the echo packet. Every 1 second, 500 transfers are done. 

However, on Windows, it seems 2 ms are costed for the echo packet to be received. Every transfer cost 3ms and about 333 transfers are done every 1 second. If I reduce the IN packet size from 64 bytes to 63 bytes, it works as expected on Windows, i.e., 500 transfers every 1 second. It's worth noting that the OUT packet size is still 64 bytes. 

Does any one know the reason of this? How can I solve it? Tell me if you need any further information.

PS: On Linux, the hidapi version is 0.10.1 hidapi-libusb. On Windows, I tried both 0.10.1 and 0.12.0, the results are the same.

    #include <stdio.h> // printf
    #include <time.h>

    #include <hidapi.h>

    int main(int argc, char* argv[])
    {
        unsigned char buf[65];
        unsigned char buf2[65];
        hid_device *handle;

        // Initialize the hidapi library
        res = hid_init();

        // Open the device using the VID, PID,
        handle = hid_open(0x28E9, 0x028A, NULL);

        int start = 1;
        int cnt = 0;
        int pre_cnt = 0;
        time_t pre_print_time = time(NULL);
        while(1) {
            buf[0] = 0; // First byte should be zero as no report ID given
            start = start * 2 % 65; // Generate the start byte by a specific pattern for debugging, you can just ignore this 
            for (int i = 1; i < 65; ++i) {
                buf[i] = start + i - 1;
            }
            hid_write(handle, buf, 65);
            int len = hid_read_timeout(handle, buf2, 65, 1000);
            int same = 0;
            for (int i = 0; i < 64; ++i) {
                if (buf[i+1] == buf2[i]) {
                    same++;
                }
            }
            cnt++;
            if (time(NULL) - pre_print_time >= 1) {
                printf("time %d, read at %d cnt %d, same:%d\n", time(NULL), len, cnt-pre_cnt, same);
                pre_print_time = time(NULL);
                pre_cnt = cnt;
            }
            // Print mismatching packets for debugging
            if (same <= 10) {
                for (int i = 0; i < 64; ++i) {
                    printf("%d ", buf[i]);
                }
                printf("\n");
                for (int i = 0; i < 64; ++i) {
                    printf("%d ", buf2[i]);
                }
                printf("\n");
            }
            // Sync IN and OUT packets, necessary on Linux
            if (same < 10) {
                hid_read_timeout(handle, buf2, 65, 100);
            }

        }
    }

Youw, 2022-09-06:
> If I reduce the IN packet size from 64 bytes to 63 bytes

Can you elaborate? Do you mean changing the size of the report in the report descriptor?

cy18, 2022-09-06:
> > If I reduce the IN packet size from 64 bytes to 63 bytes
> 
> Can you elaborate? Do you mean changing the size of the report in the report descriptor?

Yes, change the size in report descriptor from 64 bytes to 63 bytes.
The sending function in MCU is also changed, i.e., send only 63 bytes.
However, nothing changed in the PC program.

Youw, 2022-09-06:
Does your device use control or interrupt output pipe?

cy18, 2022-09-06:
> Does your device use control or interrupt output pipe?

Interrupt output pipe.

Youw, 2022-09-06:
I can only assume that maybe this is some Windows HID driver specifics or a bug.
*Maybe* Widows HID Driver expects that a single packet may be at most 63 bytes size (e.g. to reserve 1 byte for the potential report ID), so with 1ms polling interval it takes 2 ticks to read 64-bytes report.

That is only my assumption. From the API/usage - nothing looks wrong.

---
Could you also share your USB descriptors and the HID Report descriptor?
Maybe someone would have better idea(s).

cy18, 2022-09-07:
> I can only assume that maybe this is some Windows HID driver specifics or a bug. _Maybe_ Widows HID Driver expects that a single packet may be at most 63 bytes size (e.g. to reserve 1 byte for the [...]

Thank you for your reply.

Well, I have also tried report descriptor with report ID set. Both report descriptor with/without ID are attached below. The result is similar. 64-byte OUT packets works as expect, but 64-byte IN packets cost 2ms, while 63-byte packets cost only 1ms. Here 64-byte packets means 1 byte report ID + 63 bytes data. Problem still exists, but at least we can rule out one possibility. 

Since I'm not an expert on USB, I don't have much clue on the problem. Please comment if you have any other assumptions, I can do some test to validate them.

Report descriptor ( This report descriptor is based on a Google search result):

    __ALIGN_BEGIN const uint8_t customhid_report_descriptor[] __ALIGN_END = {
            0x06, 0x00, 0xff, // USAGE_PAGE (Vendor Defined Page 1)
            0x09, 0x00,       // USAGE (Custom Device)
            0xa1, 0x01,       // COLLECTION (Application)
            0x15, 0x00,       //   LOGICAL_MINIMUM (0)
            0x26, 0xff, 0x00, //   LOGICAL_MAXIMUM (255)
            0x75, 0x08,       //   REPORT_SIZE (8)
            0x95, IN_SIZE,    //   REPORT_COUNT (64)
            0x09, 0x01,       //   USAGE (Vendor Usage 1)
            0x81, 0x02,       //   INPUT (Data,Var,Abs)
            0x95, OUT_SIZE,   //   REPORT_COUNT (64)
            0x09, 0x01,       //   USAGE (Vendor Usage 1)
            0x91, 0x02,       //   OUTPUT (Data,Var,Abs)
            0xc0              // END_COLLECTION
    };

Report descriptor with report ID (This report descriptor is based on the example code of the MCU):

    __ALIGN_BEGIN const uint8_t customhid_report_descriptor[] __ALIGN_END =
    {
        0x06, 0x00, 0xFF,  /* USAGE_PAGE (Vendor Defined: 0xFF00) */
        0x09, 0x00,        /* USAGE (Custom Device)               */
        0xa1, 0x01,        /* COLLECTION (Application)            */

        0x85, 0x11,     /* REPORT_ID (0x11)          */
        0x09, 0x01,     /* USAGE (Vendor Usage 1)    */
        0x75, 0x08,     /* REPORT_SIZE (8)           */
        0x95, 64,       /* REPORT_COUNT (64)         */
        0x91, 0x82,     /* OUTPUT (Data,Var,Abs,Vol) */

        0x85, 0x15,     /* REPORT_ID (0x15)          */
        0x09, 0x02,     /* USAGE (Vendor Usage 1)    */
        0x75, 0x08,     /* REPORT_SIZE (8)           */
        0x95, 64,       /* REPORT_COUNT (64)         */
        0x81, 0x82,     /* INPUT (Data,Var,Abs,Vol)  */

        0xc0            /* END_COLLECTION            */
    };

USB descriptor (Also based on the example code of the MCU):

    __ALIGN_BEGIN const usb_desc_dev custom_hid_dev_desc __ALIGN_END =
    {
        .header =
        {
            .bLength          = USB_DEV_DESC_LEN,
            .bDescriptorType  = USB_DESCTYPE_DEV,
        },
        .bcdUSB                = 0x0200U,
        .bDeviceClass          = 0x00U,
        .bDeviceSubClass       = 0x00U,
        .bDeviceProtocol       = 0x00U,
        .bMaxPacketSize0       = USB_FS_EP0_MAX_LEN,
        .idVendor              = USBD_VID,
        .idProduct             = USBD_PID,
        .bcdDevice             = 0x0100U,
        .iManufacturer         = STR_IDX_MFC,
        .iProduct              = STR_IDX_PRODUCT,
        .iSerialNumber         = STR_IDX_SERIAL,
        .bNumberConfigurations = USBD_CFG_MAX_NUM,
    };

    /* USB device configuration descriptor */
    __ALIGN_BEGIN const usb_hid_desc_config_set custom_hid_config_desc __ALIGN_END = 
    {
        .config =
        {
            .header =
            {
                .bLength         = sizeof(usb_desc_config),
                .bDescriptorType = USB_DESCTYPE_CONFIG
            },
            .wTotalLength         = DESC_LEN_CONFIG,
            .bNumInterfaces       = 0x01U,
            .bConfigurationValue  = 0x01U,
            .iConfiguration       = 0x00U,
            .bmAttributes         = 0x80U,
            .bMaxPower            = 0x32U
        },

        .hid_itf =
        {
            .header =
            {
                .bLength         = sizeof(usb_desc_itf),
                .bDescriptorType = USB_DESCTYPE_ITF
            },
            .bInterfaceNumber     = 0x00U,
            .bAlternateSetting    = 0x00U,
            .bNumEndpoints        = 0x02U,
            .bInterfaceClass      = USB_HID_CLASS,
            .bInterfaceSubClass   = 0x00U,
            .bInterfaceProtocol   = 0x00U,
            .iInterface           = 0x00U
        },

        .hid_vendor =
        {
            .header =
            {
                .bLength         = sizeof(usb_desc_hid),
                .bDescriptorType = USB_DESCTYPE_HID
            },
            .bcdHID               = 0x0111U,
            .bCountryCode         = 0x00U,
            .bNumDescriptors      = 0x01U,
            .bDescriptorType      = USB_DESCTYPE_REPORT,
            .wDescriptorLength    = DESC_LEN_REPORT,
        },

        .hid_epin =
        {
            .header =
            {
                .bLength         = sizeof(usb_desc_ep),
                .bDescriptorType = USB_DESCTYPE_EP
            },
            .bEndpointAddress     = CUSTOMHID_IN_EP,
            .bmAttributes         = USB_EP_ATTR_INT,
            .wMaxPacketSize       = CUSTOMHID_IN_PACKET,
            .bInterval            = 1u
        },

        .hid_epout =
        {
            .header =
            {
                .bLength         = sizeof(usb_desc_ep),
                .bDescriptorType = USB_DESCTYPE_EP
            },
            .bEndpointAddress     = CUSTOMHID_OUT_EP,
            .bmAttributes         = USB_EP_ATTR_INT,
            .wMaxPacketSize       = CUSTOMHID_OUT_PACKET,
            .bInterval            = 1u
        }
    };

Youw, 2022-09-07:
What caught my eye:
1) 
> .bConfigurationValue  = 0x01U,

Do yo have multiple configurations? If only one, I'd expect this to be `0`.
But this shouldn't affect the packets size.

2)
> Report descriptor with report ID
> ...
>    `0x85, 0x11,     /* REPORT_ID (0x11)          */`
>    `0x95, 64,       /* REPORT_COUNT (64)         */`

So you have a report ID + 64-bytes report content, which effectively makes it a 65-bytes report data/packet size.
For USB2.0 - this would make 2 USB transfers at best.

cy18, 2022-09-07:
> What caught my eye:
> 
> 
> > .bConfigurationValue  = 0x01U,
> 
> Do yo have multiple configurations? If only one, I'd expect this to be `0`. But this shouldn't affect the packets size.
> 
> 
> > Report descriptor with report ID
> > ...
> > `0x85, 0x11,     /* REPORT_ID (0x11)          */`
> > `0x95, 64,       /* REPORT_COUNT (64)         */`
> 
> So you have a report ID + 64-bytes report content, which effectively makes it a 65-bytes report data/packet size. For USB2.0 - this would make 2 USB transfers at best.
> 
> > 64-byte IN packets cost 2ms. ... Here 64-byte packets means 1 byte report ID + 63 bytes data.
> 
> At least one of - your verbal description or the HID Report are not what you did/meant.

I'm so sorry for the mistake. The second one is not the one I tested. I get the second one from a dirty git revision. The origin value is 63 bytes, while I mistakenly thought it should be 64, so I edited it before posting. I have just did a test. When REPORT_COUNT is 63, it still cost 2 ms and reducing it to 62 would make it normal.

Actually I have got some progress now. Because of a lucky accident, I plug the device to another USB port. You know what, it now works as expected. It seems the problem has something to do with the USB port. However, I'm still trying to find the root cause. Both ports are USB 3.0 ports, so it has nothing to do with USB version.

cy18, 2022-09-07:
> UsbTreeView

From USBTreeView, I cannot tell much difference. I have checked up the BIOS setting and find nothing usefull. I'll try upgrading the BIOS firmware.

[image]

Both controller comes from AMD. When attached to the upper controller, it's OK. While if attached to the second one, it's slow. 

Device info when attached to the upper controller. 

		=========================== USB Port3 ===========================

	Connection Status        : 0x01 (Device is connected)
	Port Chain               : 1-3
	Properties               : 0x01
	 IsUserConnectable       : yes
	 PortIsDebugCapable      : no
	 PortHasMultiCompanions  : no
	 PortConnectorIsTypeC    : no
	ConnectionIndex          : 0x03 (Port 3)
	CompanionIndex           : 0
	 CompanionHubSymLnk      : USB#ROOT_HUB30#5&857d0&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
	 CompanionPortNumber     : 0x07 (Port 7)
	 -> CompanionPortChain   : 1-7

		  ========================== Summary =========================
	Vendor ID                : 0x28E9 (GigaDevice Semiconductor (Beijing) Inc.)
	Product ID               : 0x128A
	USB Version              : 2.0 -> but Device is Full-Speed only
	Port maximum Speed       : High-Speed (Companion Port 1-7 supports SuperSpeed)
	Device maximum Speed     : Full-Speed
	Device Connection Speed  : Full-Speed
	Self powered             : no
	Demanded Current         : 100 mA
	Used Endpoints           : 3

		  ======================== USB Device ========================

			+++++++++++++++++ Device Information ++++++++++++++++++
	Device Description       : USB 输入设备
	Device Path              : \\?\USB#VID_28E9&PID_128A#676A4F990132#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
	Kernel Name              : \Device\USBPDO-7
	Device ID                : USB\VID_28E9&PID_128A\676A4F990132
	Hardware IDs             : USB\VID_28E9&PID_128A&REV_0100 USB\VID_28E9&PID_128A
	Driver KeyName           : {745a17a0-74d3-11d0-b6fe-00a0c90f57da}\0094 (GUID_DEVCLASS_HIDCLASS)
	Driver                   : \SystemRoot\System32\drivers\hidusb.sys (Version: 10.0.22000.1  Date: 2021-06-05)
	Driver Inf               : C:\Windows\inf\input.inf
	Legacy BusType           : PNPBus
	Class                    : HIDClass
	Class GUID               : {745a17a0-74d3-11d0-b6fe-00a0c90f57da} (GUID_DEVCLASS_HIDCLASS)
	Service                  : HidUsb
	Enumerator               : USB
	Location Info            : Port_#0003.Hub_#0002
	Location IDs             : PCIROOT(0)#PCI(0801)#PCI(0003)#USBROOT(0)#USB(3), ACPI(_SB_)#ACPI(PCI0)#ACPI(GP13)#ACPI(XHC0)#USBROOT(0)#USB(3)
	Container ID             : {cf7fc3d7-ba0e-595a-8a31-41b21d9bc0a3}
	Manufacturer Info        : (标准系统设备)
	Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
	Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
	Problem Code             : 0
	HcDisableSelectiveSuspend: 0
	EnableSelectiveSuspend   : 0
	SelectiveSuspendEnabled  : 0
	EnhancedPowerMgmtEnabled : 1
	IdleInWorkingState       : 0
	WakeFromSleepState       : 0
	Power State              : D0 (supported: D0, D2, D3, wake from D0, wake from D2)
	 Child Device 1          : 符合 HID 标准的供应商定义设备
	  Device Path            : \\?\HID#VID_28E9&PID_128A#7&2938e9d4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} (GUID_DEVINTERFACE_HID)
	  Kernel Name            : \Device\000000b1
	  Device ID              : HID\VID_28E9&PID_128A\7&2938E9D4&0&0000
[... 332 lines cut ...]

		  -------------------- String Descriptors -------------------
				 ------ String Descriptor 0 ------
	bLength                  : 0x04 (4 bytes)
	bDescriptorType          : 0x03 (String Descriptor)
	Language ID[0]           : 0x0409 (English - United States)
	Data (HexDump)           : 04 03 09 04                                       ....
				 ------ String Descriptor 1 ------
	bLength                  : 0x16 (22 bytes)
	bDescriptorType          : 0x03 (String Descriptor)
	Language 0x0409          : "GigaDevice"
	Data (HexDump)           : 16 03 47 00 69 00 67 00 61 00 44 00 65 00 76 00   ..G.i.g.a.D.e.v.
							   69 00 63 00 65 00                                 i.c.e.
				 ------ String Descriptor 2 ------
	bLength                  : 0x16 (22 bytes)
	bDescriptorType          : 0x03 (String Descriptor)
	Language 0x0409          : "Controller"
	Data (HexDump)           : 16 03 43 00 6F 00 6E 00 74 00 72 00 6F 00 6C 00   ..C.o.n.t.r.o.l.
							   6C 00 65 00 72 00                                 l.e.r.
				 ------ String Descriptor 3 ------
	bLength                  : 0x1A (26 bytes)
	bDescriptorType          : 0x03 (String Descriptor)
	Language 0x0409          : "676A4F990132"
	Data (HexDump)           : 1A 03 36 00 37 00 36 00 41 00 34 00 46 00 39 00   ..6.7.6.A.4.F.9.
							   39 00 30 00 31 00 33 00 32 00                     9.0.1.3.2.

cy18, 2022-09-07:
After upgrading BIOS and chipset driver, the problem still exists.

Youw, 2022-09-07:
Just be aware, if this was your intention or not:
> USB Version              : 2.0 -> **but Device is Full-Speed only**
> Port maximum Speed       : High-Speed (Companion Port 1-7 supports SuperSpeed)
> Device maximum Speed     : **Full-Speed**

But it shouldn't matter either, USB1.1 (Full-Speed) does support 64-bytes packets just fine.

---
The only difference I could find among controllers - supported power states:
`Power State              : D0 (supported: D0, D2, D3, wake from D0, wake from D2)` - for Upper controller.
`Power State              : D0 (supported: D0, D3, wake from D0)` - for Lower controller.

---
Just to rule out another posibility - disconnect all other USB devices (I see you have 2 Keyboards) from Lower controller and try again.

cy18, 2022-09-08:
> 

About the two keyboards, the first one is the receiver of a wireless mouse, and the second one is a wired keyboard. I have tried removing the mouse and connecting the wired keyboard to another USB controller, it does not work. Actually, the wired keyboard is connected to the upper controller at first place. I did not try removing the keyboard because I have to use keyboard to do the test. Maybe I can try removing the keyboard and start the testing program by some other means. 

It seems upgrading the BIOS or something else has broke my OS. The system crashes to blue screen for at least 10 times. So I tried reinstall the Windows 11. Unfortunately, the problem still exists.

Youw, 2022-09-08:
I meant disconnect keyboard/mouse from the controller where your device is connected.
You can plug it into the *other* controller.

cy18, 2022-09-08:
> I meant disconnect keyboard/mouse from the controller where your device is connected. You can plug it into the _other_ controller.

Thank you. I tried connecting the keyboard to both controllers. The results are the same.

cy18, 2022-09-08:
I just tried removing all USB devices except the custom HID device, then run the testing program from remote desktop. The problem still exists.

mcuee, 2022-09-11:
@cy18 
From your description this seems to be a host USB controller (or root hub) driver issue. Can you try another PC (say an Intel based PC) to see if you have the same problem or not.

cy18, 2022-09-12:
> @cy18 From your description this seems to be a host USB controller (or root hub) driver issue. Can you try another PC (say an Intel based PC) to see if you have the same problem or not.

I have tried on another Intel baed PC which runs Windows 10. There's no problem.

Youw, 2022-09-12:
Unless there are more findings, this looks like a controller/driver bug, and has nothing to do with HIDAPI.

Just for completeness of investigation:
>On Linux, it works as expected.

On Linux on the same machine?
If not, would be interesting to try on the same machine/controller, but using hidraw/libusb backends on Linux.

cy18, 2022-09-13:
> Unless there are more findings, this looks like a controller/driver bug, and has nothing to do with HIDAPI.
>
> Just for completeness of investigation:
>
>
> On Linux on the same machine? If not, would be [...]

I have tried testing it with a Ubuntu live CD on the same machine. The result is quite strange. 

On a USB port, which works normally on windows, the corresponding result of Linux is slow. However, on another USB port, which is slow on Windows, the result of Linux is normal.

Further more, when the result is slow, the specific count of transfers also differs. On both Windows and Linux, both normal ones gives roughly 500 transfers every seconds. However, the result of slow ones differs. On windows, it gives 333 or 334 every second. While on Linux, it gives 343 transfers every seconds. 

In conclusion, for two different USB ports of two different controllers, say port 1 and port 2. On Windows, port 1 gives 500 transfers every second and port 2 gives 333 or 334 transfers every second. While on Linux, port 1 gives 343 transfers every second while port 2 gives 343 transfers every second.

Youw, 2022-09-13:
> I have tried libusb and hidraw, the results are the same.

That can only mean the driver issue. I don't think me or HIDAPI community could be of any help here.

closed 2022-09-13 by Youw