libusb/hidapi #454: 32bit Usage items are not supported by get_usage of libusb backend

JoergAtGithub, 2022-09-11:
The USB standard allows it to specify a usage in two ways:
- Fully qualified as one single 4 byte report descriptor Usage item
- Seperated in an 1 or 2 byte Usage-Page item, followed by an 1 or 2 byte Usage item
The get_usage function of the libusb backend supports only the second.

From USB HID spec 1.11 page 17:
[image]

From USB HID spec 1.11 page 35:
[image]

mcuee, 2023-05-10:
Is it because most of the USB HID Devices use the second way that we have not seen bug report for thr libusb backend?

Youw, 2023-05-10:
Yes, likely

PR #601 (2023-08-26): hidraw: report only Top-Level Usage_page/Usage pairs as 'unique' devices 
To be consistent with Windows/macOS (OS-level implementations).

Fixes: #298
Fixes: #454
commit 8a4de63b (2023-08-26): hidraw: report only Top-Level Usage_page/Usage pairs as 'unique' devices  (#601)

- unique Top-Level Usage_page/Usage: to be consistent with Windows/macOS (OS-level implementations).
- support for single Usage Page item for several UsageIDs; 
- support for 32bit Usage items;

Fixes: #298
Fixes: #454
closed 2023-08-26 by Youw