libusb/hidapi #458: pi user does not have access to usb relays

staebchen0, 2022-09-14:
hi,
System Info:
raspberry pi 4
pyhton 3.9

I cannot address the USB relay with the pi user.
```
pi@raspi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> dev = hid.device()
>>> dev.open(5824,1503)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hid.pyx", line 127, in hid.device.open
OSError: open failed
>>> 
```

When I run the code with root privileges it works
```
root@raspi:/home/pi# python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import hid
>>> dev = hid.device()
>>> dev.open(5824,1503)
>>> 

```

I created the following role with root (there is no other way) 99-personal.rules Folder: /etc/udev/rules.d

> SUBSYSTEM=="usb", ATTR{idVendor}=="5824",ATTR{idProduct}=="1503", MODE="0666"
> KERNEL=="hidraw*", ATTRS{idVendor}=="5824", ATTRS{idProduct}=="1503", MODE="0660", GROUP="plugdev", TAG+="systemd" ENV{SYSTEMD_WANTS}= "usbrelayd.service"
> KERNEL=="hidraw*", ATTRS{idVendor}=="5824", ATTRS{idProduct}=="1503", MODE="0660", GROUP="plugdev", SYMLINK+="usbrelay%b"

updated with:
`sudo udevadm control --reload`

What do I have to do so that the pi user also has the right to execute?

Youw, 2022-09-14:
you're having udev/configuration issue, not an issue with HIDAPI

closed 2022-09-14 by Youw

staebchen0, 2022-09-14:
yes, i know... do you have a tip?

Youw, 2022-09-14:
- `ls -l /dev/hidraw*` - this way you can check file permissions of your device(s)
- play around with udev rules until your user has read/write access to your device

- >KERNEL=="hidraw*", ATTRS{idVendor}=="5824", ATTRS{idProduct}=="1503", MODE="0660", GROUP="plugdev"

  make sure your user is in `plugdev` group

Youw, 2022-09-14:
A general comment:
This is a development repository of HIDAPI library, not a support forum.
Stackoverflow is much better platform for questions/search for tips.

staebchen0, 2022-09-14:
thanks for the info and sorry!