libusb/hidapi #460: set_report issue on macOS

mattbeghin, 2022-09-26:
I need to write data like this to activate DualShock3 controller:
```
uint8_t dataBlob[] = { 0x42, 0x0C, 0x00, 0x00};
IOHIDDeviceSetReport(device, kIOHIDReportTypeFeature, 0xF4, dataBlob, sizeof(dataBlob));
```

If I use hid_write, it will use the first byte of the data buffer as the report_id (0xF4 in this example), and when report_id != 0 it will also send this byte in the data, because of

```
static int set_report(...)
    ...
    if (report_id == 0x0) {
        /* Not using numbered Reports.
           Don't send the report number. */
        data_to_send = data+1;
        length_to_send = length-1;
    }
   ...
```

The test here looks wrong, it should be "if (report_id != 0x0)". Can you confirm ?

Youw, 2022-09-26:
Your device either uses numbered reports, in which case each report starts with its ID, or you have a single report (w/o ID) and you explicitly specify 0 as a report ID as far as `hid_write` conserns.

Unlike `hid_write`, `IOHIDDeviceSetReport` requires not having a report ID as a first byte of report data, if the device does not use numbered reports.

mattbeghin, 2022-09-26:
Then I don't understand how I can send the report mentioned above with the actual hidapi code on macOS. I want to send a report with ID 0xF4 but if I include this ID in my data buffer it will include the report ID in the data too...

Youw, 2022-09-26:
> it will include the report ID in the data too...

That is as per HID Specification. See for instance HID USB 1.11 [Specification](https://www.usb.org/document-library/device-class-definition-hid-111), section 8.2.

closed 2022-09-26 by Youw

mattbeghin, 2022-09-26:
After replacing the test in set_report / macOS version "if (report_id == 0x0)" with "if (report_id != 0x0)", I can replace 

```
uint8_t dataBlob[] = { 0x42, 0x0C, 0x00, 0x00 };
IOHIDDeviceSetReport(device, kIOHIDReportTypeFeature, 0xF4, dataBlob, sizeof(dataBlob));
```
with
```
uint8_t dataBlob[] = { 0xF4, 0x42, 0x0C, 0x00, 0x00 };
hid_send_feature_report(m_device,dataBlob,sizeof(dataBlob));
```

I'll try to read more about HID. If you think this issue is not valid don't hesitate to close it.
Thanks!