libusb/hidapi #462: dispatch.sr.ht scheduled shutdown

Qbicz, 2022-10-13:
Got this email. Not sure if hidapi still depends on `dispatch.sr.ht` for Linux/BSD builds, or only on `builds.sr.ht`?

```
You are receiving this email because you have one or more tasks
configured on [dispatch.sr.ht](http://dispatch.sr.ht/). dispatch has been deprecated and is
scheduled to be shut down on January 1st, 2023. You will need to
migrate your tasks to another solution before this date.

To continue running builds on [builds.sr.ht](http://builds.sr.ht/) from GitHub commits and pull
requests, consider hottub:

https://sr.ht/~emersion/hottub/

To continue running builds on [builds.sr.ht](http://builds.sr.ht/) from GitLab commits and
merge requests, consider dalligi:

https://sr.ht/~emersion/dalligi/

However, we would also like to request that you consider moving away
from [builds.sr.ht](http://builds.sr.ht/) as your CI platform of choice if you intend to only
use builds and not host any of your other services on a free software
platform. Part of the reason dispatch is being deprecated is because it
is an outsized strain on our resources and largely exists to support
non-free platform integrations, which does not align with our mission.
If this describes your project, please consider migrating to a free
software platform or reducing your usage of [builds.sr.ht](http://builds.sr.ht/).

If you have any questions, please reply to this email. Thank you!

--
Drew DeVault
```

Youw, 2022-10-13:
@z3ntu could you please check?

z3ntu, 2022-10-13:
dispatch.sr.ht is basically the bridge between github (and gitlab) and builds.sr.ht so yes this project is affected. 

But it should be fairly trivial to migrate to the hosted instance of hottub

(note: I just tried installing it on another organization I have admin and use builds.sr.ht and am getting an error, I just asked the admin on IRC, I'll keep you updated)

z3ntu, 2022-10-27:
I was just able to install hottub bridge on another organization where I use builds.sr.ht for CI, so seems to work now.

Basically once you get "Installation complete!" from hottub page then it should just work and trigger on PRs as usual (of course if you also have dispatch.sr.ht still installed then it'll build twice, so uninstall that one when hottub works)

z3ntu, 2022-10-27:
And I just requested install of the github app on this repo, organization admin needs to approve.

mcuee, 2022-10-28:
Approvd.

z3ntu, 2022-10-30:
Hm I can't seem to get to the sr.ht oauth dialog page that would finalize the installation. When I have full permission in an organization it also says at the bottom:

> Next: you’ll be directed to the GitHub App’s site to complete setup.

But I don't see how this is possible for libusb organization from my account...

z3ntu, 2022-10-30:
Or @mcuee maybe we can try manually, can you give me the installation ID of the github app? You should be able to see this in the URL bar when you go to https://github.com/organizations/libusb/settings/installations and click "Configure" on "hottub bridge", then there's some number at the end of the URL.

mcuee, 2022-10-30:
> Or @mcuee maybe we can try manually, can you give me the installation ID of the github app? You should be able to see this in the URL bar when you go to [...]

https://github.com/organizations/libusb/settings/installations/30672218

mcuee, 2022-10-30:
@Youw @Qbicz @todbot @jdk 
I have chnaged the roles of hidapi members to the owner for libusb organization.

mcuee, 2022-10-30:
@z3ntu 
Do you want me to add you to libusb organization as well?

z3ntu, 2022-10-30:
In theory hottub bridge should be operational now for the repo.

(for reference, I went to https://hottub.emersion.fr/post-install?installation_id=30672218 and granted oauth access on sr.ht from there)

z3ntu, 2022-10-30:
Seems to work fine I think? https://github.com/libusb/hidapi/pull/475 Not sure how the "Required" status for some of the checks works, maybe this needs some tweaking?

Then we can remove the dispatch.sr.ht integration so builds don't run twice for each configuration.

Youw, 2022-10-30:
> maybe this needs some tweaking?

I'll handle that.

> we can remove the dispatch.sr.ht integration

Lets do it right away

Youw, 2022-10-30:
What I've noticed: for quite some time not all builds.sr.ht are being run.
E.g.: in #467 `freebsd` and `netbsd` are missing, and in #464 - `alpine` and `freebsd`.

Is there a limitation on the number of jobs run?

z3ntu, 2022-10-30:
IIRC at least for dispatch the number of jobs was limited to 4 at a time to avoid CI spam or something?

Hottub seems to have the same restrictions: https://git.sr.ht/~emersion/hottub/commit/c7d8441b6363ccf6c281e79a3a3cc7aa74ed46cb

Youw, 2022-10-30:
That is a bummer.
Any ideas what can we do?
The only first thing that comes to my mind is to decrease the number of jobs, e.g. disable `alpine` (we already have Linux builds testing under Ubuntu), and, maybe lose one of the `bsd` builds.

At least I don't see a way to have a single job for multiple platforms.

z3ntu, 2022-10-30:
I'd imagine Alpine and Arch Linux builds could be moved to GitHub actions? I don't think they have *BSD environments available? Fedora-mingw might also be possible to move, I didn't look much more into it.

z3ntu, 2022-10-30:
By the way Alpine is "special" as in that is uses musl libc and not glibc

Youw, 2022-10-30:
> could be moved to GitHub actions?

Only if used with Docker or something.
Github Actions [only supports](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources) Ubuntu Linux of-the-box.

z3ntu, 2022-10-30:
Oh.. I guess might be an option then for Linux ones at least. I'd keep *BSD ones on sr.ht then because those you can't just put those in Linux containers of course.

mcuee, 2022-10-30:
> > could be moved to GitHub actions?
> 
> Only if used with Docker or something. Github Actions [only supports](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners#supported-runners-and-hardware-resources) Ubuntu Linux of-the-box.

Do you want to consider 3rd github action for BSDs?
1) https://github.com/cross-platform-actions/action
2) https://github.com/vmactions

mcuee, 2022-10-30:
BTW,  not exactly related, but Arduino is using the Crossbuild container for many things, including avrdude static build for Linux (32/64bit x86/ARM), macOS and Windows. Their main goal is to have one binary to work out of the box, even for quite old Linux distros.
https://github.com/arduino/avrdude-packing/blob/main/.github/workflows/release.yml
https://github.com/arduino/crossbuild/blob/main/deps/build_libs.sh

Youw, 2022-10-31:
> Do you want to consider 3rd github action for BSDs?
> 
> 1. https://github.com/cross-platform-actions/action
> 2. https://github.com/vmactions

Oh, these are very interesting-looking. I'd definitely want to look at those sometime.
I only wounder how big the overhead is for the VM to prepare. Both projects claim to aim the low overhead, but I didn't see any numbers in the high-level READMEs.

Youw, 2022-10-31:
I any case I believe we're done with migration to `hottub`. Hence closing this issue.
#476 and #477 to continue the newly raised conversations.

closed 2022-10-31 by Youw