libusb/hidapi #472: Feature report works on windows but fails on linux

luke876, 2022-10-23:
I am trying to control the lighting on my keyboard with hidapi. It works perfectly on windows, but doesn't on linux.
It seems like it's disconnecting as soon as I try to send data to it.

```
$ sudo ./hidapitester -v --vidpid 258a:0059 -l 65 --open --send-feature 10,1,1,2,41,17,0,5,5,255,0,0,0,1
Looking for vid/pid 0x258A / 0x0059  (9610 / 89)
Set buflen to 65
Opening device, vid/pid: 0x258A/0x0059
Writing 65-byte feature report...wrote -1 bytes:
 0A 01 01 02 29 11 00 05 05 FF 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00
Closing device
```

[kernel.log](https://github.com/libusb/hidapi/files/9845631/kernel.log)

Youw, 2022-10-23:
can you share your HID descriptor?

luke876, 2022-10-23:
not sure why there are two but here it is
```
Device Path: /dev/hidraw6
Raw Name: SINO WEALTH RK Bluetooth Keyboar
Raw Phys: usb-0000:07:00.4-2.4/input0
Raw Info:
	bustype: 3 (USB)
	vendor: 0x258a
	product: 0x0059
Report Descriptor Size: 67
Report Descriptor:
05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01 
95 08 75 01 81 02 95 01 75 08 81 03 95 06 75 08 
15 00 26 FF 00 05 07 19 00 2A FF 00 81 00 25 01 
95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 
91 03 C0 

---
Device Path: /dev/hidraw4
Raw Name: SINO WEALTH RK Bluetooth Keyboar
Raw Phys: usb-0000:07:00.4-2.4/input1
Raw Info:
	bustype: 3 (USB)
	vendor: 0x258a
	product: 0x0059
Report Descriptor Size: 125
Report Descriptor:
05 01 09 80 A1 01 85 01 19 81 29 83 15 00 25 01 
75 01 95 03 81 02 95 05 81 01 C0 05 0C 09 01 A1 
01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 
10 81 00 C0 06 00 FF 09 01 A1 01 85 05 15 00 26 
FF 00 19 01 29 02 75 08 95 05 B1 02 C0 05 01 09 
06 A1 01 85 06 05 07 19 04 29 70 15 00 25 01 75 
01 95 78 81 02 C0 06 00 FF 09 01 A1 01 85 0A 15 
00 26 FF 00 09 00 75 08 95 40 B1 02 C0
```

Youw, 2022-10-23:
> not sure why there are two

That could be exactly your issue.
Why exactly there is two devices - that's a separate question. Maybe there is two different devices on different USB interfaces (try [hidtest](https://github.com/libusb/hidapi/tree/master/hidtest) - it prints a little more info, including USB interface the device is defined at).

When you're trying to open a device with `--vidpid 258a:0059` filter, it would open only a first device that matches it, and the order of devices reported is not specified, and often different on different platforms (i.e. Windows vs Linux).

Try https://eleccelerator.com/usbdescreqparser/ to parse your reports.

Looks like your first device is the keyboard, and the seconds one is a special (Sys Control) device. And only the second device has a feature report with id `0A(10)` (the one you're trying to use).

Make sure to use `--usagePage` and `--usage` to open a device that has a specific collection (`0xFF`/`0x01` in your case).

luke876, 2022-10-23:
You are right, I didn't realize there could be multiple devices with vid and pid of 258a:0059.
Both of these work. Thanks for your help.
`sudo ./hidapitester --open-path /dev/hidraw4 -l 65 --send-feature 10,1,1,2,41,17,0,5,5,255,0,0,0,1`
`sudo ./hidapitester --vidpid 258a:0059 --usagePage 1 --usage 6  --open -l 65 --send-feature 10,1,1,2,41,17,0,5,5,255,0,0,0,1`

closed 2022-10-23 by luke876