libusb/hidapi #474: cant open device on raspberry pi 3b+

hannescam, 2022-10-25:
pi@led-panel:~/hidapi/hidtest $ ./test 
hidapi test/example tool. Compiled with hidapi version 0.13.0, runtime version 0.13.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 17ef 6019
  path: 1-1.1.3:1.0
  serial_number: (null)
  Manufacturer: (null)
  Product:      (null)
  Release:      7200
  Interface:    0
  Usage (page): 0x0 (0x0)
  Bus type: 1

unable to open device
pi@led-panel:~/hidapi/hidtest $ 

i have only a usb mouse connected on 64 bit raspbian 
please help

Youw, 2022-10-25:
This is a HIDAPI library bug tracker, not a support forum.

And mouses/keyboards are not supported (and never will be) by HIDAPI.

closed 2022-10-25 by Youw

hannescam, 2022-10-25:
and where can i get support?

hannescam, 2022-10-25:
thankyou