libusb/hidapi #476: Consider Docker images for Alpine and Fedora/MinGW builds

Youw, 2022-10-31:
As per suggestion: https://github.com/libusb/hidapi/issues/462#issuecomment-1296360252
1) To avoid using `builds.sr.ht` and its limitations of only 4 builds at once
2) To have all builds in Github Actions only

mcuee, 2022-10-31:
FYI this docker image approach can be very fast.
Reference: the following is from Arduino avrdude-packing projects which contains six builds (as well as two other actions) at one go.
https://github.com/arduino/avrdude-packing/actions/workflows/release.yml

mcuee, 2022-10-31:
libusb Linux github action also involves using Docker Container for the umockdev test (simulated USB device).

https://github.com/libusb/libusb/actions/runs/2989246626/jobs/4792835136

Youw, 2023-04-30:
I've checked Github Actions Docker Fedora MingGW build - it is insanly fast, i.e. practivally no time/performance overhead because of the Docker image used.
It looks like Github Actions using some caching for Docker images - it takes only ~ 5-6 seconds to initialize.

Youw, 2023-04-30:
The Archlinux takes insignificantly longer to initialize - ~10-11 sec.

Youw, 2023-04-30:
Alpine Docker build is even faster than Fedora.
If it wasn't for a bug I hit in `musl` - it would be so easy to make it work.

Anyways - the 3 PRs complete this story.

mcuee, 2023-05-01:
Good work!

PR #537 (2023-05-01): ci: Docker build for Alpine
Use GithubActions Docker build for Alpine builds.
To avoid using `builds.sr.ht`.
Closes: #476
commit 45a4ba5b (2023-05-01): ci: Docker build for Alpine (#537)

Use GithubActions Docker build for Alpine builds.
To avoid using `builds.sr.ht`.
Closes: #476
closed 2023-05-01 by Youw