libusb/hidapi #477: Consider 3rdparty Github Actions for *bsd builds

Youw, 2022-10-31:
As per: https://github.com/libusb/hidapi/issues/462#issuecomment-1296381299
Possible solutions:
1. https://github.com/cross-platform-actions/action
2. https://github.com/vmactions

Motivation:
1. To avoid using `builds.sr.ht` and its limitations of only 4 builds at once
2. To have all builds in Github Actions only

Youw, 2022-10-31:
I only wounder how big the overhead is for the VM to prepare. Both projects claim to aim the low overhead, but I didn't see any numbers in the high-level READMEs.

mcuee, 2022-10-31:
> I only wonder how big the overhead is for the VM to prepare. Both projects claim to aim the low overhead, but I didn't see any numbers in the high-level READMEs.

I found one project using both (cross-platform-action for BSDs and vmactions for Solaris). Indeed the run seems to take quite some time. Since we do not have so many commits, it may still be acceptable.
https://github.com/gansm/finalcut/actions

mcuee, 2022-11-04:
I tried the NetBSD one on libusb-compat-0.1 and it is not too bad at 32minutes. But I need to check why OpenBSD failed.
https://github.com/mcuee/libusb-compat-0.1/actions/runs/3385196122

Youw, 2022-11-04:
I tend to thing that 1 min vs 32 min might be worth efforts to keep using `builds.sr.ht`. I don't see too much issues with it.
476# should be a good alternative (still needs to be tested, but anyway).

mcuee, 2022-11-06:
In this case, we can close this one.

closed 2022-11-06 by mcuee