libusb/hidapi #478: Consider using umockdev hidraw device for testing under Linux

mcuee, 2022-10-31:
There is an example device for hidraw from umockdev project. Maybe we can integrate it for CI under Linux.
https://github.com/martinpitt/umockdev/tree/main/devices/hidraw

mcuee, 2022-10-31:
Ref: libusb project uses umockdev for Linux testing using a container.
https://github.com/libusb/libusb/blob/master/tests/umockdev.c
https://github.com/libusb/libusb/blob/master/.github/workflows/linux.yml

Youw, 2023-03-12:
The alternative: `uhid`.
In one of the issues there was a [sample](https://github.com/libusb/hidapi/issues/514#issue-1616173667). The sample implementation actually has several issues/bugs, but the overall aproach is fine.

mcuee, 2023-05-14:
I am not so sure if this is possible or not.
https://github.com/vpelletier/python-functionfs

mcuee, 2023-05-14:
BTW, other than CI, it is also good to see if we have some known good HID device or FW to test HIDAPI.

Just wondering what is the test device you are using to test HIDAPI?

For me I usually use whatever HID device I have with me like the following for general test using hidtest.
1) Logitech USB Receiver
2) Plantronics USB Headset
3) My laptop has built-in I2C HID device (Synaptics Touch Pad)
4) USB Mouse/Keyboard
5) Bluetooth Mouse (Microsoft 3600 and maybe others)
6) Logitech Dual Action USB Gamepad
7) Wireless Xbox Gamepad (old version) 
8) ESP32S based BLE HID example

For more real tests with hidapitester or simple test codes.  I do not know much about the FW development other than simple modifications of the USB PIC USB codes and maybe a little bit with USB AVRs.
1) Microchip PIC18F87J50 with custom HID FW (8 bit USB PIC18 MCU). I have USB PIC24 and PIC32 board as well but seldom use them.
2) Arduino Leonardo and Pro Micro based HID example (8 bit USB AVR MCU)
3) Teensy 2.0 and 2.0+ based HID example (8 bit USB AVR MCU)
4) Various STM32 MCU demo boards -- but I do not know much about the FW development
5) Raspberry Pi Pico using C or MicroPython --I do not know much about the FW development.
6) ESP32C3/S2/S3 based boards using C or MicroPython -- again I do not know much about the FW development

Then I use some HID devices to test hidapi indirectly using avrdude (CMSIS DAP v1 compliant).

Youw, 2023-05-14:
Unfortunately I don't have any devices dedicated for testing, only a few of my peripherals, and a few devices from my employer (I'm under NDA not to disclose what those are), which have a few fixed input/output/interrupt reports, that I usually use for some sanity testing.

Youw, 2023-05-14:
Yeah, would be great to have 1-2 FW samples for some well known simple/cheap device to be able to test most (all?) of the HIDAPI.

mcuee, 2023-05-14:
@todbot has some test firmware for hidapitester which can be used for hidapi testing as well.
https://github.com/todbot/hidapitester/tree/master/test_hardware

1) ProMicroRawHID -- USB AVR Atmega32U4
2) TeensyRawHID -- can be used by different Teensy boards (USB AVR and Arm Cortex M MCUs)
3) two tinyusb examples which can be used by many ARM core MCU development boards including STM32 and Raspberry Pi Pico, as well as non-ARM core MCUs like ESP32 S2/S3 or MSP430

TinyUSB 
https://github.com/hathach/tinyusb

mcuee, 2023-05-14:
Jan Axelson has quite some HID examples for things like EZUSB FX2 and Microchip USB PIC.
http://janaxelson.com/hidpage.htm

mcuee, 2023-05-14:
Simple test for Jan Axelson [FX2HID example (0925:1234)](http://janaxelson.com/files/fx2hid.zip) -- loop back of Input/Output report (no report ID) using hidapitest.

For this FW, I have to explicitly send output report ID 0.

```
PS C:\work\hid\hidapitester> .\hidapitester --vidpid 0925:1234 --list-detail
0925/1234: CYPRESS - EZ-USB FX2 HID USBHIDIO
  vendorId:      0x0925
  productId:     0x1234
  usagePage:     0xFFA0
  usage:         0x0001
  serial_number:
  interface:     0
  path: \\?\HID#VID_0925&PID_1234#7&21072e4a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

PS C:\work\hid\hidapitester> .\hidapitester --vidpid 0925:1234 --open --send-output 0,8,9 --read-input
Opening device, vid/pid: 0x0925/0x1234
Writing output report of 64-bytes...wrote 3 bytes:
 00 08 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 2 bytes:
 08 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
PS C:\work\hid\hidapitester> .\hidapitester --vidpid 0925:1234 --open -l 3 --send-output 0,8,9 --read-input
Opening device, vid/pid: 0x0925/0x1234
Writing output report of 3-bytes...wrote 3 bytes:
 00 08 09
Reading 3-byte input report 0, 250 msec timeout...read 2 bytes:
 08 09 00
Closing device
```

mcuee, 2023-05-14:
Simple test using hidapitester for the [Microchip Simple HID demo](https://github.com/Microchip-MPLAB-Harmony/usb_apps_device/tree/master/apps/hid_basic) (which is the base for the FW used in hidtest).

```
1)  toggle LED
PS C:\work\hid\hidapitester> .\hidapitester --vidpid 04d8:003f --open -l 2 --send-output 0x01,0x80
Opening device, vid/pid: 0x04D8/0x003F
Writing output report of 2-bytes...wrote 65 bytes:
 01 80
Closing device
2) read button status
PS C:\work\hid\hidapitester> .\hidapitester --vidpid 04d8:003f --open -l 2 --send-output 0x01,0x81 --read-input
Opening device, vid/pid: 0x04D8/0x003F
Writing output report of 2-bytes...wrote 65 bytes:
 01 81
Reading 2-byte input report 0, 250 msec timeout...read 2 bytes:
 81 01
Closing device
```

mcuee, 2023-05-14:
@Youw 

For libusb testing, I usually recommend Cypress EZ-USB FX3 (SuperSpeed). Then EZ-USB FX2LP (High-Speed).
https://github.com/libusb/libusb/wiki/FAQ#do-you-have-a-list-of-known-good-devices-to-test-libusb

I think for HIDAPI testing, maybe FX3 is a bit overkill but you may want to get cheap EZ-USB FX2LP breakout from online marketplace like AliExpress at about US$5. For example, I have the following board.
[image]

mcuee, 2023-05-15:
For EZ-USB FX2LP, previous examples are using Keil UV2 and Keil C51 toolchain.

Lastes from Cypress (now part of Infineon) is using Eclipse based IDE with Open Source SDCC toolchain.
https://community.infineon.com/t5/Knowledge-Base-Articles/FX2-FX2LP-SDCC-Eclipse-Project-Interrupt-not-Serviced-KBA231372/ta-p/250672

Then there are quite some open source FW library available, using sdcc.
https://github.com/djmuhlestein/fx2lib
https://github.com/whitequark/libfx2

JoergAtGithub, 2023-05-15:
It should be a device, where you can program any HID ReportDescriptor, PhysicalDescriptor and StringDescriptor on low-level (byte-wise). Not something where you configure everything on high-level, because for testing it makes a difference how we order the items in the ReportDescriptor.

mcuee, 2023-05-15:
> It should be a device, where you can program any HID ReportDescriptor, PhysicalDescriptor and StringDescriptor on low-level (byte-wise). Not something where you configure everything on high-level, because for testing it makes a difference how we order the items in the ReportDescriptor.

I believe most of the MCU development board will allow this, including the EZ-USB FX2LP (actually the example code from Jan Axelson uses assembly language for the descriptor).

mcuee, 2023-05-17:
> Edit: take note that in order to develop high speed HID USB device using FX2LP with wMaxPacketSize > 64 Bytes, we can not use EP1IN or EP1OUT, as their buf size is only 64 bytes. We should probably use EP2 and EP6 (buffer size equals to 512 Bytes or 1024 Bytes). So the modification is not as simple. 

I combined the codes from BulkLoop (change to Interrupt Endpoint) and FX2HID and create the new FX2HID FW using EP2 (OUT) and EP6 (IN) and it seems to work fine, with wMaxPacketSize = 512 Bytes and Report Size = 512 Bytes.

```
C:\work\libusb\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester --vidpid 0925:1234 --open -l 513 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,2,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,3,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,4,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,5,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,6,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,7,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,8,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64  --read-input
Opening device, vid/pid: 0x0925/0x1234
Writing output report of 513-bytes...wrote 513 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 02 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 03 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 04 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 05 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 06 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 07 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 08 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40
Reading 513-byte input report 0, 250 msec timeout...read 512 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 02 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 03 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 04 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 05 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 06 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 07 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 08 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 00
Closing device

```

mcuee, 2023-05-17:
1024 bytes wMaxPacketSize and HID INPUT/OUTPUT report size.
It is a bit strange for the last two bytes. I am not so sure what is the problem of the extra bytes of 0x48.

```
C:\work\libusb\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester --vidpid 0925:1234 --open -l 1025 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,2,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,3,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,4,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,5,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,6,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,7,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,8,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,9,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,10,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,11,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,12,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,13,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,14,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,15,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,16,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64  --read-input
Opening device, vid/pid: 0x0925/0x1234
Writing output report of 1025-bytes...wrote 1025 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 02 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 03 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 04 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 05 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 06 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 07 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 08 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 09 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 0A 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40 0B 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
[... cut ...]
 0F 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 10 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 48
 48
Closing device
```
[fx2hid_1024bytes_packet.zip](https://github.com/libusb/hidapi/files/11496555/fx2hid_1024bytes_packet.zip)

mcuee, 2023-05-17:
BTW, I think the above implementation is not fully correct as I have not implemented getting the Input/Output report using Control Transfer. But I think it is good enough for Input/Output transfer using the Interrupt IN/OUT endpoint.

mcuee, 2023-05-23:
> BTW, I think the above implementation is not fully correct as I have not implemented getting the Input/Output report using Control Transfer. I will try that as well later. But I think it is good enough for Input/Output transfer using the Interrupt IN/OUT endpoint.
> 
> I will also try to add Feature reports as well.

For this purpose, I have the modified example from Jan Axelson, using USB PIC (I have PIC18F87J50 USB PIM, other USB PICs should work).
```
C:\work\libusb\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester --vidpid 0925:7001 --open -l 64 --send-output 2,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63  --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 02 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report 0, 250 msec timeout...read 64 bytes:
 01 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
C:\work\libusb\hidapitester [master ≡ +3 ~0 -0 !]> .\hidapitester --vidpid 0925:7001 --open -l 64 --send-feature 3,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63  --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 03 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte feature report, report_id 4...read 64 bytes:
 04 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
```

mcuee, 2023-06-02:
Testing results of two examples from hidapitester.
https://github.com/todbot/hidapitester/issues/20

There are four test FWs in hidapitester. I have only tested the first two examples: ProMicroRawHID and TeensyRawHid. I will try to get the other tinyusb based examples working later.
https://github.com/todbot/hidapitester/tree/master/test_hardware

mcuee, 2023-06-02:
> There are four test FWs in hidapitester. I have only tested the first two examples: ProMicroRawHID and TeensyRawHid. I will try to get the other tinyusb based examples working later.
https://github.com/todbot/hidapitester/tree/master/test_hardware

Here are the test results for SAMD21 (using Arduino M0 clone) hidtest-tinyusb example.
```
Available modes:
  Mode:0   ('IN/OUT reports 32 bytes, no reportIds')
    vid/pid/mfg/prod: 27B8/EE32: hidapitester - INOUT 32bytes
  Mode:1   ('IN/OUT reports 32 bytes, with reportId 1')
    vid/pid/mfg/prod: 27B8/EE33: hidapitester - INOUT 32bytes rId1
  Mode:2   ('Teensy RAWHID like, 64-byte IN/OUT reports, no reportId')
    vid/pid/mfg/prod: 27B8/EEEE: hidapitester - FakeTeensy
  Mode:3   ('blink(1) like, FEATURE reports 1 (8-bytes) and 2 (60-bytes)')
    vid/pid/mfg/prod: 27B8/4444: hidapitester - blink(1) in name only
```

1) Mode 0, echo ON. You can see that hidapitester requires an extra 0 for the OUTPUT report (which will not appear on the wire). For reading of the INPUT report, there will be no extra zero shown.

```
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 27B8:EE32 --open --open -l 33 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32 --read-input
Opening device, vid/pid: 0x27B8/0xEE32
Opening device, vid/pid: 0x27B8/0xEE32
Writing output report of 33-bytes...wrote 33 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20
Reading 33-byte input report 0, 250 msec timeout...read 32 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 00
```

2) Mode 1, echo ON. Report ID is 1 for both INPUT and OUTPUT report. You can see that 0x21,0x22,0x23 were not send on the wire.
```
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 27B8:EE33 --open --open -l 36 --send-output 1,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35 --read-input
Opening device, vid/pid: 0x27B8/0xEE33
Opening device, vid/pid: 0x27B8/0xEE33
Writing output report of 36-bytes...wrote 33 bytes:
 01 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23
Reading 36-byte input report 0, 250 msec timeout...read 33 bytes:
 01 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 00 00 00
Closing device
```

3) Mode 2, echo ON. You can see that hidapitester requires an extra 0 for the OUTPUT report (which will not appear on the wire). For reading of the INPUT report, there will be no extra zero shown.
```
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 27B8:EEEE --open --open -l 65 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59.60,61,62,63,64 --read-input
Opening device, vid/pid: 0x27B8/0xEEEE
Opening device, vid/pid: 0x27B8/0xEEEE
Writing output report of 65-bytes...wrote 65 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3D 3E 3F 40
 00
Reading 65-byte input report 0, 250 msec timeout...read 64 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3D 3E 3F 40 00
 00
Closing device
```
4) Mode 3, echo ON

mcuee, 2023-06-02:
Here are the test results for SAMD21 (using Arduino M0 clone) hidtest_tinyusb_static example.

Mode: blink1 mode, echo ON.
```
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 27B8:01ED --open --open -l 61 --send-feature 2,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60 --read-feature 2
Opening device, vid/pid: 0x27B8/0x01ED
Opening device, vid/pid: 0x27B8/0x01ED
Writing 61-byte feature report...wrote 61 bytes:
 02 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C
Reading 61-byte feature report, report_id 2...read 61 bytes:
 02 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C
Closing device
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 27B8:01ED --open --open -l 9 --send-feature 1,1,2,3,4,5,6,7,8  --read-feature 1                                                                                                        Opening device, vid/pid: 0x27B8/0x01ED
Opening device, vid/pid: 0x27B8/0x01ED
Writing 9-byte feature report...wrote 9 bytes:
 01 01 02 03 04 05 06 07 08
Reading 9-byte feature report, report_id 1...read 9 bytes:
 01 01 02 03 04 05 06 07 08
Closing device
```

mcuee, 2023-06-02:
BTW, Adafruit Arduino TinyUSB library has a generic hid example as well and it should work on many boards.

```
PS C:\work\hid\hidapitester> .\hidapitester.exe --vidpid 239A:801E --open -l 65 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,
12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,
52,53,54,55,56,57,58,59,60,61,62,63,64 --read-input
Opening device, vid/pid: 0x239A/0x801E
Writing output report of 65-bytes...wrote 65 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
 40
Reading 65-byte input report 0, 250 msec timeout...read 64 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40
 00
Closing device
```

mcuee, 2023-06-02:
@Youw and @todbot 

I tend to think the three examples from hidapitester can be of good use to hidapi project testing.
https://github.com/todbot/hidapitester/tree/master/test_hardware

1) ProMicroRawHID: this should work on many Microchip Atmega32U4 based boards like Arduino Micro and Leonardo. The underlying library seems to have a problem with Feature report but it is not difficult to fix.
https://github.com/todbot/hidapitester/blob/master/test_hardware/ProMicroRawHID/ProMicroRawHID.ino

2) hidtest_tinyusb: this should work on many tinyusb supported boards. I have only tested SAMD21 based boards but maybe it can work with other boards as well with some minor changes.
https://github.com/todbot/hidapitester/blob/master/test_hardware/hidtest_tinyusb/hidtest_tinyusb.ino

Maybe we can adapt hidtest.c to work with one of them and use Readme to document the example code for another one.

Youw, 2023-06-02:
"3" That would be great.
   "5" Agree

mcuee, 2023-06-11:
@todbot 

Sorry to disturb you. I believe there is a stupid mistake here in `code.py` so maybe you can help. I am not familiar with Circuit Python.

Code.py is working fine for Generic USB HID device emulation.
```
import usb_hid

# This is only one example of a gamepad descriptor, and may not suit your needs.
CUSTOM_REPORT_DESCRIPTOR = bytes((
    0x06, 0x00, 0xff,  # Usage page (Vendor Defined Page1)
    0x09, 0x01,        # Usage (Vendor Page 1)
    0xA1, 0x01,        # Collection (Application)
    
    0x85, 0x01,        # Report ID (1)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, 0x40,        # Report Count (64 fields)
    0x75, 0x08,        # Report Size (8 bits or 1 byte)
    0x95, 0x40,        # Report Count (64)
    0x82, 0x02, 0x01,  # Input (Data,Var,Abs,Buf)
    
    0x85, 0x02,        # Report ID (2)
    0x09, 0x00,        # Usage (Undefined)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, 0x40,        # Report Count (64 fields)
    0x75, 0x08,        # Report Size (8 bits or 1 byte)
    0x95, 0x40,        # Report Count (64)
    0x92, 0x02, 0x01,  # Output (Data,Var,Abs,Buf)
     
    0xC0,        # End Collection
))

custom_device = usb_hid.Device(
    report_descriptor=CUSTOM_REPORT_DESCRIPTOR,
    usage_page=0xff00,         # Vendor defined
    usage=0x01,                # Vendor page 1
    report_ids=(1, 2),
    in_report_lengths=(64,0),
    out_report_lengths=(0, 64),
)

usb_hid.enable(
    (usb_hid.Device.CONSUMER_CONTROL, # strange that I need to keep one of the three default ones to get this to work
     custom_device)
)
```

But sending input report (looping back from output report) does not work.

```
import time
import usb_hid
import adafruit_hid

custom_device = adafruit_hid.find_device(usb_hid.devices, usage_page=0xff00, usage=0x01)

print("custom_device:", custom_device)

while True:
    report = bytearray(64)  # must be same size as specified in HID Report Descriptor in boot.py    
    report = custom_device.get_last_received_report(2);
    print(["%02x" % x for x in report])
    time.sleep(1)
    report[0] = 1
    custom_device.send_report(report, 1);
```

hidapitester output
```
(py310x64venv) PS C:\work\hid\hidapitester> .\hidapitester --vidpid 2e8a:102e --usagePage=0xff00 --usage 1 --open --list-detail
Opening device, vid/pid:0x2E8A/0x102E, usagePage/usage: FF00/1
Device opened
2E8A/102E: VCC-GND Studio - CircuitPython HID
  vendorId:      0x2E8A
  productId:     0x102E
  usagePage:     0xFF00
  usage:         0x0001
  serial_number: DE6185100F4D6522
  interface:     3
  path: \\?\HID#VID_2E8A&PID_102E&MI_03&Col02#8&15f7af61&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}

Closing device

(py310x64venv) PS C:\work\hid\hidapitester> .\hidapitester --vidpid 2e8a:102e --usagePage=0xff00 --usage 1 --open --send-output 2,3,4,5 --read-input 1
Opening device, vid/pid:0x2E8A/0x102E, usagePage/usage: FF00/1
Device opened
Writing output report of 64-bytes...wrote 65 bytes:
 02 03 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 0 bytes:
Closing device
```

todbot, 2023-06-11:
Hi @mcuee,
The code looks correct from inspection.  I'll look into this when I get back to my bench tomorrow.

As for why you need two descriptors in `enable()`, you're probably running into a Python-ism where one-item tuples are converted to a non-list item. To get around this, add a comma to your one-item tuple.  That is, do this:
```py
usb_hid.enable( (custom_device,) )
```

mcuee, 2023-06-11:
@todbot 

Yes you are right. The following works.
```
usb_hid.enable( (custom_device,) )
```

Relevant discussion here.
* https://github.com/adafruit/Adafruit_CircuitPython_HID/issues/115

todbot, 2023-06-14:
I have had problems with 64-byte reports with report IDs.  Not sure if this is a MCU buffer limitation (which is 64 bytes usually), an issue with hidapi, or something else.

Here is a modified set of above code that does work, up to 63 byte reports:

```py
# rawhid_code.py
import time
import usb_hid
import adafruit_hid

custom_device = adafruit_hid.find_device(usb_hid.devices, usage_page=0xff00, usage=0x01)

print("custom_device: %04x %04x" % (custom_device.usage_page, custom_device.usage) )

while True:
    out_report = custom_device.get_last_received_report(2) # out from computer
    if out_report:
        print("len:",len(out_report),["%02x" % x for x in out_report])
        time.sleep(0.5)
        print("sending copy on reportid 1")
        in_report = bytearray(out_report)  # copy in case we want to modify
        custom_device.send_report(in_report, 1);  # in to computer
```

```py 
# rawhid_boot.py --
# any changes here only reflected after board reset

import usb_hid

REPORT_COUNT = 63  # size of report in bytes

CUSTOM_REPORT_DESCRIPTOR = bytes((
    0x06, 0x00, 0xff,  # Usage page (Vendor Defined Page1)
    0x09, 0x01,        # Usage (Vendor Page 1)
    0xA1, 0x01,        # Collection (Application)

    0x85, 0x01,        # Report ID (1)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, REPORT_COUNT, # Report Count (64 fields)
    0x82, 0x02, 0x01,  # Input (Data,Var,Abs,Buf)

    0x85, 0x02,        # Report ID (2)
    0x09, 0x00,        # Usage (Undefined)
    0x09, 0x00,        # Usage Page (Undefined)
    0x15, 0x00,        # Logical Minimum (0)
    0x26, 0xFF, 0x00,  # Logical Maximum (255)
    0x75, 0x08,        # Report Size (8 bits)
    0x95, REPORT_COUNT, # Report Count (64 fields)
    0x92, 0x02, 0x01,  # Output (Data,Var,Abs,Buf)

    0xC0,        # End Collection
))

custom_device = usb_hid.Device(
    report_descriptor=CUSTOM_REPORT_DESCRIPTOR,
    usage_page=0xff00,         # Vendor defined
    usage=0x01,                # Vendor page 1
    report_ids=(1, 2),
    in_report_lengths=(REPORT_COUNT,0),
    out_report_lengths=(0, REPORT_COUNT),
)

usb_hid.enable( (custom_device,) )
```

And running hidapitester:

```sh
./hidapitester --usagePage 0xff00 --usage 1 --open -l 64 --send-output 2,3,4,5 --timeout 1000 --read-input 1
Opening device, vid/pid:0x0000/0x0000, usagePage/usage: FF00/1
Device opened
Writing output report of 64-bytes...wrote 64 bytes:
 02 03 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 1000 msec timeout...read 64 bytes:
 01 03 04 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

Note that hidapitester is given `-l 64` to make a report that is REPORT_COUNT+1 long

mcuee, 2023-06-14:
> I have had problems with 64-byte reports with report IDs. Not sure if this is a MCU buffer limitation (which is 64 bytes usually), an issue with hidapi, or something else.

@todbot 

Thanks a lot for the help. 

Interesting and this is exactly the same problem I have with my USB PIC using Jan Axelson's generic HID example as well. 63 bytes works but not 64 bytes. 
[Firmware_generic_hid4_63B.zip](https://github.com/libusb/hidapi/files/11751347/Firmware_generic_hid4_63B.zip)

> Note that hidapitester is given -l 64 to make a report that is REPORT_COUNT+1 long.

This is one potential issue of hidapitester I would like to check with you and @Youw. I am thinking this should be worked-around inside hidapitest for Windows (report ID 0 needs to be in the buffer but not really appear in the USB BUS. That is probably why we need one more bytes of bufer than the length of report size. I believe this is not necessary for Linux and Windows. I could be wrong though.

todbot, 2023-06-15:
> > Note that hidapitester is given -l 64 to make a report that is REPORT_COUNT+1 long.
> 
> This is one potential issue of hidapitester I would like to check with you and @Youw.

This reflects how `hidapi` works, where the first byte is the reportId if it exists, which I believe came from how libusb works.  It's always bugged me and has led to so much confusion, but I'm not sure how we could get around it.

mcuee, 2023-06-15:
> This reflects how hidapi works, where the first byte of the send buffer is the reportId if it exists, which I believe came from how libusb works. It's always bugged me and has led to so much confusion, but I'm not sure how we could get around it.

@Youw 

Please take a look when you got the time. Thanks.

mcuee, 2023-06-15:
@Youw and @todbot and @JoergAtGithub 

I think we have gone through this report length previously. Maybe we have to revisit them.

I think the following PR fixes the issue for reading back the input report and it is correct.
* https://github.com/libusb/hidapi/pull/232

I think we need to revisit for Feature report. I think the following commit is kind of correct if we look at from Windows driver point of view. But from a cross-platform API point of view, we may need to revist to align with Linux and macOS, which means we need to focus on the data on the USB bus and we need to drop the extra 0 in front when the report ID is 0 for Windows.
https://github.com/libusb/hidapi/commit/6fcb0bb2282dbb505561a53689eba0c0536570d9

Same for Output report. Right now we may report one more byte than real data on the bus due to Windows driver behavior, when the report ID is 0.

All in all, we may need to align Windows HIDAPI behavior along with Linux and macOS. Need to check with the libusb backend as well.

Another related old PR for macOS.
* https://github.com/libusb/hidapi/pull/70

mcuee, 2023-06-15:
There are many confusions under libusb Windows HID backend as well. I think there are more problems with libusb Windows HID backend (libusb project recommends people to use HIDAPI and not libusb, so it is a legacy thingy).

* https://github.com/libusb/libusb/pull/1285 (this may not be totally correct)
* https://github.com/libusb/libusb/pull/1217 II believe this is wrong)
* https://github.com/libusb/libusb/pull/986 (I tend to think this is mostly correct)

There is another interesting issue there as well (hopefully not affecting HIDAPI).
* https://github.com/libusb/libusb/pull/226

mcuee, 2023-06-15:
@tormodvolden 

FYI as well. I believe https://github.com/libusb/libusb/pull/1217 is incorrect.

Youw, 2023-06-15:
Unfortunately I've never had good test device(s) to check all: input/output/feature/interrupt, with and w/o report ID and with different lengths.

All being said, I think we need to focus to make the behavior (e.g. in terms of the return length, etc.) be consistent on all platforms, specifically to match the documentation.

> Please take a look when you got the time. Thanks.

I'm a little lost in the context and worst of all - cannot properly check it locally...

mcuee, 2023-06-16:
> Unfortunately I've never had good test device(s) to check all: input/output/feature/interrupt, with and w/o report ID and with different lengths.

I will recommend the following devices which will work with the examples here. They are of relative low cost from online stores like AliExpress.
https://github.com/todbot/hidapitester/tree/master/test_hardware

1) Arduino Micro or Leonardo clone (ATmega32U4)
2) Raspeberry Pico (official or from 3rd parties)
3) Microchip/Atmel SAM D21 Mini breakout boards

> All being said, I think we need to focus to make the behavior (e.g. in terms of the return length, etc.) be consistent on all platforms, specifically to match the documentation.

Yes I agree. 

> > Please take a look when you got the time. Thanks.
> 
> I'm a little lost in the context and worst of all - cannot properly check it locally...

I understand. I will try to summarize and create another issue. And hopefully you can get one or two of the test devices without too much difficulties.

Youw, 2023-06-16:
> Arduino Micro or Leonardo clone

Hm, I actually have an Arduino somewhere. Haven't used it for years. Maybe I could use it.
Now it is only the question if finding fime for it.

mcuee, 2023-06-16:
> > Arduino Micro or Leonardo clone
> 
> Hm, I actually have an Arduino somewhere. Haven't used it for years. Maybe I could use it. Now it is only the question if finding fime for it.

No problem, take your time.

The Arduino Uno and Maga 2560 (official version and some clones) come with ATmega16U2. It is a bit tricky to get the rawhid example working.
https://github.com/NicoHood/HID/wiki/RawHID-API

Arduino Leonardo or Arduino Micro or Pro Micro or clones come with ATmega32U4 and then you should have no issues to get rawhid example working.
https://github.com/todbot/hidapitester/tree/master/test_hardware/ProMicroRawHID

mcuee, 2023-06-16:
I am going to move this to discussion and create a new issue to make this one clean. Sorry I have put too much info here and people may get confused.