libusb/hidapi #482: Linux/CMake: find_package(hidapi) cannot find specific version

stsichler, 2022-11-16:
I have built hidapi (0.12.0) from source (using git clone from git tag hidapi-0.12.0) and installed it into a local directory (/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu). 
When I now try to import it into my CMake build system using `find_package(hidapi 0.12.0 CONFIG REQUIRED)` or `find_package(hidapi 0.12.0 REQUIRED)`, CMake rejects the installation. 
CMake Output:
```
CMake Error at xxxx (find_package):
  Could not find a configuration file for package "hidapi" that exactly
  matches requested version "0.12.0".

  The following configuration files were considered but not accepted:

    /home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi/hidapi-config.cmake, version: unknown
```
Note that **it is accepted** when I remove the package version requirement from `find_package()`.

This is strange, because all required files seem to be correctly installed:
```
$ find /home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi
/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi
/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi/hidapi-config.cmake
/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi/libhidapi-release.cmake
/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi/libhidapi.cmake
/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/lib/cmake/hidapi/libhidapi-version.cmake
```

stsichler, 2022-11-16:
Ok, renaming the file `libhidapi-version.cmake` to `hidapi-config-version.cmake` **solves** the problem.

Youw, 2022-11-16:
> In addition, still the question remains whether the other files libhidapi*.cmake are named correctly.

Those are correct, otherwise you wouldn't get CMake targets for HIDAPI working.

PR #484 (2022-11-22): fixed CMake problems (#482, #483)
Fixes: #482, #483
commit 39734493 (2022-11-22): Fix CMake export files (#484)

Fixes: #482, #483
closed 2022-11-22 by Youw