libusb/hidapi #483: Linux/CMake: INTERFACE_INCLUDE_DIRECTORIES property of target hidapi::hidapi not correctly set in libhidapi.cmake (v0.12.0)

stsichler, 2022-11-16:
I have built hidapi (0.12.0) from source (using git clone from git tag hidapi-0.12.0) and installed it into a local directory (/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu) using the `-DCMAKE_INSTALL_PREFIX=/home/xxxx/.lib/hidapi-0.12.0/x86_64-linux-gnu/` CMake configure option.
`hidapi` properly installes into the correct location, but the resulting contents of the generated `libhidapi.cmake` file includes these lines:
```
set_target_properties(hidapi::include PROPERTIES
  INTERFACE_INCLUDE_DIRECTORIES "/hidapi"
)
```
**Note that installation prefix is missing here** from the INTERFACE_INCLUDE_DIRECTORIES path.
In consequence, CMake outputs the following error when trying to consume the `hidapi::hidapi` target:
```
CMake Error in CMakeLists.txt:
  Imported target "hidapi::hidapi" includes non-existent path

    "/hidapi"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

stsichler, 2022-11-16:
Reason found:
`src/CMakeLists.txt` uses `CMAKE_INSTALL_INCLUDEDIR` for definition of target include directories of `hidapi_include` before it is defined by `GNUInstallDirs`.

See this [patch](https://github.com/libusb/hidapi/files/10026378/patch.txt) solving the problem.

Youw, 2022-11-22:
Interesting fact: `CMAKE_INSTALL_INCLUDEDIR` is empty only during the initial CMake run.
That's probably why I didn't catch it myself - I usually run `cmake` many times while I test anything locally.
Definitely need a good CI check for that.

closed 2022-11-22 by Youw