libusb/hidapi #487: hidapi windows doesn't see USBRelay

lfini, 2022-12-01:
I have an USBRelays mini board with USB ID: 0519:2018 vendor is Star micronics (sometimes Usbcreatefun).

It works smoothly with python hidapi on Linux, but I'm not able to use it on Windows 10.

I installed python hidapi (and hidapi.dll) and I can use hid.enumerate() to list attached devices. But the USBRelay board doesn't show (and dev.open(0x519, 0x2918) gives IOError).

hidapi seems OK because I can see other HID devices when I plug them in

Any hints?

Youw, 2022-12-02:
1) what version of hidapi.dll?

2) try [hidtest](https://github.com/libusb/hidapi/tree/master/hidtest) (from master) and lets see what is shows

3) share your device's HID Report descriptor (it is [easy](https://www.systutorials.com/docs/linux/man/8-usbhid-dump/) to do so on Linux)

lfini, 2022-12-03:
1. I'm not sure about hidapi.dll version. I downloaded it fro github a few days ago. Comments in hidapi.h show: copyrighr 2022. 

2. I'll do that, but see my final remark

3.  here's the output from usbhid-dump:

001:006:000:DESCRIPTOR         1670057100.575673
 06 00 FF 09 01 A1 01 09 02 15 00 26 00 FF 75 08
 95 08 81 06 09 02 15 00 26 00 FF 75 08 95 08 91
 06 C0

But I've some news: I could try again on a brand new PC with windows 11: the USBRelay works as expected just with Python 3.11 and hidapi.dll freshly installed.

The laptop for my previous test is Win 11 home all the same, but I messed it up with many attempts with PyUSB and the zelig driver installer. So now when I plug in the relay it shows up as a COM port (but flagged as 'non working'). I do not know how to remove completely the drivers provided by zelig

On the working PC, instead, the relay shows up as "vendor defined HID compatible device"

Youw, 2022-12-03:
> it shows up as a COM port

That's the reason why it doesn't work. COM port is not a HID device.

closed 2022-12-03 by Youw

Youw, 2022-12-03:
Follow from "Step 2. Open the Device Manager": https://www.digitalcitizen.life/completely-remove-faulty-drivers-windows/