libusb/hidapi #493: need longer timeout with send_feature_report() for libusb under Linux

elfmimi, 2023-01-03:
Some devices may require much longer timeout when submitting hid_send_feature_report() .

for such devices hid_send_feature_report() just notifies failure with return value of -1 .

I checked that internal error code libusb is reporting is LIBUSB_ERROR_TIMEOUT ( -7 ) .

Changing value here, from 1000 to, say, 5000 does make difference.
https://github.com/libusb/hidapi/blob/c3a6150d8801f79a59034045275176e9738b13e4/libusb/hid.c#L1471

You curious what the device is?
It is USB-HID control capable LCD monitor from EIZO .  ex.  FlexScan EV2670
Most requests to the monitor, for example a request to change speaker volume, work fine with short timeout .
But one specific request that is to enable or disable picture-by-picture mode only works with long timeout .

**So it is better to have a measure to somehow specify custom value for timeout.**

Youw, 2023-01-03:
I guess it was only a matter of time when that magic number would backfire at us.
On the other hand it's been years and no one complained, until now.

But we definitely need a workaround for that. I think a [libusb-specific](https://github.com/libusb/hidapi/blob/master/libusb/hidapi_libusb.h) option (e.g. `int hid_libusb_set_output_timeout(hid_device *dev, unsigned int timeout)` to be used with `hid_write` and `hid_send_feature_report` would suffice.

elfmimi, 2023-01-03:
Thank you for your quick response. I'm afraid I'm not sure if I want to do a PR myself for now.
It's like I spent way too much effort into finding this out , and so I felt like leaving a short note here.

Easiest workaround is to use libhidapi-hidraw instead of libhidapi-libusb .

Youw, 2023-01-03:
> Easiest workaround is to use libhidapi-hidraw instead of libhidapi-libusb .

I'd say - a perfect one.

Youw, 2023-01-03:
For a historical reference:
HIDRAW has this timeout hard-coded as well: https://github.com/torvalds/linux/blob/69b41ac87e4a664de78a395ff97166f0b2943210/drivers/hid/usbhid/hid-core.c#L931
And it is set to [5](https://github.com/torvalds/linux/blob/69b41ac87e4a664de78a395ff97166f0b2943210/include/linux/usb.h#L1877) seconds.

mcuee, 2023-01-19:
I think we may want to promote the use of the hidraw backend for Linux and not the libusb backend.

Youw, 2023-01-19:
Totally agree. But that does not exclude having a potential fix for libusb backend.

oistra-ah, 2024-03-21:
Hi,
could please someone give a hint: is it possible to increase timeout  for send_feature_report() on Windows? Thanks.
@Youw

mcuee, 2024-03-22:
> Hi, could please someone give a hint: is it possible to increase timeout for send_feature_report() on Windows? I also have a device, which takes a more than 5 sec sometimes for response. Thanks. 

That is interesting, what is the device? The original issue only mentions Linux.

oistra-ah, 2024-03-22:
> > Hi, could please someone give a hint: is it possible to increase timeout for send_feature_report() on Windows? I also have a device, which takes a more than 5 sec sometimes for response. Thanks.
> 
> That is interesting, what is the device? The original issue only mentions Linux and only libusb backend.

Sorry for asking in this topic. I've just created #670. Device is a tiny USB climate sensor on the stm32f4