libusb/hidapi #498: HIDAPI homebrew formular problem

mcuee, 2023-01-14:
This is not really an HIDAPI issue but Homebrew issue.

They do not provide static linking library under Homebrew.

Formula:
https://github.com/Homebrew/homebrew-core/blob/master/Formula/hidapi.rb

Youw, 2023-01-14:
Does Homebrew provides static linking for other libraries?
Does anyone requires a static HIDAPI specifically from Homebrew?

Youw, 2023-01-14:
For (if) a library provided in both shares/static by Homebrew - what is the mechanism to swith among them by the user/client?

mcuee, 2023-01-14:
> Does Homebrew provides static linking for other libraries? Does anyone requires a static HIDAPI specifically from Homebrew?

Yes. For example, there are libftdi1 (CMake based) and libusb-1.0 (auto-tools build) static libraries.

Please refer to the following two avrdude issues.
* https://github.com/avrdudes/avrdude/issues/1159#issuecomment-1382628402

```
mcuee@mcuees-Mac-mini lib % ls -la libhidapi*
lrwxr-xr-x  1 mcuee  admin  50 Jan 14 09:23 libhidapi.0.13.1.dylib -> ../Cellar/hidapi/0.13.1/lib/libhidapi.0.13.1.dylib
lrwxr-xr-x  1 mcuee  admin  45 Jan 14 09:23 libhidapi.0.dylib -> ../Cellar/hidapi/0.13.1/lib/libhidapi.0.dylib
lrwxr-xr-x  1 mcuee  admin  43 Jan 14 09:23 libhidapi.dylib -> ../Cellar/hidapi/0.13.1/lib/libhidapi.dylib

mcuee@mcuees-Mac-mini lib % ls -la libusb*   
lrwxr-xr-x  1 mcuee  admin  52 Dec  5 19:22 libusb-0.1.4.dylib -> ../Cellar/libusb-compat/0.1.8/lib/libusb-0.1.4.dylib
lrwxr-xr-x  1 mcuee  admin  52 Dec 15 20:48 libusb-1.0.0.dylib -> ../Cellar/libusb/HEAD-f3619c4/lib/libusb-1.0.0.dylib
lrwxr-xr-x  1 mcuee  admin  46 Dec 15 20:48 libusb-1.0.a -> ../Cellar/libusb/HEAD-f3619c4/lib/libusb-1.0.a
lrwxr-xr-x  1 mcuee  admin  50 Dec 15 20:48 libusb-1.0.dylib -> ../Cellar/libusb/HEAD-f3619c4/lib/libusb-1.0.dylib
lrwxr-xr-x  1 mcuee  admin  42 Dec  5 19:22 libusb.a -> ../Cellar/libusb-compat/0.1.8/lib/libusb.a
lrwxr-xr-x  1 mcuee  admin  46 Dec  5 19:22 libusb.dylib -> ../Cellar/libusb-compat/0.1.8/lib/libusb.dylib
...
mcuee@mcuees-Mac-mini lib % ls -la libftdi*
lrwxr-xr-x  1 mcuee  admin  48 Dec 15 20:48 libftdi1.2.5.0.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdi1.2.5.0.dylib
lrwxr-xr-x  1 mcuee  admin  44 Dec 15 20:48 libftdi1.2.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdi1.2.dylib
lrwxr-xr-x  1 mcuee  admin  38 Dec 15 20:48 libftdi1.a -> ../Cellar/libftdi/1.5_2/lib/libftdi1.a
lrwxr-xr-x  1 mcuee  admin  42 Dec 15 20:48 libftdi1.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdi1.dylib
lrwxr-xr-x  1 mcuee  admin  50 Dec 15 20:48 libftdipp1.2.5.0.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdipp1.2.5.0.dylib
lrwxr-xr-x  1 mcuee  admin  46 Dec 15 20:48 libftdipp1.3.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdipp1.3.dylib
lrwxr-xr-x  1 mcuee  admin  40 Dec 15 20:48 libftdipp1.a -> ../Cellar/libftdi/1.5_2/lib/libftdipp1.a
lrwxr-xr-x  1 mcuee  admin  44 Dec 15 20:48 libftdipp1.dylib -> ../Cellar/libftdi/1.5_2/lib/libftdipp1.dylib
```

mcuee, 2023-01-14:
> For (if) a library provided in both shares/static by Homebrew - what is the mechanism to swith among them by the user/client?

Exampe from avrdude: for sure  it is not fool-proof, if static library is not present, it will fall back to dynamic library.
https://github.com/avrdudes/avrdude/blob/main/CMakeLists.txt#L120-L136
```
if(USE_STATIC_LIBS)
    set(PREFERRED_LIBELF libelf.a elf)
    set(PREFERRED_LIBUSB libusb.a usb)
    set(PREFERRED_LIBUSB_1_0 libusb-1.0.a usb-1.0)
    set(PREFERRED_LIBHIDAPI libhidapi.a libhidapi-libusb.a libhidapi-hidraw.a hidapi hidapi-libusb hidapi-hidraw)
    set(PREFERRED_LIBFTDI libftdi.a ftdi)
    set(PREFERRED_LIBFTDI1 libftdi1.a ftdi1)
    set(PREFERRED_LIBREADLINE libreadline.a)
else()
    set(PREFERRED_LIBELF elf)
    set(PREFERRED_LIBUSB usb)
    set(PREFERRED_LIBUSB_1_0 usb-1.0)
    set(PREFERRED_LIBHIDAPI hidapi hidapi-libusb hidapi-hidraw)
    set(PREFERRED_LIBFTDI ftdi)
    set(PREFERRED_LIBFTDI1 ftdi1)
    set(PREFERRED_LIBREADLINE readline)
endif()
```

mcuee, 2023-01-19:
BTW, avrdude has its own HIDAPI fork but it is now using the same Windows codes as HIDAPI 0.13.0.
https://github.com/avrdudes/libhidapi

Youw, 2023-01-19:
From [avrdudes/libhidapi](https://github.com/avrdudes/libhidapi) README:

> Currently, the Windows related code does no longer required any patches, so it is identical with the upstream version.

I'd say that is a great win for HIDAPI - we where focused on making the library better based on HID Standard and unification of the behavior across platforms and as a result eliminated a (platform-specific) issue _some_ project had.

mcuee, 2023-01-20:
I think I will close this issue as it is not really an HIDAPI issue, but rather a Homebrew formula problem.

closed 2023-01-20 by mcuee