libusb/hidapi #502: Unable to retrieve bus_type for some devices on Windows

AlexGuo1998, 2023-02-04:
On windows, we currently inspect the parent devnode to get `bus_type`:

https://github.com/libusb/hidapi/blob/4ebce6b5059b086d05ca7e091ce04a5fd08ac3ac/windows/hid.c#L498-L552

However, some filtered HID device (e.g. [imbushuo/mac-precision-touchpad](https://github.com/imbushuo/mac-precision-touchpad)) whose parent is the filter driver, which don't have a correct `compatible_id`. The device tree is like this:

```
...                                           (page usage)
Microsoft Bluetooth Enumerator
|- Bluetooth HID device                                   <- 3
   |- Apple Multi-touch Auxiliary Services    (FF00 0014)
   |- Apple Multi-touch Trackpad HID Filter               <- 2
      |- Microsoft Input Configuration Device (000D 000E)
      |- HID Compactible Trackpad             (000D 0005) <- 1
```

If we want to know the `bus_type` for (1), according to the code, we're inspecting the `compatible_id` for (2), which is empty. 

However if we go one step further we hit (3), which has the correct `compatible_id`: `BTHENUM\{GUID}`. Now we know it's a Bluetooth device.

---

I suggest we go all the way up, until to the root to find a recognizable `compatible_id`, and stop after we found one.

I don't really know if this is the correct / sane thing to do. Could you give me some ideas?

---

BTW, on a UEFI laptop some devices have a ancestor with a `compatible_id` like `*PNP0A03`, but no more recognizable value. Do you know what that is? (Maybe we can open another issue for this?)

Youw, 2023-02-04:
is #464 of any relevance to the issue described here?

AlexGuo1998, 2023-02-05:
I'm not quite sure but maybe no.

In this situation, we don't even find `USB` in any of the `compatible_id`s, while #464 attempts to add some extra logic in `if(wcsstr(compatible_id, L"USB") != NULL)`

JoergAtGithub, 2023-02-05:
@DJm00n can you help with this Bluetooth bus_type issue?

Youw, 2023-02-05:
> I suggest we go all the way up, until to the root to find a recognizable compatible_id, and stop after we found one.

This does sounds like a good solution. But I guess i has to be in line with other changes (including #464).

DJm00n, 2023-03-13:
@AlexGuo1998 thank you for reporting. I'll review `cr = CM_Get_Parent(&dev_node, dev_node, 0);` after #464 get merged.

PS: according to the `C:\Windows\INF\pci.inf` `PNP0A03` is a "PCI Bus".

mcuee, 2023-05-02:
@DJm00n 

#464 has been merged. Maybe you want to take a look at this issue now.