libusb/hidapi #503: macOS: IOHIDDeviceSetReport failed: (0xE00002BC) (iokit/common) general error

benma, 2023-02-06:
Hi

I communicate to a HID device and get this error every once in a while, in around 5% of the messages sent to the device. The messages are all the same and work most of the time.

When compiling this on an old macbook running SDK 10.13, everything works. When compiling in on a new macbook running SDK 13.1, the above issue appears.

Here is the line that produces the error: https://github.com/libusb/hidapi/blob/4ebce6b5059b086d05ca7e091ce04a5fd08ac3ac/mac/hid.c#L1053

When I simply override this to ignore this specific error, everything works normally. The device does receive the data and responds to it normally.

Any idea what could cause this, or how I could debug this?

I also checked dmesg. It shows this when the error happens:

```
IOHIDLibUserClient:0x100002065 setReport failed: 0xe00002bc
```

Seems like this error might coming from `IOHIDLibUserClient::setReport()` here (found via a Google search):

https://opensource.apple.com/source/IOHIDFamily/IOHIDFamily-1633.81.2/IOHIDFamily/IOHIDLibUserClient.cpp.auto.html

mcuee, 2023-02-07:
@benma 

Maybe you want to post the HID report descriptor of this device and the output of hidtest for this device.

You may also try to use hidapitester to test your device.
https://github.com/todbot/hidapitester

Youw, 2023-02-07:
> When compiling this on an old macbook running SDK 10.13, everything works. When compiling in on a new macbook running SDK 13.1, the above issue appears.

What if you _compile_ it on an old macbook and _run_ on the new one.
And the other way around.

Is it consistent on the same device or by the same compilation SDK?

benma, 2023-02-07:
@mcuee thanks, I will look into it.

@Youw when compiling on the old macbook, the program runs fine on old and on new macbooks. Only when compiling on the new macbook the issue appears.

Youw, 2023-02-07:
Sounds like an SDK bug.
- Can you share the output of: `otool -l libhidapi.dylib` and `otool -L libhidapi.dylib`?
- Try specifying `CMAKE_OSX_DEPLOYMENT_TARGET` CMake argument on the "new" machine to something older, e.g.: `-DCMAKE_OSX_DEPLOYMENT_TARGET=10.13` (and share the `otool` output for that variant of binary as well);

benma, 2023-02-07:
@Youw thanks. I agree it kind of sounds like an SDK bug, but I wonder if that the newer SDK might correclty be reporting an error which previous SDKs did not report, though it doesn't help that the error code is the generic error.

The deployment target is already set to 10.13, which is visible in the otool output of the program. I don't have the macbook at hand right now, so I can only paste the full otool output later.

The program contains a statically linked hidapi by nature of using it in Go using [this library](https://github.com/karalabe/usb/) (which I already locally patched to use the newest hidapi version, as the one in that library is outdated).

Youw, 2023-02-08:
> I wonder if that the newer SDK might correctly be reporting an error which previous SDKs did not report

you said it yourself: `When I simply override this to ignore this specific error, everything works normally. The device does receive the data and responds to it normally`

it sounds like there is no actual error, since the device responds the command just fine

---
One more thing to note: can you confirm there is no multithreading involved in the code related to HIDAPI?

benma, 2023-02-08:
It gets stranger! I downloaded Xcode 9.4.1 (SDK 10.13) and Xcode 10.3 (SDK 10.14) on the new MacBook and compiled the program using these Sdks. The issue still appears.

 I did the above by setting `SDKROOT=Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk` and using clang in `Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin`).

There is no multithreading involved.

The `otool -L` output of the test program:

```
 otool -L hidapi-test 
hidapi-test:
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1575.17.0)
	/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.250.1)
	/usr/lib/libresolv.9.dylib (compatibility version 1.0.0, current version 1.0.0)
```

The `otool -l` output of the program when compiled against SDK 10.14 on the new macbook:

```
Load command 7
          cmd LC_LOAD_DYLINKER
      cmdsize 32
         name /usr/lib/dyld (offset 12)
Load command 8
     cmd LC_UUID
 cmdsize 24
    uuid 3025B189-35B9-3F42-85B8-258CDA37BFB0
Load command 9
      cmd LC_VERSION_MIN_MACOSX
  cmdsize 16
  version 10.11
      sdk 10.14
Load command 10
      cmd LC_SOURCE_VERSION
  cmdsize 16
  version 0.0
Load command 11
       cmd LC_MAIN
   cmdsize 24
  entryoff 381568
 stacksize 0
Load command 12
          cmd LC_LOAD_DYLIB
      cmdsize 104
         name /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (offset 24)
   time stamp 2 Thu Jan  1 01:00:02 1970
      current version 1575.17.0
compatibility version 150.0.0
Load command 13
          cmd LC_LOAD_DYLIB
      cmdsize 88
         name /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (offset 24)
   time stamp 2 Thu Jan  1 01:00:02 1970
      current version 275.0.0
compatibility version 1.0.0
Load command 14
          cmd LC_LOAD_DYLIB
      cmdsize 56
         name /usr/lib/libobjc.A.dylib (offset 24)
   time stamp 2 Thu Jan  1 01:00:02 1970
      current version 228.0.0
compatibility version 1.0.0
Load command 15
          cmd LC_LOAD_DYLIB
      cmdsize 56
         name /usr/lib/libSystem.B.dylib (offset 24)
   time stamp 2 Thu Jan  1 01:00:02 1970
      current version 1252.250.1
compatibility version 1.0.0
Load command 16
          cmd LC_LOAD_DYLIB
      cmdsize 56
         name /usr/lib/libresolv.9.dylib (offset 24)
   time stamp 2 Thu Jan  1 01:00:02 1970
      current version 1.0.0
compatibility version 1.0.0
Load command 17
      cmd LC_FUNCTION_STARTS
  cmdsize 16
  dataoff 1544320
 datasize 3368
Load command 18
      cmd LC_DATA_IN_CODE
  cmdsize 16
  dataoff 1547688
 datasize 184
```

Do you know what `current version` refers to? In the otool output of the program built on the old mac, when run on the new mac, contains the same libs but different current versions, e.g:

```
/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1455.122.0)
```

(1575.17.0 vs 1455.122.0). This seems to be the only difference I could still find. The IOKit lib has the same currrent version.

Youw, 2023-02-08:
> (1575.17.0 vs 1455.122.0). This seems to be the only difference I could still find. The IOKit lib has the same currrent version.

I was thinking that probably the version of a target binary differes, but I though the difference will be in `IOKit` (that's where `IOHIDDeviceSetReport` sits), and not in `CoreFoundation`. That is a surprise.

---
I don't know _exactly_ what that is or how it works, but it might just be what is causing it.

Or not.
Just to exclude the other factors (e.g. Go runtime, or something) - please repeat your tests using [hidapitester](https://github.com/todbot/hidapitester) (or make your own small C tool if you like).

Youw, 2023-02-08:
In any case - this is the first time I see a behavior similar to this at all.

benma, 2023-02-10:
Spend more time debugging, and this is what I have so far:

When I run a hid_write/hid_read loop in C, there is no issue! 

When I run the same code linked in a Go program (loop inside the C function called from Go), there is also no issue.

Only when I have the loop in Go and call the C functions to write and read, I get the issue.

If I add `runtime.Gosched()` (invoke the scheduler) in each loop iteration, the issue disappears.

I am still trying to figure out what is going, why the issue appeared in the first place, and why `runtime.Gosched()` fixes it, but I think it's safe to say that the issue is not with this excellent library, so I will close the issue here.

Thanks for your time! I will update this issue anyway if I figure out the details, in case someone else might run into the same issue.

closed 2023-02-10 by benma

Youw, 2023-02-10:
Yeap, multithreading.
I'm pretty sure this explicains the root cause: https://stackoverflow.com/a/13108469/3697939

Just to confirm - what are the Go (compiler) versions on your old/new systems?

benma, 2023-02-11:
@Youw 

> Yeap, multithreading.

I am not so sure - the Go program that exhibits the issue is only a main(){} function containing a loop calling hid_write and hid_read - no goroutines or multithreading. Even if there was multithreading, I'd expect no issues even if not yielding from the goroutine doing the hid_write/hid_read loop. I feel like something else is going on. I thought maybe the garbage collector might be interfering in unexpected ways, but turning off the GC also didn't change anything. 

I tried multiple Go versions between 1.16 and 1.20, all of them have that issue.

Youw, 2023-02-11:
> I'm pretty sure this explicains the root cause: https://stackoverflow.com/a/13108469/3697939

As of Go 1.5, GOMAXPROCS is set to the number of cores of the hardware: [golang.org/doc/go1.5#runtime](http://golang.org/doc/go1.5#runtime)

> I tried multiple Go versions between 1.16 and 1.20, all of them have that issue.

All of those are >= 1.5. I'd say that's expected.

Please confirm the Go version on you "old" mac machine.

---
As an experiment (to confirm or rule out the "multithreading" theory):
modify  HIDAPI source to print the current thread ID inside the set_report function

benma, 2023-02-12:
The Go version on the old mac was 1.18. Everything in the build environment from the old and new mac is the same except the macOS and SDK versions.

I printed the thread ID using `pthread_self()` (`thrd_current` doesn't exist on macOS) and it is the same for all calls, including the ones that fail.

benma, 2023-02-12:
By the way, as an experiment, when I run `runtime.Gosched()` but also have the hid_write/hid_read loop switch OS threads (by using goroutines), the issue appears again. Adding `runtime.LockOSThread()` to fix the goroutine to the OS thread fixes it. In other words, both are required - no multithreading as well as calling `runtime.Gosched()` (skipping either one of them results in the issue reappearing).

Youw, 2023-02-12:
Interesting, I never studied what is  under the hood of goroutines, I only assumed it is a variation of multithreading.

Maybe something similar to coroutines in C++20 (which still might have problems that look like a multithreading problems, when cross-language integration doesn't take those into account).

I think this sample should be a good addition to future "Multithreading support" of Readme.

Youw, 2023-02-12:
Thanks for investigation.

benma, 2023-02-13:
Investigation is not over ;) It is still unclear how the simple single-threaded (no goroutines) loop could exhibit multithreading issues.

As mentioned above, the calls to hid_read hid_write are completely serialized, and the thread ID is constant for all calls too. There must be something else going on, like a memory bug somewhere.

Youw, 2023-02-20:
> I printed the thread ID using `pthread_self()`

I've just looked up some [code](https://github.com/libusb/libusb/blob/master/libusb/os/threads_posix.c#L96-L103) - on macOS for Thread ID one you either use `pthread_threadid_np` or `pthread_mach_thread_np(pthread_self())`.

If you're still up for investigation, can you try that as well?

benma, 2023-02-20:
@Youw for my info, how could that give more insight than `ptread_self()`, which also returns a thread ID? I can try anyway when I find some time. I actually want to double check anyway if the thread ID was constant for all calls, as this was the only problem in the theory that the `hid_` calls needed to all run in the smae OS thread. I kind of hope I made a mistake before when I checked this. In any case, it seems a bit crazy that different `hid_` calls would need to be in the same thread. It seems as if it should work regardless, as long as the calls are serialized of course (mutex).

Youw, 2023-02-20:
> It seems as if it should work regardless, as long as the calls are serialized of course (mutex).

That I agree.

> which also returns a thread ID?

I believe that is not exactly true. 
For starters: from the Linux MAN page:
> The thread ID returned by pthread_self() is not the same thing as
       the kernel thread ID returned by a call to [gettid(2)](https://man7.org/linux/man-pages/man2/gettid.2.html).

And, a thread might be created by something other than `pthread`. I'm not sure how `pthread_self` would behave in that case.

benma, 2023-02-22:
I logged the thread ID as reported by pthread_threadid_np in every hid_read and every hid_write and the value is constant for all of them. The failure happens anyway. Very strange.

In my app moving all calls explicitly to a single OS thread (and invoking the Go scheduler explicitly via runtime.Gosched or implicitly by using Go operations that invoke it, such as reading/writing channels) fixes the issue in practice. Fixing to a single OS thread is also needed otherwise the same error happens.

Very strange behaviour.