libusb/hidapi #506: hid_write Time Is Abnormal.

FeiSong123, 2023-02-12:
I'm calling hid_ When writing, it usually takes 0.3 ms, but it takes more than 5 ms every three to four cycles, hid_ Write is called every 5ms.

Youw, 2023-02-12:
Please share the version of HIDAPI, and you target platform/backend.

FeiSong123, 2023-02-12:
0.13.1  Upper computer Ubuntu 20.04 Lower computer stm32H7. Only tried in Ubuntu.

FeiSong123, 2023-02-12:
I also want to know on which system the libusb in this folder is used. We are using linux now.
[image]

JoergAtGithub, 2023-02-12:
Which backend is used, depends on you application. On Linux, you can link either libusb or linux (hidraw).

5ms to process an OutputReport is not unusual, I know similar times from Windows. This is because hid_write is made synchronous by waiting for the transaction to be completed. 
In theory, an overlapping IO should be possible to achieve a higher OutputReport rate - but this isn't supported by hidapi currently.

FeiSong123, 2023-02-13:
Is 13-14ms normal? So you mean there is no way to solve this problem at present?

mcuee, 2023-02-15:
> Which backend is used, depends on you application. On Linux, you can link either libusb or linux (hidraw).
>
> 5ms to process an OutputReport is not unusual, I know similar times from Windows. This is [...]

Basically you are sayin there is no asynchronous API support in hidapi as of now, right?

mcuee, 2023-02-15:
> Is 13-14ms normal? So you mean there is no way to solve this problem at present?

Have you tried to see if there is a difference using the hidraw backend or libusb backend? I will recommend you to try the hidraw backend to see if that helps.

The OS may play a part as well. Have you tried to use Windows or other Linux distro to see if the behavior is the same or not.

Youw, 2023-02-15:
> Basically you are sayin there is no asynchronous I/O support in hidapi as of now, right?

Technically, the only async I/O is hid_read (the input reports from interrupt pipe). Everything else is done syncronously.
And that is actually a limitation of most backends.

mcuee, 2023-02-15:
Technically there are two ways to write HID Output report -- through the control pipe and the interrupt out pipe (if the device supports this optional feature). For the interrupt output pipe, theoretically we can use either synchronous or asynchronous transfers.

Youw, 2023-02-15:
Right, the question is either the backend supports it:
- on Windows we could WriteFile with Overlaped result;
- on macOS there is `IOHIDDeviceSetReportWithCallback` (if that one is [even functional](https://github.com/libusb/hidapi/issues/385));
- on libusb of course that is possible;
- on hidraw - there is `write` but no indication if it actually respects `O_NONBLOCK` file flag (like it is mentioned for `read`;

mcuee, 2023-02-22:
@FeiSong123 

Just wondering if you have tried the following suggestion?

1) Have you tried to see if there is a difference using the hidraw backend or libusb backend? I will recommend you to try the hidraw backend to see if that helps.

2) The OS may play a part as well. Have you tried to use Windows or other Linux distro to see if the behavior is the same or not.

mcuee, 2023-02-22:
I see. Thanks for the updates.

Since you have found another solution and not using hidapi any more, I will close this issue.

mcuee, 2023-02-22:
> Right, the question is either the backend supports it:
>
> * on Windows we could WriteFile with Overlaped result;
> * on macOS there is `IOHIDDeviceSetReportWithCallback` (if that one is [even [...]

In this case, I guess we will not pursue this improvement any time soon.

closed 2023-02-22 by mcuee

JoergAtGithub, 2023-02-26:
At least there is code on GitHub, which uses IOHIDDeviceSetReportWithCallback: https://github.com/andrei-ch/g403-rgb-led-colors/blob/9d4501b213c2b65bad01615190ac14e36363ead8/G403/AppDelegate.m#L196-L201

JoergAtGithub, 2023-02-26:
While Google implemented for Chromium it's own function for asyncronous SetReport on MacOS:
https://chromium.googlesource.com/chromium/src/+/refs/changes/62/3314562/3/services/device/hid/hid_connection_mac.cc#155