libusb/hidapi #508: HID read normal but write return value is always 63

arilink, 2023-02-16:
hid_read is normal, but hid_write always returns 63 and no out message is detected by bus hound

arilink, 2023-02-17:
[image]
[image]

arilink, 2023-02-17:
I use bus hound to send data to the device, and bus hound can also observe hid out messages, but the return value of hid api write is always 63 and bus hound does not capture out messages

Youw, 2023-02-17:
Depending on device capabilities - the Output report (hid_write) might be sent over Interrupt pipe or over Control pipe.
Input reports (hid_read) are always received over Interrupt pipe.

Try catching Usb Control too.

arilink, 2023-02-17:
The device has a separate out endpoint, will it still send data through the control pipeline?

Youw, 2023-02-17:
It should, but just to be sure.

One more thing: so the data is not being received by the device?

arilink, 2023-02-17:
No The device has not received the data sent from the host computer

arilink, 2023-02-17:
My device defines the out endpoint and uses bus hound to send commands to the device indicating that the device is working fine but I use hidapi to send commands to the device for control and out endpoints and no data is coming out

JoergAtGithub, 2023-02-17:
Does the device describes the report, which you try to send, in the HID ReportDescriptor?

arilink, 2023-02-17:
Yes. The device report descriptor has the out descriptor

JoergAtGithub, 2023-02-17:
Could you please share the HID ReportDescriptor.

arilink, 2023-02-17:
I can't send data to the device using libusb but my friend can send data to the device using a tool made in c#  which is weird

arilink, 2023-02-17:
my email is root@arilink.com    You send me an email and I will send you the descriptor by email

arilink, 2023-02-17:
[USBlyzer Report.pdf](https://github.com/libusb/hidapi/files/10768646/USBlyzer.Report.pdf)
This is HID ReportDescriptor
Thank you very much for your help

JoergAtGithub, 2023-02-17:
Which ReportID has the OutputReport which you try to send, and what is the size in data in bytes you try to send?

arilink, 2023-02-17:
I want to send data to report id = 1

Youw, 2023-02-17:
Can you share the sample of you code that is using HIDAPI?

JoergAtGithub, 2023-02-17:
There is no OutputReport with Report ID 1 and size of 8 bytes in this report descriptor.

arilink, 2023-02-17:
[code.txt](https://github.com/libusb/hidapi/files/10769072/code.txt)
This is code

arilink, 2023-02-17:
[image]
Should I add the out description to the report id 1 like this?

JoergAtGithub, 2023-02-17:
The descriptor is fine, you just need to send the same number of bytes as defined in the descriptor. Otherwise Windows rejects it.

arilink, 2023-02-17:
This device belongs to the client, and I am only responsible for developing the upper computer and communicating with it. The information the client gave me was to send data less than 64 bytes to report 1, but I did not find the relevant definition of out in the top collection

arilink, 2023-02-17:
Any suggestions I'm confused about what to fix

JoergAtGithub, 2023-02-17:
For ReportID 1: Try to send 14 bytes to the joystick and 1 byte to the keyboard

arilink, 2023-02-17:
I send 14 bytes and the result is the same: 63

arilink, 2023-02-17:
> For ReportID 1: Try to send 14 bytes to the joystick and 1 byte to the keyboard

why 14 bytes?

JoergAtGithub, 2023-02-17:
You need to sum up all bits of the Report. All ReportSize x ReportCount. The byte with the ReportID comes on top.

Youw, 2023-02-17:
> I send 14 bytes and the result is the same: 63

I believe it is a limitation of WinAPI. It expects/reports buffers of size that is the size of largest report.

Youw, 2023-02-17:
Can you debug the HIDAPI and check if your device is opened in ReadWrite or ReadOnly mode?
This is the fallback and the description why it is needed: https://github.com/libusb/hidapi/blob/4ebce6b5059b086d05ca7e091ce04a5fd08ac3ac/windows/hid.c#L841

arilink, 2023-02-17:
This report descriptor is actually divided into 2 parts: 1 is JoyStick and 2 is USB HID->PID(about game force feed back). I want to send data to the Device without interfering with 2 Does this mean I need to add the relevant description of out after JoyStick?

arilink, 2023-02-17:
How to check device is opened in ReadWrite or ReadOnly mode?

Youw, 2023-02-17:
Step into the `hid_open_path` and check if the code I mentioned above (`open_device(interface_path, FALSE);`) is executed or does it succeed at `open_device(interface_path, TRUE);`.

arilink, 2023-02-17:
The Qt Creator Cannot step into hid_open_path

Youw, 2023-02-17:
> The Qt Creator Cannot step into hid_open_path

Probably because you didn't build it yourself.

arilink, 2023-02-17:
Yes. i use hidapi  dll

arilink, 2023-02-17:
[image]
Can I read and write correctly if I add the right part to where the left arrow points?

Youw, 2023-02-17:
I'm not sure I follow.

Have you tried to filter/open the device with `usage_page/usage`=`1/5`? Does the hid_write works in that case?

arilink, 2023-02-18:
[image]

Youw, 2023-02-18:
Based on this:
[image]
I assume you should have one.

Can you share the entire HID Report Descriptor as reported by the device?
Can be done on macOS on Linux (e.g. with https://github.com/DIGImend/usbhid-dump).

JoergAtGithub, 2023-02-18:
> Can you share the entire HID Report Descriptor as reported by the device?

He did above: https://github.com/libusb/hidapi/issues/508#issuecomment-1434859965

Youw, 2023-02-18:
Ah, sorry, I missed that. I'll check, thanks.

mcuee, 2023-02-22:
It seems to me the following code is not really opening the right device path.

```
int MainWindow::hid_print()
{
    hid_device_info *hid_info;//usb链表
    hid_device_info *tar_info;
    char *path = nullptr;
        /*打开指定VID PID设备*/
    hid_info = hid_enumerate(0x11ff,0x1112);
    /*遍历所有信息并打印*/
    for(;hid_info != nullptr;hid_info = hid_info->next){
        if(hid_info->interface_number==0)
        {
            qDebug("设备接口号:%d",hid_info->interface_number);
            qDebug("厂商字符串:%ls",hid_info->manufacturer_string);
            qDebug("设备字符串:%ls",hid_info->product_string);
            qDebug("版本号:%d",hid_info->release_number);
            qDebug("地址:%s",hid_info->path);
            tar_info = hid_info;
            path = hid_info->path;
        }

    }
    qDebug("地址:%s",path);
    /*释放链表*/
    hid_free_enumeration(hid_info);
    //handle = hid_open(0x11ff,0x1112,nullptr);
    handle = hid_open_path(path);
```

@arilink 

You may want to print out the output from the hidtest program and from there you should know how to open the right device path.
https://github.com/libusb/hidapi/tree/master/hidtest

arilink, 2023-02-27:
@mcuee  @Youw  @triplef   The problem has been fixed. Thank you very much for your help

closed 2023-02-27 by arilink

Youw, 2023-02-27:
Care to share what was the issue or how it was fixed?

arilink, 2023-03-03:
You may not believe that it is a CPU problem. The computer with the problem I tested was installed with an AMD processor. Using BusHound, I have been unable to capture the out data in the program, but I switched to another computer with an Intel processor. Everything is normal. The two computers. The windows system version is the same as the bushound and QT software versions, so it has nothing to do with HIDAPI

mcuee, 2023-03-03:
I tend to think this is rather a chipset problem. I remember AMD platforms sometimes have problem for USB because of USB Bus Filter. This might cause problems for Bus Hound.

Youw, 2023-03-03:
Yeah, for some reason I'm not suprised either.

arilink, 2023-03-04:
@mcuee  @Youw  @triplef  Yes, I have consulted some engineers and they said that it does happen. Fortunately, my problem has been solved. Thank you very much for your help