libusb/hidapi #519: Compatibility problem

arilink, 2023-03-24:
I used QT+libusb to develop a hid read and write software. Everything worked well on my computer, but I could not communicate on the client's computer. I used the latest release of dll.

arilink, 2023-03-24:
After further analysis, it is located that the problem is due to the USB device has 3 interfaces, we hid read and write is the default interface, leading to some computer read and write failure. Is there any other way to read and write the specified interface?

JoergAtGithub, 2023-03-24:
Use hid_enumerates to get a list of struct hid_device_info. This struct contains interface number and the path needed to call hid_open.
You just need to find your device in the list and than open it by path.

Youw, 2023-03-24:
> QT+libusb

`libusb` or `HIDAPI`?

> Is there any other way to read and write the specified interface?

There is no such concept as to "read or write the specific (USB) interface".
If you're using HIDAPI, then you're always working with HID `device`s, which you should explicitly open by:
- `hid_enumerate` - to find your device (including the filtering based on the interface number, if your device is a USB device;
- `hid_open_path` with `path` from a device you've found on a previouse step;

arilink, 2023-03-25:
@Youw @JoergAtGithub  Ok  I will try it

arilink, 2023-04-07:
use path to control device is work. Pressure test OK . thank you very much

arilink, 2023-04-07:
I will close this  issue

closed 2023-04-07 by arilink