libusb/hidapi #529: No way to get report descriptions

tksuoran, 2023-04-12:
Having access to report descriptions would make it possible to parse reports from device like multi-axis controllers.

It looks like `hid_get_report_descriptor()` currently returns a string and as such is ill-suited for to assist report parsing.

It looks like `hid_winapi_descriptor_reconstruct_pp_data()` already does most of the work. It would be nice to get access to information similar to what can be obtained by `HidP_GetCaps()` / `HidP_GetButtonCaps()` / `HidP_GetValueCaps()`.

tksuoran, 2023-04-12:
Alternatively, something simple to enumerate buttons and values, and able to get their values parsed from reports might be even nicer.

Youw, 2023-04-12:
`hid_get_report_descriptor` does not return a string. It returns a binary representation of HID Report Descriptor.
On all platforms except Windows - it returns the same binary report, as reported by the device itself.
On Windows it returns the same thing as `hid_winapi_descriptor_reconstruct_pp_data` but for a specific device.

> It would be nice to get access to information similar to what can be obtained by HidP_GetCaps() / HidP_GetButtonCaps() / HidP_GetValueCaps().

That is some platform/Windows-specific API.
What you can do instead, is to take the output of `hid_get_report_descriptor` and parse it using one of many HID Report Descriptor parsers, e.g.: https://github.com/listff/hidparser.

Youw, 2023-04-12:
And because there is so many different parsers available for HID Descriptor - there is no need to have one as part of the HIDAPI.

tksuoran, 2023-04-14:
Could you please tell me an example of a parser that works on Windows and linux - Thanks!

tksuoran, 2023-04-14:
Thanks, I those look promising.

closed 2023-04-14 by tksuoran