libusb/hidapi #531: hid_device_info->interface_number zero on macOS 13.3

mcuee, 2023-04-15:
### Discussed in https://github.com/libusb/hidapi/discussions/527

<div type='discussions-op-text'>

<sup>Originally posted by **dingsijie-nreal** April 11, 2023</sup>
I've noticed comment here: https://github.com/libusb/hidapi/blob/master/mac/hid.c#L543
The interesting thing is: macOS Monterey(12.6.3) works perfectly with my usb gadget, I got expected interface numbers.
the output of `test.c` is shown below:
```
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime ve
Compile-time version matches runtime version of hidapi.

Device Found
  type: 3318 0424
  path: DevSrvsID:4298350859
  serial_number: A00016:33:33
  Manufacturer: xxxx
  Product:    xxxx
  Release:      1
  Interface:    3
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
Device Found
  type: 3318 0424
  path: DevSrvsID:4298350872
  serial_number: A00016:33:33
  Manufacturer: xxxx
  Product:      xxxxx
  Release:      1
  Interface:    5
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
Device Found
  type: 3318 0424
  path: DevSrvsID:4298350862
  serial_number: A00016:33:33
  Manufacturer: xxxx
  Product:      xxxxx
  Release:      1
  Interface:    4
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
```
But, after upgrading to macOS 13.3 the result changes to flowing:
```
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 3318 0424
  path: DevSrvsID:4294984005
  serial_number: A00016:33:33
  Manufacturer: xxxxx
  Product:      xxxxx
  Release:      1
  Interface:    0
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
Device Found
  type: 3318 0424
  path: DevSrvsID:4294984010
  serial_number: A00016:33:33
  Manufacturer: xxxxx
  Product:      xxxxx
  Release:      1
  Interface:    0
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
Device Found
  type: 3318 0424
  path: DevSrvsID:4294984008
  serial_number: A00016:33:33
  Manufacturer: xxxxx
  Product:      xxxxx
  Release:      1
  Interface:    0
  Usage (page): 0x0 (0x41)
  Bus type: 1

  Report Descriptor: (27 bytes)
0x05, 0x41, 0x09, 0x00, 0xa1, 0x01, 0x15, 0x00, 0x25, 0xff, 
0x19, 0x01, 0x29, 0x08, 0x95, 0x40, 0x75, 0x08, 0x81, 0x02, 
0x19, 0x01, 0x29, 0x08, 0x91, 0x02, 0xc0, 
```
My mission it to discriminate 3 different components on my USB gadget, and the interface number was the key that I rely on.
But with 0 interfaces numbers, I don't know how to do my job.
Any one get any ideas? Thanks a lot.
 </div>

mcuee, 2023-04-15:
I think this is a real issue.

Last time I can get proper interface number on my Mac Mini M1.
https://github.com/libusb/hidapi/issues/249#issuecomment-905417116

With the latest macOS Ventura 13.3.1 on my Mac Mini M1, interface number becomes 0 for the Logitech USB receiver (046d:c52b).

```
mcuee@mcuees-Mac-mini hidapi % sudo ./hidtest/hidtest
Password:
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 046d c52b
  path: DevSrvsID:4294970451
  serial_number: 
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    0
  Usage (page): 0x2 (0x1)
  Bus type: 1

  Report Descriptor: (148 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0x05, 
0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x03, 0x75, 0x10, 0x95, 
0x02, 0x15, 0x01, 0x26, 0xff, 0x02, 0x19, 0x01, 0x2a, 0xff, 
0x02, 0x81, 0x00, 0xc0, 0x05, 0x01, 0x09, 0x80, 0xa1, 0x01, 
0x85, 0x04, 0x75, 0x02, 0x95, 0x01, 0x15, 0x01, 0x25, 0x03, 
0x09, 0x82, 0x09, 0x81, 0x09, 0x83, 0x81, 0x60, 0x75, 0x06, 
0x81, 0x03, 0xc0, 0x06, 0xbc, 0xff, 0x09, 0x88, 0xa1, 0x01, 
0x85, 0x08, 0x19, 0x01, 0x29, 0xff, 0x15, 0x01, 0x26, 0xff, 
0x00, 0x75, 0x08, 0x95, 0x01, 0x81, 0x00, 0xc0, 
Device Found
  type: 046d c52b
  path: DevSrvsID:4294970451
  serial_number: 
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    0
  Usage (page): 0x1 (0x1)
  Bus type: 1

  Report Descriptor: (148 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0x05, 
0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x03, 0x75, 0x10, 0x95, 
0x02, 0x15, 0x01, 0x26, 0xff, 0x02, 0x19, 0x01, 0x2a, 0xff, 
0x02, 0x81, 0x00, 0xc0, 0x05, 0x01, 0x09, 0x80, 0xa1, 0x01, 
0x85, 0x04, 0x75, 0x02, 0x95, 0x01, 0x15, 0x01, 0x25, 0x03, 
0x09, 0x82, 0x09, 0x81, 0x09, 0x83, 0x81, 0x60, 0x75, 0x06, 
0x81, 0x03, 0xc0, 0x06, 0xbc, 0xff, 0x09, 0x88, 0xa1, 0x01, 
0x85, 0x08, 0x19, 0x01, 0x29, 0xff, 0x15, 0x01, 0x26, 0xff, 
0x00, 0x75, 0x08, 0x95, 0x01, 0x81, 0x00, 0xc0, 
Device Found
  type: 046d c52b
  path: DevSrvsID:4294970451
[... 72 lines cut ...]
0x02, 0x81, 0x00, 0xc0, 0x05, 0x01, 0x09, 0x80, 0xa1, 0x01, 
0x85, 0x04, 0x75, 0x02, 0x95, 0x01, 0x15, 0x01, 0x25, 0x03, 
0x09, 0x82, 0x09, 0x81, 0x09, 0x83, 0x81, 0x60, 0x75, 0x06, 
0x81, 0x03, 0xc0, 0x06, 0xbc, 0xff, 0x09, 0x88, 0xa1, 0x01, 
0x85, 0x08, 0x19, 0x01, 0x29, 0xff, 0x15, 0x01, 0x26, 0xff, 
0x00, 0x75, 0x08, 0x95, 0x01, 0x81, 0x00, 0xc0, 
Device Found
  type: 046d c52b
  path: DevSrvsID:4294970444
  serial_number: 
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    0
  Usage (page): 0x6 (0x1)
  Bus type: 1

  Report Descriptor: (59 bytes)
0x05, 0x01, 0x09, 0x06, 0xa1, 0x01, 0x05, 0x07, 0x19, 0xe0, 
0x29, 0xe7, 0x15, 0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x08, 
0x81, 0x02, 0x81, 0x03, 0x95, 0x05, 0x05, 0x08, 0x19, 0x01, 
0x29, 0x05, 0x91, 0x02, 0x95, 0x01, 0x75, 0x03, 0x91, 0x01, 
0x95, 0x06, 0x75, 0x08, 0x15, 0x00, 0x26, 0xa4, 0x00, 0x05, 
0x07, 0x19, 0x00, 0x2a, 0xa4, 0x00, 0x81, 0x00, 0xc0, 
```

mcuee, 2023-04-15:
From @dingsijie-nreal

> Well, I noticed this PR #530 It was submitted by one of my coworkers. The PR is a workaround we found. For now I think
> 
> https://github.com/libusb/hidapi/blob/9f185ec51025695ab004085600ddaafb252e1e42/mac/hid.c#L550
> 
> is the
> only "proper" way to get the interface number. But it just doesn't work on the latest macOS.
> We've posted a question at https://developer.apple.com/forums/thread/728001.
> No one replys... Maybe you can also have a look at the post, and say something there.

mcuee, 2023-04-15:
@Youw @dingsijie-nreal @pengxianheng-nreal

Somehow libusb is able to print the interface number.

```
mcuee@mcuees-Mac-mini libusb % ./examples/xusb 046d:c52b
Using libusb v1.0.26.11791

Opening device 046D:C52B...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 046D:C52B
         bcdDevice: 2410
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 84
         descriptor length: 9
             nb interfaces: 3
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.01
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 08
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 82
           max packet size: 0008
          polling interval: 02
              interface[2]: id = 2
interface[2].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 83
           max packet size: 0020
          polling interval: 02

Kernel driver attached for interface 0: yes

Claiming interface 0...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
libusb: warning [darwin_detach_kernel_driver] USB device capture requires either an entitlement (com.apple.vm.device-access) or root privilege
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-3
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed (error 1) Other error

Kernel driver attached for interface 1: yes

Claiming interface 1...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
libusb: warning [darwin_detach_kernel_driver] USB device capture requires either an entitlement (com.apple.vm.device-access) or root privilege
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-3
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed (error 1) Other error

Kernel driver attached for interface 2: yes

Claiming interface 2...
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
libusb: warning [darwin_detach_kernel_driver] USB device capture requires either an entitlement (com.apple.vm.device-access) or root privilege
libusb: info [darwin_capture_claim_interface] failed to auto-detach the kernel driver for this device, ret=-3
libusb: info [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
   Failed (error 1) Other error
[... 115 lines cut ...]
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      98
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               2
can't get device qualifier: No such file or directory
can't get debug descriptor: No such file or directory
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

```

mcuee, 2023-04-15:
PR #530 is a workaround and it seems to work. However, I believe there is a better solution.
* #530 

```
mcuee@mcuees-Mac-mini hidapi_pr530 % ./hidtest/hidtest 
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 046d 1000
  path: DevSrvsID:4294971188
  serial_number: 
  Manufacturer: Logitech
  Product:      LogiVirtualHIDDevice
  Release:      1
  Interface:    2
  Usage (page): 0x2 (0x1)
  Bus type: 0

  Report Descriptor: (69 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 
Device Found
  type: 046d 1000
  path: DevSrvsID:4294971188
  serial_number: 
  Manufacturer: Logitech
  Product:      LogiVirtualHIDDevice
  Release:      1
  Interface:    2
  Usage (page): 0x1 (0x1)
  Bus type: 0

  Report Descriptor: (69 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 
Device Found
  type: 046d c52b
  path: DevSrvsID:4294971164
  serial_number: 
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    2
  Usage (page): 0x1 (0xff00)
  Bus type: 1

  Report Descriptor: (98 bytes)
0x06, 0x00, 0xff, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x10, 0x75, 
0x08, 0x95, 0x06, 0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 0x01, 
0x81, 0x00, 0x09, 0x01, 0x91, 0x00, 0xc0, 0x06, 0x00, 0xff, 
0x09, 0x02, 0xa1, 0x01, 0x85, 0x11, 0x75, 0x08, 0x95, 0x13, 
0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 0x02, 0x81, 0x00, 0x09, 
0x02, 0x91, 0x00, 0xc0, 0x06, 0x00, 0xff, 0x09, 0x04, 0xa1, 
0x01, 0x85, 0x20, 0x75, 0x08, 0x95, 0x0e, 0x15, 0x00, 0x26, 
0xff, 0x00, 0x09, 0x41, 0x81, 0x00, 0x09, 0x41, 0x91, 0x00, 
0x85, 0x21, 0x95, 0x1f, 0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 
0x42, 0x81, 0x00, 0x09, 0x42, 0x91, 0x00, 0xc0, 
Device Found
  type: 046d c52b
[... 171 lines cut ...]
  serial_number: 
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    1
  Usage (page): 0x88 (0xffbc)
  Bus type: 1

  Report Descriptor: (148 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0x05, 
0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x03, 0x75, 0x10, 0x95, 
0x02, 0x15, 0x01, 0x26, 0xff, 0x02, 0x19, 0x01, 0x2a, 0xff, 
0x02, 0x81, 0x00, 0xc0, 0x05, 0x01, 0x09, 0x80, 0xa1, 0x01, 
0x85, 0x04, 0x75, 0x02, 0x95, 0x01, 0x15, 0x01, 0x25, 0x03, 
0x09, 0x82, 0x09, 0x81, 0x09, 0x83, 0x81, 0x60, 0x75, 0x06, 
0x81, 0x03, 0xc0, 0x06, 0xbc, 0xff, 0x09, 0x88, 0xa1, 0x01, 
0x85, 0x08, 0x19, 0x01, 0x29, 0xff, 0x15, 0x01, 0x26, 0xff, 
0x00, 0x75, 0x08, 0x95, 0x01, 0x81, 0x00, 0xc0, 
```

mcuee, 2023-04-15:
Just a quick test using my mod hidapi-libusb under macOS.

```
mcuee@mcuees-Mac-mini libusb % ./hidtest-libusb 
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 046d c52b
  path: 2-2.4.4:1.0
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    0
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: Unable to open device by path
Device Found
  type: 046d c52b
  path: 2-2.4.4:1.1
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    1
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: Unable to open device by path
Device Found
  type: 046d c52b
  path: 2-2.4.4:1.2
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    2
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: Unable to open device by path
unable to open device

mcuee@mcuees-Mac-mini libusb % sudo ./hidtest-libusb
Password:
hidapi test/example tool. Compiled with hidapi version 0.14.0, runtime version 0.14.0.
Compile-time version matches runtime version of hidapi.

Device Found
  type: 046d c52b
  path: 2-2.4.4:1.0
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    0
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: (59 bytes)
0x05, 0x01, 0x09, 0x06, 0xa1, 0x01, 0x05, 0x07, 0x19, 0xe0, 
0x29, 0xe7, 0x15, 0x00, 0x25, 0x01, 0x75, 0x01, 0x95, 0x08, 
0x81, 0x02, 0x81, 0x03, 0x95, 0x05, 0x05, 0x08, 0x19, 0x01, 
0x29, 0x05, 0x91, 0x02, 0x95, 0x01, 0x75, 0x03, 0x91, 0x01, 
0x95, 0x06, 0x75, 0x08, 0x15, 0x00, 0x26, 0xa4, 0x00, 0x05, 
0x07, 0x19, 0x00, 0x2a, 0xa4, 0x00, 0x81, 0x00, 0xc0, 
Device Found
  type: 046d c52b
  path: 2-2.4.4:1.1
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    1
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: (148 bytes)
0x05, 0x01, 0x09, 0x02, 0xa1, 0x01, 0x85, 0x02, 0x09, 0x01, 
0xa1, 0x00, 0x05, 0x09, 0x19, 0x01, 0x29, 0x10, 0x15, 0x00, 
0x25, 0x01, 0x95, 0x10, 0x75, 0x01, 0x81, 0x02, 0x05, 0x01, 
0x16, 0x01, 0xf8, 0x26, 0xff, 0x07, 0x75, 0x0c, 0x95, 0x02, 
0x09, 0x30, 0x09, 0x31, 0x81, 0x06, 0x15, 0x81, 0x25, 0x7f, 
0x75, 0x08, 0x95, 0x01, 0x09, 0x38, 0x81, 0x06, 0x05, 0x0c, 
0x0a, 0x38, 0x02, 0x95, 0x01, 0x81, 0x06, 0xc0, 0xc0, 0x05, 
0x0c, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x03, 0x75, 0x10, 0x95, 
0x02, 0x15, 0x01, 0x26, 0xff, 0x02, 0x19, 0x01, 0x2a, 0xff, 
0x02, 0x81, 0x00, 0xc0, 0x05, 0x01, 0x09, 0x80, 0xa1, 0x01, 
0x85, 0x04, 0x75, 0x02, 0x95, 0x01, 0x15, 0x01, 0x25, 0x03, 
0x09, 0x82, 0x09, 0x81, 0x09, 0x83, 0x81, 0x60, 0x75, 0x06, 
0x81, 0x03, 0xc0, 0x06, 0xbc, 0xff, 0x09, 0x88, 0xa1, 0x01, 
0x85, 0x08, 0x19, 0x01, 0x29, 0xff, 0x15, 0x01, 0x26, 0xff, 
0x00, 0x75, 0x08, 0x95, 0x01, 0x81, 0x00, 0xc0, 
Device Found
  type: 046d c52b
  path: 2-2.4.4:1.2
  serial_number: (null)
  Manufacturer: Logitech
  Product:      USB Receiver
  Release:      2410
  Interface:    2
  Usage (page): 0x0 (0x0)
  Bus type: 1

  Report Descriptor: (98 bytes)
0x06, 0x00, 0xff, 0x09, 0x01, 0xa1, 0x01, 0x85, 0x10, 0x75, 
0x08, 0x95, 0x06, 0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 0x01, 
0x81, 0x00, 0x09, 0x01, 0x91, 0x00, 0xc0, 0x06, 0x00, 0xff, 
0x09, 0x02, 0xa1, 0x01, 0x85, 0x11, 0x75, 0x08, 0x95, 0x13, 
0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 0x02, 0x81, 0x00, 0x09, 
0x02, 0x91, 0x00, 0xc0, 0x06, 0x00, 0xff, 0x09, 0x04, 0xa1, 
0x01, 0x85, 0x20, 0x75, 0x08, 0x95, 0x0e, 0x15, 0x00, 0x26, 
0xff, 0x00, 0x09, 0x41, 0x81, 0x00, 0x09, 0x41, 0x91, 0x00, 
0x85, 0x21, 0x95, 0x1f, 0x15, 0x00, 0x26, 0xff, 0x00, 0x09, 
0x42, 0x81, 0x00, 0x09, 0x42, 0x91, 0x00, 0xc0, 
unable to open device
```

mcuee, 2023-04-15:
@Youw 

I am thinking that libusb codes may be of some reference to sort out the issue here.[
https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L1342-L1377

Youw, 2023-04-15:
> Somehow libusb is able to print the interface number.

Yeah, libusb and HIDAPI using different API/subsystems to get the information about USB device.
libusb gets it directly from USB subsystem, which would be broken all at once if the interface number wasn't availale.
HIDAPI uses HidManager macOS API which is either changed or broken in the latest macOS 13.3.1.

dingsijie-nreal, 2023-04-16:
Yeah, I've noticed that libusb is able to fetch the correct interface number, but as @Youw has pointed out, hidapi doesn't use libusb on mac. Our app code currently uses hidapi API because it is easier and nicer to use than libusb. Maybe it is a real bug on macOS 13.3?
Now what I'm thinking more about is that shouldn't apple already received reports about this issue? Is @mcuee 's logitech USB receiver working fine on macOS 13.3? Or, maybe a usb device with none zero interface number itself is a less common design?

mcuee, 2023-04-16:
The USB receiver works fine (together with the mouse and keyboard). In fact yesterday I tried hidpp macOS fork and it works fine.
* https://github.com/cvuchener/hidpp/issues/17#issuecomment-1509613098

It is probably just the paticual macOS API is broken. Sometimes Apple will fix it in the future updates. Sometimes it may take time.
Ref: https://developer.apple.com/forums/thread/728001

mcuee, 2023-04-16:
> Better way is to submit a bug report to Apple.

@dingsijie-nreal
It will be good to file a feedback to Apple. 
https://developer.apple.com/bug-reporting/

You can see from the following libusb issue that Apple fixed the issue pretty fast in the paticular case. But sometimes it takes time.
Ref: https://github.com/libusb/libusb/issues/1156

dingsijie-nreal, 2023-04-17:
feedback posted. https://feedbackassistant.apple.com/feedback/12125162

Youw, 2023-04-17:
Is there some kind of visibility configuration for the feedback? I'm unable to open/see it.

mcuee, 2023-04-17:
> Is there some kind of visibility configuration for the feedback? I'm unable to open/see it.

Same here.

dingsijie-nreal, 2023-04-17:
> 
Only apple can see it according to:
https://developer.apple.com/forums/thread/700518
Also, I didn't notice any visibility setting when I was posting the feedback.

Youw, 2023-04-30:
After digging deeper into the problem in the context of #534 - I'm starting to think that this is not a bug in macOS.
It looks like HIDAPI used an undocumented property of the `IOHIDDevice` registry entry, and Apple finally cleaned (some?/all?) undocumented properties from the IOHIDDevice registry.

But it is still a clear statement that `IOHIDDevice` entry in the registry is one of the child entries of the `IOUSBHostInterface` entry for USB HID devices, which definitely has the InterfaceNumber property (otherwise at least libusb would break completely).

Youw, 2023-04-30:
@dingsijie-nreal see if you (or you colleague?) could check the #534 fix.

mcuee, 2023-05-01:
PR #534 works fine for me.

PR #534 (2023-05-05): macos: improve USB interface number retrival
Fixes: #531
closed 2023-05-05 by Youw