libusb/hidapi #532: The write method is called continuously in the thread

Alvazz, 2023-04-17:
I run the write method in a thread, which means that I keep sending data to the hid hardware device, can I do this?

Youw, 2023-04-17:
Generally, sending data continuesly is definitely something HIDAPI is designed for.
But if you send data in one thread and doing something else in another w/o syncronisation - that is not thread-safe.

As of current moment, all HIDAPI function over a particular hid_device must be done either in one thread, or serialized using some syncronisation mechanism.
The only exception is `hid_read(_timeout)` - that one might be used in another thread w/o syncronisation.

Alvazz, 2023-04-19:
[image]
I took a look at the Hidapi.c file and found that the place that suggested this error is here.

Youw, 2023-04-19:
Your `writeOtherData` and `writeMusicData` are not syncronized - that is a race condition and not a supported scenario.

Alvazz, 2023-04-19:
Thank you for your enthusiastic reply, so how should I operate? For example can I use a lock-free queue to achieve this? My idea is, for example, if I have 10 pieces of data that need to be sent to the target device continuously, can I create 10 lock-free queues, insert each piece of corresponding data into each lock-free queue, and then Sent by 10 threads separately, can I do this? Or, hid_write can only be used in one thread, and hid_write cannot be called in multiple threads at the same time?

closed 2023-04-20 by Youw