libusb/hidapi #540: hidapi 0.14.0 release

mcuee, 2023-05-07:
Now we have zero outstanding PRs. Just wondering if it is good to have 0.14.0 release soon.

If I look at the Project, it is for hotplug support, which will take sometime.
https://github.com/libusb/hidapi/projects/7

mcuee, 2023-05-10:
@Youw 

One of the good argument for a new release is because of #531. PR #534 has been merged to fix the issue for macOS 13.3 users.

mcuee, 2023-05-10:
Reference:
* https://github.com/ruabmbua/hidapi-rs/issues/115

mcuee, 2023-05-10:
Reference:
* https://github.com/ruabmbua/hidapi-rs/issues/104

Youw, 2023-05-12:
I think it makes sense to fix issues found by Coverity before the release.
#541, #542 
I want to run Coverity on all platforms and apply fixes if any.
Then we're good to release.

Youw, 2023-05-22:
https://github.com/libusb/hidapi/releases/tag/hidapi-0.14.0

closed 2023-05-22 by Youw