libusb/hidapi #567: connection-callback  branch cpu usage is very high

Alvazz, 2023-05-25:
pls check it!

```
if (hid_init())
		return;

	hid_hotplug_register_callback(0x0213, 0xA22C,
				      HID_API_HOTPLUG_EVENT_DEVICE_ARRIVED |
					      HID_API_HOTPLUG_EVENT_DEVICE_LEFT,
				      HID_API_HOTPLUG_ENUMERATE,
				      device_callback, NULL, &token2);
```

Alvazz, 2023-05-25:
Let me describe it more clearly, when I don't plug in the device (0x0213, 0xA22C), the CPU usage is not high, but when I plug in the device, the CPU instantly rises to 9%~11%

mcuee, 2023-05-27:
Please provide more details about the device (eg: output from hidtest) and OS you are using.

d3xMachina, 2023-06-06:
On my end I only have 0.05% cpu usage when a device gets connected with the connection-callback branch and my application does a number of things at the same time. My config is Windows 11 and a 13600k cpu.

Make sure you keep your code in the callback as concise as possible. You can use a producer-consumer pattern with a FIFO to process those connections/disconnections in another thread.

mcuee, 2023-08-14:
@Alvazz 

Any updates? Thanks.

We will close the issue if there are no responses.

closed 2024-11-20 by Youw