libusb/hidapi #581: Building libusb support on Windows

slouken, 2023-05-26:
SDL has a change to allow building hidapi with libusb support on Windows. It was done by abstracting the threading interface, so it can be replaced by other implementations, and the pthread implementation is provided by default.

Please let me know if you'd like this change as a PR!
https://github.com/libsdl-org/SDL/commit/57b33aec01a86c28b30797ecbae394a2f07ddd42

mcuee, 2023-05-27:
Yes please file a PR.

This will also address the following issue.
* https://github.com/libusb/hidapi/issues/319

mcuee, 2023-05-27:
@slouken 

If possible, you may also want to add macOS support. macOS libusb support is not so useful for real application (requires sudo or some difficult-to-attain entitlement), but it may be useful to troubleshooting purpose and maybe useful for some devices.

slouken, 2023-05-27:
The current code works on macOS without any changes. I tested with it a few days ago.

closed 2023-06-06 by Youw