libusb/hidapi #589: hidapi Windows reports one byter longer of HID Output Report written than Linux and macOS

mcuee, 2023-06-16:
Discusison here:
https://github.com/libusb/hidapi/issues/478#issuecomment-1592887588

Testing device: Circuit Python rawhid example, no report IDs. 
* https://github.com/adafruit/Adafruit_CircuitPython_HID/issues/115#issuecomment-1587535723

boot.py
```
import usb_hid

RAWHID_REPORT_DESCRIPTOR = bytes((
    0x06, 0x00, 0xFF,  # Usage Page (Vendor Defined 0xFF00)
    0x09, 0x01,        # Usage (0x01)
    0xA1, 0x01,        # Collection (Application)
    0x09, 0x02,        #   Usage (0x02)
    0x15, 0x00,        #   Logical Minimum (0)
    0x26, 0xFF, 0x00,  #   Logical Maximum (255)
    0x75, 0x08,        #   Report Size (8)
    0x95, 0x40,        #   Report Count (64)
    0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x03,        #   Usage (0x03)
    0x15, 0x00,        #   Logical Minimum (0)
    0x26, 0xFF, 0x00,  #   Logical Maximum (255)
    0x75, 0x08,        #   Report Size (8)
    0x95, 0x40,        #   Report Count (64)
    0x91, 0x02,        #   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0xC0,              # End Collection
))

raw_hid = usb_hid.Device(
    report_descriptor=RAWHID_REPORT_DESCRIPTOR,
    usage_page=0xFF00,
    usage=0x01,
    report_ids=(0,),
    in_report_lengths=(64,),
    out_report_lengths=(64,),
)

usb_hid.enable((raw_hid,))
```

code.py
```
import usb_hid
import time

d = usb_hid.devices[0]

while True:
    report = bytearray(64)  # must be same size as specified in HID Report Descriptor in boot.py    

    report[0] = 1
    report[1] = 2
    report[2] = 3
    report[3] = 4
    report[63] = 64
    d.send_report(report)
    time.sleep(1)
    print(d.get_last_received_report())
```

hidapitester output under Windows, it says 65 bytes written
```
PS C:\work\hid\hidapitester> .\hidapitester --vidpid 2e8a:102e --open --length 64 --send-output 1,2,3,4,5,6,7,8
Opening device, vid/pid: 0x2E8A/0x102E
Writing output report of 64-bytes...wrote 65 bytes:
 01 02 03 04 05 06 07 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

hidapitester output under Linux: it says 64 bytes written
```
mcuee@UbuntuSwift3:~/build/hid/hidapitester$ sudo ./hidapitester --vidpid 2e8a:102e --open --length 64 --send-output 1,2,3,4,5,6,7,8
Opening device, vid/pid: 0x2E8A/0x102E
Writing output report of 64-bytes...wrote 64 bytes:
 01 02 03 04 05 06 07 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Closing device
```

hidtest info about the device (using Linux to get the real HID report descriptors)
```
Device Found
  type: 2e8a 102e
  path: /dev/hidraw3
  serial_number: DE6185100F4D6522
  Manufacturer: VCC-GND Studio
  Product:      YD-RP2040
  Release:      100
  Interface:    3
  Usage (page): 0x1 (0xff00)
  Bus type: 1 (USB)

  Report Descriptor: (34 bytes)
0x06, 0x00, 0xff, 0x09, 0x01, 0xa1, 0x01, 0x09, 0x02, 0x15, 
0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 0x40, 0x81, 0x02, 
0x09, 0x03, 0x15, 0x00, 0x26, 0xff, 0x00, 0x75, 0x08, 0x95, 
0x40, 0x91, 0x02, 0xc0,
```

mcuee, 2023-06-16:
Relevant discussion before, which fixed the reported Input Report number of reading bytes
* https://github.com/libusb/hidapi/pull/232

Probably similar patch is requrired for Output report.

The Windows behavior is not wrong per se, Windows HID driver indeed adds the report ID 0 as the first byte in the buffer. However the report ID 0 is not really send on the USB bus. So we should not report 65 bytes in the above example, but only 64 bytes.

Since HIDAPI is a cross-platform library, I think we should make Windows to behave the same as Linux and macOS.

mcuee, 2023-06-16:
Test using Jan Axelson's generic HID example here.
http://janaxelson.com/hidpage.htm

Windows reported 3 for Feature report writing and reading, one byter longer than the real device (2 bytes Input report, Output Report and Feature report, no report IDs). And same for Output report, Windows reports 3 bytes written.

Only for Input report, Windows is correct to report 2 bytes have been read, thanks to PR #232.

```
ROM struct{BYTE report[HID_RPT01_SIZE];}hid_rpt01={
 {
   0x06, 0x00, 0xFF,               // Usage page (Vendor Defined Page1)
   0x09, 0x01,                     // Usage (Vendor Usage 1)
   0xA1, 0x01,                     // Collection (Application)

   // Global items apply to all reports below

   0x15, 0x00,                     // Logical Minimum (0)
   0x26, 0xFF, 0x00,               // Logical Maximum (255)
   0x75, 0x08,                     // Report Size (8 bits)
   0x95, 0x02,                     // Report Count (2 fields)

   // Input report
   0x85, 0x01,                     // Report ID (1)
   0x09, 0x00,                     // Usage (Undefined)
   0x82, 0x02, 0x01,               // Input (Data,Var,Abs,Buf)

   // Output report
   0x85, 0x02,                     // Report ID (2)
   0x09, 0x00,                     // Usage (Undefined)
   0x92, 0x02, 0x01,               // Output (Data,Var,Abs,Buf)

   // Feature report
   0x85, 0x03,                     // Report ID (3)
   0x09, 0x00,                     // Usage (Undefined)
   0xB2, 0x02, 0x01,               // Feature (Data,Var,Abs,Buf)

   // Feature report
   0x85, 0x04,                     // Report ID (4)
   0x09, 0x00,                     // Usage (Undefined)
   0xB2, 0x02, 0x01,               // Feature (Data,Var,Abs,Buf)

   0xC0                            // end collection
 }

PS C:\work\hid\hidapitester> .\hidapitester --vidpid 0925:7001 --open -l 3 --send-feature 0,1,2 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 01 02
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 01 02
Closing device

PS C:\work\hid\hidapitester> .\hidapitester --vidpid 0925:7001 --open -l 3 --send-output 0,1,2 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 01 02
Reading 3-byte input report 0, 250 msec timeout...read 2 bytes:
 01 02 00
Closing device
```

libusb-1.0.26 version works with the device using my test code.
https://github.com/mcuee/picusb/tree/master/lvr_genhid
```
$ ./libusb1_lvrhid4.exe
Successfully find the LVR Generic HID device
Successfully set usb configuration 1
Successfully claimed interface
Testing control transfer using loop back test of feature report
20, 20; 21, 21;
Testing interrupt transfer using loop back test of input/output report
40, 40; 41, 41;
```

mcuee, 2023-06-16:
So I am pretty sure we need to fix number of bytes written Output Report under Windows. 

I am not so sure about Feature report yet.

mcuee, 2023-06-16:
@JoergAtGithub

I think you have very good understanding of the issue and the necessary test devices. Please take a look at this issue as well when you got the chance.

mcuee, 2023-06-16:
@Youw 

I think for Feature report, Linux hidraw implemenation is kind of the same as Windows. So that probably explains why I see the same behavior under Windows and Linux, since we just report the buffer length (without minus one for the case when there is no repor ID) and not the data on the USB bus.

Now the question is whether we need to minus 1 for the case of no Report ID, to reflect the data on the USB Bus, and not the OS driver.

https://www.kernel.org/doc/Documentation/hid/hidraw.txt

>HIDIOCSFEATURE(len): Send a Feature Report
This ioctl will send a feature report to the device.  Per the HID
specification, feature reports are always sent using the control endpoint.
Set the first byte of the supplied buffer to the report number.  For devices
which do not use numbered reports, set the first byte to 0. The report data
begins in the second byte. Make sure to set len accordingly, to one more
than the length of the report (to account for the report number).
>
> HIDIOCGFEATURE(len): Get a Feature Report
This ioctl will request a feature report from the device using the control
endpoint.  The first byte of the supplied buffer should be set to the report
number of the requested report.  For devices which do not use numbered
reports, set the first byte to 0.  The report will be returned starting at
the first byte of the buffer (ie: the report number is not returned).

mcuee, 2023-06-16:
As of now, hidapitester using hidapi-libusb (custom build) behaves pretty same as hidapi-hidraw. So that is a good thing.
```
mcuee@UbuntuSwift3:~/build/hid/hidapitester$ sudo ./hidapitester_libusb  --vidpid 0925:7001 --open -l 3 --send-output 0,1,2 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 01 02
Reading 3-byte input report 0, 250 msec timeout...read 2 bytes:
 01 02 00
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester$ sudo ./hidapitester_libusb  --vidpid 0925:7001 --open -l 2 --send-output 1,2 --read-input
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 2-bytes...wrote 2 bytes:
 01 02
Reading 2-byte input report 0, 250 msec timeout...read 2 bytes:
 01 02
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester$ sudo ./hidapitester  --vidpid 0925:7001 --open -l 3 --send-feature 0,1,2 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 01 02
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 01 02
Closing device
```

Youw, 2023-06-16:
> 			@returns
>				This function returns the number of bytes read plus
>				one for the report ID (which is still in the first
>				byte), or -1 on error.
>				Call hid_error(dev) to get the failure reason.
>		*/
>		int HID_API_EXPORT HID_API_CALL hid_get_feature_report(hid_device *dev, unsigned char *data, size_t length);

As per HIDAPI documentation, we should include the `0` report ID in the return. Is it included?

JoergAtGithub, 2023-06-16:
`0` is a placeholder an not a Report ID. This is because the HID class specification specifies _"If a Report ID tag was used in the Report descriptor, all reports include a single byte ID prefix. If the Report ID tag was not used, all values are returned in a single report and a prefix ID is not included in that report."_ and further about the representation in the ReportDescriptor  _"Set Report ID to 0 (zero) if Report IDs are not used."_

Therefore only values from 1 to 255 can be Report IDs. The value 0 can be used in the ReportDescriptor, to define that the reports do not contain a ReportID.

Windows implements it as specified in the class specification and adds the Report ID prefix byte only for devices that use Report IDs.

I know that we need to keep backward compatibility, but from a clean API I would expect a clear seperation between address and data in two arguments.

mcuee, 2023-06-16:
> I know that we need to keep backward compatibility, but from a clean API I would expect a clear seperation between address and data in two arguments.

@JoergAtGithub 

Just wondering what you mean by the above. Are you proposing some new APIs? I think that will be okay if there are real benefits. We can always keep the older APIs (and in the future to deprecate them in HIDAPI 1.x version if necessary).

mcuee, 2023-06-17:
Just tested under macOS (13.4, Mac Mini M1) and the behavior is the same as Linux. So only Windows is out for the Output Report.

```
mcuee@mcuees-Mac-mini hidapitester % ./hidapitester  --vidpid 0925:7001 --open -l 3 --send-output 0,1,2 --read-input       
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 01 02
Reading 3-byte input report 0, 250 msec timeout...read 2 bytes:
 01 02 00
Closing device

mcuee@mcuees-Mac-mini hidapitester % ./hidapitester  --vidpid 0925:7001 --open -l 2 --send-output 1,2 --read-input  
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 2-bytes...wrote 2 bytes:
 01 02
Reading 2-byte input report 0, 250 msec timeout...read 2 bytes:
 01 02
Closing device

mcuee@mcuees-Mac-mini hidapitester % ./hidapitester --vidpid 0925:7001 --open -l 3 --send-feature 0,1,2 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 01 02
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 01 02
Closing device
```

Youw, 2023-06-17:
> Just wondering what you mean by the above.

I believe that is a though process "out loud", i.e. "it is probably would be clean to have an API where we send the report ID as a separate argument and only the report body in the buffer".

I think that will not change anything in terms of this issue. We still have to fight with different drivers/OS implementation to match the behavior the way we want. And because as per current documentation - we should always count the report ID as part of the data returned - having it as a separate argument would essentially mean `-1` to the result.
But it will complicate some backend implementations, e.g. hidraw, which expects a single buffer as well. And if report ID will be a separate argument - we would have to make an intermediate buffer/copy before we can pass it into the underlying API.

JoergAtGithub, 2023-06-17:
But pre-pending the `0` byte is also an operation, which requires intermediate buffer/copy.

Youw, 2023-06-17:
That's exactly my point.
Having report ID at the beginning of the buffer - means not extra buffer "prepending" on HIDAPI side.

JoergAtGithub, 2023-06-17:
If you want to unify the behaviour between platforms you need either to prepend the `0` byte on Windows send functions for devices that don't use a Report ID or you need to remove the `0` byte for the other backends.
Because we have send and receive functions, there will be always a case, where you have to prepend the `0`.
Performance wise I would say receiving reports is a more critical operation than sending.

mcuee, 2023-06-17:
> That's exactly my point. Having report ID at the beginning of the buffer - means not extra buffer "prepending" on HIDAPI side.

There are two aspects of the issue.

1. Buffer handling -- platform specific. I think the current codes already do that. For example, Windows codes have already done that, at least for Output Report and Feature Report (first byte of the report is the report ID, 0 when there is no report ID). And I believe all platforms are doing that for Feature report, based on my testing results.

I tend to believe the current codes are already fine in this aspect, i.e, we do not have issues in terms of getting thngs done (the device will receive the right data, host will also receive the right data).

2. Read/Write lentgh reporting

I think this is the issue now. It is not consistent.

Two ways of reporting -- now we are mixing the two.
a) lengh of the buffer used by underlying OS driver -- Feature Report (including the 0 byte for device without Report ID). Currently I believe there is no behaviour differences for Feature Report across platforms.

b) length of the data bytes appearing on the USB Bus -- Input Report  (not including the 0 byte for device without Report ID). For Windows, this could be due to the way we handle Input Report differently from Output Report and Feature report. Currently I believe there is no behaviour differences for Input Report across platforms.

c) mix of the two ways -- Output Report (Windows always including the 0 byte for device without Report ID, other platforms may or may not including). So Windows may report one byte more than the other platforms or may report the same length as the other platforms. I think we should at least fix this one.

3. Potential Solution for Issue 2.
a) If we really want to unify all platforms in terms of the read/write length report mechanism, then I tend to htink Option b is the way to go for all three types of reports. But then it may require some jobs across all platforms and for Output and Feature reports..

b) If we just want to unify the behavior across platforms, then we only need to fix Output Report write length reporting to the user. I tend to think we only need to make Windows consistent with others. So only one platform (Windows) and one report type (Output Report) needs to be fixed.

Youw, 2023-06-17:
> If we just want to unify the behavior across platforms, then we only need to fix Output Report write length reporting to the user. I tend to think we only need to make Windows consistent with others. So only one platform (Windows) and one report type (Output Report) needs to be fixed.

+1

mcuee, 2023-06-17:
> > If we just want to unify the behavior across platforms, then we only need to fix Output Report write length reporting to the user. I tend to think we only need to make Windows consistent with others. So only one platform (Windows) and one report type (Output Report) needs to be fixed.
> 
> +1

Okay, I agree this is the way to go to keep backwards compatibillity and with minimum changes.

We may also need to improve the documentation as well.

mcuee, 2023-06-18:
I will try to compare the Linux hidraw implementation versus Windows implementation.

One thing hidapitester does not test is to use `hid_get_input_report` to get input report using Control Transfer. It only uses `hid_read_timeout`. So this may hide the issues with regard to Input Report.

1) Linux hidraw and Input Report

Linux hidraw docuemtation shows there will be no discrepancies using `read` method in `hid_read_timeout` and using `HIDIOCSINPUT` ioctl in `hid_get_input_report`.

`read` method: For devices which do not use numbered reports, the report data will begin at the first byte.

`HIDIOCGINPUT` ioctl: For devices which do not use numbered reports, the report data will begin at the first byte of the returned buffer.

2) Windows HID API and Input Report
Windows `hid_read_timeout` implementation uses ReadFile which should give the same behavior as above.

Windows `hid_get_input_report` uses IOCTL_HID_GET_INPUT_REPORT, which will add the report ID 0 for devices which do not use numbered reports (no Report ID)

**So we will have a discrepancy here within the Windows implementations betwen using `hid_read_timeout and `hid_read`.**

```
int HID_API_EXPORT HID_API_CALL hid_get_input_report(hid_device *dev, unsigned char *data, size_t length)
{
	/* We could use HidD_GetInputReport() instead, but it doesn't give us an actual length, unfortunately */
	return hid_get_report(dev, IOCTL_HID_GET_INPUT_REPORT, data, length);
}
```

By the way, Windows `hid_get_report` has the same behavior as the above `hid_get_input_report`. It is mainly used for Feature Report (which is consistent with other platforms) but it is not used by Windows `hid_get_input_report` implementations in the end. Actually we can use it for `hid_get_input_report` without the following codes which is correct for Feature report. Then it will be consistent with `hid_read`.
```
	/* When numbered reports aren't used,
	   bytes_returned seem to include only what is actually received from the device
	   (not including the first byte with 0, as an indication "no numbered reports"). */
	if (data[0] == 0x0) {
		bytes_returned++;
	}
```

**Reference for Windows**
https://learn.microsoft.com/en-us/windows-hardware/drivers/ddi/hidclass/ni-hidclass-ioctl_hid_get_input_report

**IOCTL_HID_GET_INPUT_REPORT IOCTL (hidclass.h)**

Input buffer
The Parameters.DeviceIoControl.OutputBufferLength member specifies the size of a requester-allocated output buffer in bytes. The HID class driver uses this buffer to return an input report.

If the collection includes report IDs, the requester must set the first byte of the output buffer to a nonzero report ID. Otherwise, the requester must set the first byte of the output buffer to zero.

Input buffer length
The size of the buffer in bytes. The buffer must be large enough to hold the input report plus one additional byte that specifies a nonzero report ID. If report ID is not used, the ID value is zero.

**Reference: https://docs.kernel.org/hid/hidraw.html**

**read()**
read() will read a queued report received from the HID device. On USB devices, the reports read using read() are the reports sent from the device on the INTERRUPT IN endpoint. By default, read() will block until there is a report available to be read. read() can be made non-blocking, by passing the O_NONBLOCK flag to open(), or by setting the O_NONBLOCK flag using fcntl().

On a device which uses numbered reports, the first byte of the returned data will be the report number; the report data follows, beginning in the second byte. For devices which do not use numbered reports, the report data will begin at the first byte.

**write()**
The write() function will write a report to the device. For USB devices, if the device has an INTERRUPT OUT endpoint, the report will be sent on that endpoint. If it does not, the report will be sent over the control endpoint, using a SET_REPORT transfer.

The first byte of the buffer passed to write() should be set to the report number. If the device does not use numbered reports, the first byte should be set to 0. The report data itself should begin at the second byte.

**HIDIOCSFEATURE(len):**
Send a Feature Report

This ioctl will send a feature report to the device. Per the HID specification, feature reports are always sent using the control endpoint. Set the first byte of the supplied buffer to the report number. For devices which do not use numbered reports, set the first byte to 0. The report data begins in the second byte. Make sure to set len accordingly, to one more than the length of the report (to account for the report number).

**HIDIOCGFEATURE(len):**
Get a Feature Report

This ioctl will request a feature report from the device using the control endpoint. The first byte of the supplied buffer should be set to the report number of the requested report. For devices which do not use numbered reports, set the first byte to 0. The returned report buffer will contain the report number in the first byte, followed by the report data read from the device. For devices which do not use numbered reports, the report data will begin at the first byte of the returned buffer.

**HIDIOCSINPUT(len):**
Send an Input Report

This ioctl will send an input report to the device, using the control endpoint. In most cases, setting an input HID report on a device is meaningless and has no effect, but some devices may choose to use this to set or reset an initial state of a report. The format of the buffer issued with this re [... cut ...] uffer issued with this report is identical to that of HIDIOCSFEATURE.

**HIDIOCGOUTPUT(len):**
Get an Output Report

This ioctl will request an output report from the device using the control endpoint. Typically, this is used to retrieve the initial state of an output report of a device, before an application updates it as necessary either via a HIDIOCSOUTPUT request, or the regular device write() interface. The format of the buffer issued with this report is identical to that of HIDIOCGFEATURE.

mcuee, 2023-06-18:
For Feature Report: (to be updated)

1) Linux hidraw for Feature Report

2) Windows:

```
int HID_API_EXPORT HID_API_CALL hid_get_feature_report(hid_device *dev, unsigned char *data, size_t length)
{
	/* We could use HidD_GetFeature() instead, but it doesn't give us an actual length, unfortunately */
	return hid_get_report(dev, IOCTL_HID_GET_FEATURE, data, length);
}
```

mcuee, 2023-06-18:
For Output Report: (to be updated)

Even though I mentioned that Windows may report one more byte than the other platforms, I tend to think it is actually according to the HIDAPI documentation.

Other platforms seems to support both -- to ignore the 0 or not to ignore the 0 report ID.

```
		/** @brief Write an Output report to a HID device.

			The first byte of @p data[] must contain the Report ID. For
			devices which only support a single report, this must be set
			to 0x0. The remaining bytes contain the report data. Since
			the Report ID is mandatory, calls to hid_write() will always
			contain one more byte than the report contains. For example,
			if a hid report is 16 bytes long, 17 bytes must be passed to
			hid_write(), the Report ID (or 0x0, for devices with a
			single report), followed by the report data (16 bytes). In
			this example, the length passed in would be 17.

			hid_write() will send the data on the first OUT endpoint, if
			one exists. If it does not, it will send the data through
			the Control Endpoint (Endpoint 0).

			@ingroup API
			@param dev A device handle returned from hid_open().
			@param data The data to send, including the report number as
				the first byte.
			@param length The length in bytes of the data to send.

			@returns
				This function returns the actual number of bytes written and
				-1 on error.
				Call hid_error(dev) to get the failure reason.
		*/
		int  HID_API_EXPORT HID_API_CALL hid_write(hid_device *dev, const unsigned char *data, size_t length);

```

mcuee, 2023-06-18:
Summary: to be updated.

It seems to me HIDAPI Windows implementation is actually consistent with the HIDAPI documentation, even though the behavior seems to be a bit different from the other platforms.

Now the question is shall we update the HIDAPI Windows implementation and documentation to make it consistent with other platforms?

mcuee, 2023-06-18:
> understand more about hidraw implementation to see if I can figure out the discrepancies between test results and hidraw documentation.

1) Existing discussions about Feature Report.

Confusions about `hid_get_feature_report`
https://github.com/libusb/hidapi/issues/229#issuecomment-810780640
https://github.com/liquidctl/liquidctl/issues/312

2) I am guessing that It could be that Linux hidraw `HIDIOCGFEATURE` documentation is wrong and it actually will add report ID 0 for devices without Report ID. If that is the case, then it can explain why I get the same results from hidraw and libusb under Linux.

mcuee, 2023-06-18:
> 1) To test `hid_get_input_report`
>
> I will try to use a modified version of hidapitester.
> * https://github.com/todbot/hidapitester/issues/4

Windows testing results with device with Report ID.

```
PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-output 2,0x41,0x42,0x43 --read-input 1
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 02 41 42
Reading 3-byte input report 1, 2000 msec timeout...read 3 bytes:
 01 41 42
Closing device

PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-output 2,0x41,0x42,0x43 --read-input-alt 1
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 02 41 42
Reading 3-byte input report, report_id 1...
read 3 bytes:
 01 41 42
Closing device

PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 4 --send-feature 3,0x31,0x32,0x33 --read-feature 4
Opening device, vid/pid: 0x0925/0x7001
Writing 4-byte feature report...wrote 4 bytes:
 03 31 32 33
Reading 4-byte feature report, report_id 4...read 3 bytes:
 04 31 32 00
Closing device
```

mcuee, 2023-06-18:
Linux test result

For devices without numbered report ID.

```
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 2 --timeout 2000 --send-output 0x41,0x42 --read-input 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 2-bytes...wrote 2 bytes:
 41 42
Reading 2-byte input report 0, 2000 msec timeout...read 2 bytes:
 41 42
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-output 0,0x41,0x42 --read-input 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 41 42
Reading 3-byte input report 0, 2000 msec timeout...read 2 bytes:
 41 42 00
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 2 --timeout 2000 --send-output 0x41,0x42 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 2-bytes...wrote 2 bytes:
 41 42
Reading 2-byte input report, report_id 0... 
read 2 bytes:
 00 00
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-output 0,0x41,0x42 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 41 42
Reading 3-byte input report, report_id 0... 
read 3 bytes:
 00 00 00
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-feature 0,0x31,0x32 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 31 32
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 31 32
Closing device
```

mcuee, 2023-06-18:
Linux libusb test results

For devices without numbered report ID.

Results are consistent with hidraw backend.

```
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-feature 0,0x31,0x32 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 31 32
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 31 32
Closing device
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 2 --timeout 2000 --send-feature 0x31,0x32 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 2-byte feature report...wrote -1 bytes:
 31 32
Reading 2-byte feature report, report_id 0...read 2 bytes:
 00 31
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 2 --timeout 2000 --send-output 0x41,0x42 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 2-bytes...wrote 2 bytes:
 41 42
Reading 2-byte input report, report_id 0... 
read 2 bytes:
 00 00 (data not correct)
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-output 0,0x41,0x42 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 3-bytes...wrote 3 bytes:
 00 41 42
Reading 3-byte input report, report_id 0... 
read 3 bytes:
 00 00 00 (data  not correct)
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 3 --timeout 2000 --send-feature 0,0x31,0x32 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 3-byte feature report...wrote 3 bytes:
 00 31 32
Reading 3-byte feature report, report_id 0...read 3 bytes:
 00 31 32
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 2 --timeout 2000 --send-feature 0x31,0x32 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 2-byte feature report...wrote -1 bytes:
 31 32
Reading 2-byte feature report, report_id 0...read 2 bytes:
 00 31
Closing device
```

mcuee, 2023-06-18:
@Youw and @todbot 

I need some help on using `hid_get_input_report` with `hidapitester`. I think I need to implement some types of time out. Currently it returns (I believe) the correct number of bytes reading but the data are not correct (all 0).

mcuee, 2023-06-18:
@Youw 

Somehow based on the test results, if I follow exactly the existing HIDAPI documentation, there are no issues, Windows implementations are consistent with Linux hidraw and Linux libusb implementations.

It is just that Linux hidraw and libusb implementations are more flexible in terms of output reports.

So in the end, maybe there is nothing to fix.

mcuee, 2023-06-18:
@todbot and @JoergAtGithub 

If possible, please test on your end to see if you come to the same conclusion as I or not. Thanks.

mcuee, 2023-06-18:
@jonasmalacofilho 

>Confusions about hid_get_feature_report
> https://github.com/libusb/hidapi/issues/229#issuecomment-810780640
> https://github.com/liquidctl/liquidctl/issues/312

You mentioned the following.
> However, the same text appears in `hid_get_feature_report` which, in turn, seems to have conflicting implementations:
>
> - the [libusb](https://github.com/libusb/hidapi/blob/6a01f3b4a8862b19a7ec768752ebcfc1a412f4b1/libusb/hid.c#L1230-L1258) implementation appears to follow the interpretation above;
> - but the [hidraw](https://github.com/libusb/hidapi/blob/6a01f3b4a8862b19a7ec768752ebcfc1a412f4b1/linux/hid.c#L1035-L1044) implementation simply wraps `HIDIOCGFEATURE`, which is documented to have a different behavior: "For devices which do not use numbered reports, the report data will begin at the first byte of the returned buffer."

If my conclusion is correct that Linux hidraw documentation about HIDIOCGFEATURE is incorrect, then your confusion will be cleared.

Now we need to confirm with Linux HID developer side.

Since you are also very familiar with hidapi and hidraw, please check on your side as well under Linux. Thanks.

mcuee, 2023-06-18:
> @Youw and @todbot
> 
> I need some help on using `hid_get_input_report` with `hidapitester`. I think I need to implement some types of time out. Currently it returns (I believe) the correct number of bytes reading but the data are not correct (all 0).

I think my modification is working even though it is a quick and dirty fix. It turns out I did not modify Jan Axeson's original HID FW correctly for devices without numbered report ID. I have fixed that for report length <=63 Bytes. I still have one issue with the FW for Report length = 64 Bytes but I will try to fix that later.

Here are the results under Windows and I can see that Windows implementation is as per the documentation, using the FW without numbered report ID and report length = 63 Bytes.

**We can see that `hid_read_timeout` gets 63 bytes and `hid_get_input_report` gets 64 bytes under Windows, which is as per the HIDAPI documentation. Whether that is a good thing or not is another story.**

```
ROM struct{BYTE report[HID_RPT01_SIZE];}hid_rpt01={
{
  	0x06, 0xA0, 0xFF,		// Usage page (vendor defined) 
  	0x09, 0x01,				// Usage ID (vendor defined)
  	0xA1, 0x01,				// Collection (application)

	// The Input report
       0x09, 0x03,     		// Usage ID - vendor defined
       0x15, 0x00,     		// Logical Minimum (0)
       0x26, 0xFF, 0x00,   	// Logical Maximum (255)
       0x75, 0x08,     		// Report Size (8 bits)
       0x95, 0x3f,     		// Report Count (63 fields)
       0x81, 0x02,     		// Input (Data, Variable, Absolute)  

	// The Output report
       0x09, 0x04,     		// Usage ID - vendor defined
       0x15, 0x00,     		// Logical Minimum (0)
       0x26, 0xFF, 0x00,   	// Logical Maximum (255)
       0x75, 0x08,     		// Report Size (8 bits)
       0x95, 0x3f,     		// Report Count (63 fields)
       0x91, 0x02,      	// Output (Data, Variable, Absolute)  

	// The Feature report
       0x09, 0x05,     		// Usage ID - vendor defined
       0x15, 0x00,     		// Logical Minimum (0)
       0x26, 0xFF, 0x00,   	// Logical Maximum (255)
       0x75, 0x08,			// Report Size (8 bits)
       0x95, 0x3f, 			// Report Count	(63 fields)				
       0xB1, 0x02,     		// Feature (Data, Variable, Absolute)  

  	0xC0}					// end collection
};

PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report 0, 4000 msec timeout...read 63 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 00
Closing device

PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report, report_id 0...
read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device

PS C:\work\hid\hidapitester_mod> .\hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-feature 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte feature report, report_id 0...read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E B1 0B 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
```

mcuee, 2023-06-18:
Linux hidraw testing results are the same as Windows if I use the same commands (as per HIDAPI documentation).
```
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report 0, 4000 msec timeout...read 63 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 00
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report, report_id 0... 
read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device

mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-feature 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte feature report, report_id 0...read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E C6 0F 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
```

mcuee, 2023-06-18:
Linux libusb testing results are the same as Windows if I use the same commands (as per HIDAPI documentation).

```
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report 0, 4000 msec timeout...read 63 bytes:
 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20
 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 00
Closing device
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-output 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-input-alt 0
Opening device, vid/pid: 0x0925/0x7001
Writing output report of 64-bytes...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte input report, report_id 0... 
read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
mcuee@UbuntuSwift3:~/build/hid/hidapitester_mod$ sudo ./hidapitester_libusb --vidpid 0925:7001 --open -l 64 --timeout 4000 --send-feature 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63 --read-feature 0
Opening device, vid/pid: 0x0925/0x7001
Writing 64-byte feature report...wrote 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Reading 64-byte feature report, report_id 0...read 64 bytes:
 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E ED 0F 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
Closing device
```

mcuee, 2023-06-18:
@Youw 

In the end, if my testing results are correct, then  there is only one thing to fix about documentation for `hid_send_feature_report`. 

It should be changed to the following.

```
@returns
				This function returns the actual number of bytes written plus
				one for the report ID (which is still in the first byte), and
				-1 on error.
				Call hid_error(dev) to get the failure reason.
```

One thing I would like to do is to check with linux-input mailing list to confirm that Linux hidraw documentation is not correct for the two IOCTLs: `HIDIOCGFEATURE` and `HIDIOCGINPUT`.

https://docs.kernel.org/hid/hidraw.html
https://github.com/torvalds/linux/blob/master/Documentation/hid/hidraw.rst

HIDIOCGFEATURE(len):
Get a Feature Report
This ioctl will request a feature report from the device using the control endpoint. The first byte of the supplied buffer should be set to the report number of the requested report. For devices which do not use numbered reports, set the first byte to 0. The returned report buffer will contain the report number in the first byte, followed by the report data read from the device. **For devices which do not use numbered reports, the report data will begin at the first byte of the returned buffer.**

I beleive this shoudl be:
**For devices which do not use numbered reports, the report data will begin at the first byte of the returned buffer.**

HIDIOCGINPUT(len):
Get an Input Report
This ioctl will request an input report from the device using the control endpoint. This is slower on most devices where a dedicated In endpoint exists for regular input reports, but allows the host to request the value of a specific report number. Typically, this is used to request the initial states of an input report of a device, before an application listens for normal reports via the regular device read() interface. The format of the buffer issued with this report is identical to that of HIDIOCGFEATURE.

Youw, 2023-06-18:
> fix about documentation for hid_send_feature_report.
> It should be changed to the following.

That I agree, it will be way more consistent overall.

---
I'd make one general statement: from a good API/architecture point of view: if we're passing a buffer (regardless of its nature) and expect it to be read or written to - it is best if API function (the return value) gives that buffer size (from the start) in case of success.
I.e.: when the report IDs are not used and the report size is 4 bytes, user code still have to prepare a buffer of 5 bytes. An in case of success - it is best to return 5 (even of there been only 4 bytes transfered over the transport bus - not sure if that's the case for HID, but _that's not my point_), to confirm that the implementation _has accounted for_ all bytes in the buffer. It also makes all validations (proxy/wrappers if needed) simpler and device-independent.

mcuee, 2023-06-18:
> I'd make one general statement: from a good API/architecture point of view: if we're passing a buffer (regardless of its nature) and expect it to be read or written to - it is best if API function (the return value) gives that buffer size (from the start) in case of success.
> I.e.: when the report IDs are not used and the report size is 4 bytes, user code still have to prepare a buffer of 5 bytes. And in case of success - it is best to return 5 (even if there been only 4 bytes transfered over the transport bus - not sure if that's the case for HID, but _that's not my point_), to confirm that the implementation _has accounted for_ all bytes in the buffer. It also makes all validations (proxy/wrappers if needed) simpler and device-independent.

@Youw 

Great. I have actually similar thoughts when I wrote the following in the above comment.
**We can see that hid_read_timeout gets 63 bytes and hid_get_input_report gets 64 bytes under Windows, which is as per the HIDAPI documentation. Whether that is a good thing or not is another story.**

If we look at the testing results, only `hid_read_timeout` and `hid_read` are not following your thoughts. We can not remove the two due to backwards compatibility reason. However, we can create two new functions which add the `fake` Report ID 0 to the buffer and then it will be in line with `hid_get_input_report`. Maybe we can call them `hid_read_timeout_1` and `hid_read_1`. Sorry for the lame names. :-)

mcuee, 2023-06-19:
> One thing I would like to do is to check with linux-input mailing list to confirm that Linux hidraw documentation is not correct for the two IOCTLs: HIDIOCGFEATURE and HIDIOCGINPUT.
> 
> https://docs.kernel.org/hid/hidraw.html
> https://github.com/torvalds/linux/blob/master/Documentation/hid/hidraw.rst

Question asked in linux-input mailing list.
https://marc.info/?l=linux-input&m=168715494222601&w=3

Youw, 2023-06-19:
> only `hid_read_timeout` and `hid_read` are not following your thoughts

I don't think we should change the behavior of that one either.
Maybe from the usage perspective it is a bit less convenient, maybe the API is a bit inconsistent _compared to other API calls_, but overall it follows the general statement noted above - the return value gives the number of bytes written to the buffer, e.g. there would be no cases where return `4` could mean that the bytes `[1-4]` are written - it is always `[0-3]`, i.e. the first 4 bytes in the buffer.
As for the handling of the report and detecting if the first byte is the report ID or not - user application has to know its device anyway, to interpret the recport correctly. Either dummy `0` report ID is prepended at the beginning or not (for devices that do not use numbered reports) - that doesn't change things drastically.

And, of course, breaking backward compativility is something we should to avoid as much as we can.

mcuee, 2023-06-19:
Modified testing code from the following Linux kernel example using hidraw for the above test device
https://github.com/torvalds/linux/blob/master/samples/hidraw/hid-example.c

```
// SPDX-License-Identifier: GPL-2.0
/*
 * Hidraw Userspace Example
 *
 * Copyright (c) 2010 Alan Ott <alan@signal11.us>
 * Copyright (c) 2010 Signal 11 Software
 *
 * The code may be used by anyone for any purpose,
 * and can serve as a starting point for developing
 * applications using hidraw.
 */

/* Linux */
#include <linux/types.h>
#include <linux/input.h>
#include <linux/hidraw.h>

/*
 * Ugly hack to work around failing compilation on systems that don't
 * yet populate new version of hidraw.h to userspace.
 */
#ifndef HIDIOCSFEATURE
#warning Please have your distro update the userspace kernel headers
#define HIDIOCSFEATURE(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x06, len)
#define HIDIOCGFEATURE(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x07, len)
#endif

/* Unix */
#include <sys/ioctl.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <unistd.h>

/* C */
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <errno.h>

const char *bus_str(int bus);

int main(int argc, char **argv)
{
	int fd;
	int i, res, desc_size = 0;
	char buf[64];
	struct hidraw_report_descriptor rpt_desc;
	struct hidraw_devinfo info;
	char *device = "/dev/hidraw1";

	if (argc > 1)
		device = argv[1];

	/* Open the Device with non-blocking reads. In real life,
	   don't use a hard coded path; use libudev instead. */
	fd = open(device, O_RDWR|O_NONBLOCK);

	if (fd < 0) {
		perror("Unable to open device");
		return 1;
	}

	memset(&rpt_desc, 0x0, sizeof(rpt_desc));
	memset(&info, 0x0, sizeof(info));
	memset(buf, 0x0, sizeof(buf));

	/* Get Report Descriptor Size */
	res = ioctl(fd, HIDIOCGRDESCSIZE, &desc_size);
	if (res < 0)
		perror("HIDIOCGRDESCSIZE");
	else
		printf("Report Descriptor Size: %d\n", desc_size);

	/* Get Report Descriptor */
	rpt_desc.size = desc_size;
	res = ioctl(fd, HIDIOCGRDESC, &rpt_desc);
	if (res < 0) {
		perror("HIDIOCGRDESC");
	} else {
		printf("Report Descriptor:\n");
		for (i = 0; i < rpt_desc.size; i++)
			printf("%hhx ", rpt_desc.value[i]);
		puts("\n");
	}

	/* Get Raw Name */
	res = ioctl(fd, HIDIOCGRAWNAME(256), buf);
	if (res < 0)
		perror("HIDIOCGRAWNAME");
	else
		printf("Raw Name: %s\n", buf);

	/* Get Physical Location */
	res = ioctl(fd, HIDIOCGRAWPHYS(256), buf);
	if (res < 0)
		perror("HIDIOCGRAWPHYS");
	else
		printf("Raw Phys: %s\n", buf);

	/* Get Raw Info */
	res = ioctl(fd, HIDIOCGRAWINFO, &info);
	if (res < 0) {
		perror("HIDIOCGRAWINFO");
	} else {
		printf("Raw Info:\n");
		printf("\tbustype: %d (%s)\n",
			info.bustype, bus_str(info.bustype));
		printf("\tvendor: 0x%04hx\n", info.vendor);
		printf("\tproduct: 0x%04hx\n", info.product);
	}

	/* Set Feature */
	buf[0] = 0x0; /* Report Number */
	for (i = 1; i < 64; i++)
		buf[i] = 0x40 + i;

	res = ioctl(fd, HIDIOCSFEATURE(64), buf);
	if (res < 0)
		perror("HIDIOCSFEATURE");
	else
		printf("ioctl HIDIOCSFEATURE returned: %d\n", res);

	/* Get Feature */
	buf[0] = 0x0; /* Report Number */
	res = ioctl(fd, HIDIOCGFEATURE(64), buf);
	if (res < 0) {
		perror("HIDIOCGFEATURE");
	} else {
		printf("ioctl HIDIOCGFEATURE returned: %d\n", res);
		printf("Report data:\n\t");
		for (i = 0; i < res; i++)
			printf("%hhx ", buf[i]);
		puts("\n");
	}

	/* Send a Report to the Device */
	buf[0] = 0x0; /* Report Number */
	for (i = 1; i < 64; i++)
		buf[i] = 0x20 + i;
	
	res = write(fd, buf, 64);
	if (res < 0) {
		printf("Error: %d\n", errno);
		perror("write");
	} else {
		printf("write() wrote %d bytes\n", res);
	}
	//sleep(1);
	/* Get a report from the device */
	res = read(fd, buf, 64);
	if (res < 0) {
		perror("read");
	} else {
		printf("read() read %d bytes:\n\t", res);
		for (i = 0; i < res; i++)
			printf("%hhx ", buf[i]);
		puts("\n");
	}
	close(fd);
	return 0;
}

const char *
bus_str(int bus)
{
	switch (bus) {
	case BUS_USB:
		return "USB";
		break;
	case BUS_HIL:
		return "HIL";
		break;
	case BUS_BLUETOOTH:
		return "Bluetooth";
		break;
	case BUS_VIRTUAL:
		return "Virtual";
		break;
	default:
		return "Other";
		break;
	}
}

mcuee@UbuntuSwift3:~/build/hid/hidraw$ gcc -o myhid-example myhid-example.c 

mcuee@UbuntuSwift3:~/build/hid/hidraw$ sudo ./myhid-example 
Report Descriptor Size: 47
Report Descriptor:
6 a0 ff 9 1 a1 1 9 3 15 0 26 ff 0 75 8 95 3f 81 2 9 4 15 0 26 ff 0 75 8 95 3f 91 2 9 5 15 0 26 ff 0 75 8 95 3f b1 2 c0 

Raw Name: Microchip Technology Inc. Generic HID
Raw Phys: usb-0000:00:14.0-1/input0
Raw Info:
	bustype: 3 (USB)
	vendor: 0x0925
	product: 0x7001
ioctl HIDIOCSFEATURE returned: 64
ioctl HIDIOCGFEATURE returned: 64
Report data:
	0 41 42 43 44 45 46 47 48 14 4 18 4 4d 72 31 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f 

write() wrote 64 bytes
read() read 63 bytes:
	21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
```

mcuee, 2023-06-19:
Moreover the kernel codes here seem to prove that the documentation
is wrong for both HIDIOCGFEATURE and HIDIOCGINPUT.

https://github.com/torvalds/linux/blob/master/include/uapi/linux/hidraw.h

```
/* The first byte of SFEATURE and GFEATURE is the report number */
#define HIDIOCSFEATURE(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x06, len)
#define HIDIOCGFEATURE(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x07, len)
#define HIDIOCGRAWUNIQ(len)     _IOC(_IOC_READ, 'H', 0x08, len)
/* The first byte of SINPUT and GINPUT is the report number */
#define HIDIOCSINPUT(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x09, len)
#define HIDIOCGINPUT(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x0A, len)
/* The first byte of SOUTPUT and GOUTPUT is the report number */
#define HIDIOCSOUTPUT(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x0B, len)
#define HIDIOCGOUTPUT(len)    _IOC(_IOC_WRITE|_IOC_READ, 'H', 0x0C, len)
```

mcuee, 2023-06-22:
> Question asked in linux-input mailing list. https://marc.info/?l=linux-input&m=168715494222601&w=3

https://lore.kernel.org/linux-input/1f24c83c-fe36-d2ab-c755-e83fc6a265eb@redhat.com/T/#m458a231a4f3c9eb91553656b9d7d50d86cacd4be

Answer from the linux-input core maintainer.

++++++++++++++++++++++
Yep, this is wrong.

This should be read:
```
The returned report buffer will contain the report number in the first 
byte or 0 when the device doesn't use numbered reports. The data read 
from the device will be stored in the following bytes.
```

FWIW, the difficulty to find out what the code does is because this part 
is handled in each HID transport driver: USB, Bluetooth, I2C.

Looking at drivers/hid/usbhid/hid-core.c, lines 869+, the function 
usbhid_get_raw_report() is the one used by hidraw in the end and is 
still the original code from Alan:

```
/* Byte 0 is the report number. Report data starts at byte 1.*/
buf[0] = report_number;
if (report_number == 0x0) {
	/* Offset the return buffer by 1, so that the report ID
	   will remain in byte 0. */
	buf++;
	count--;
	skipped_report_id = 1;
}
```

mcuee, 2023-06-22:
@Youw @tormodvolden @jonasmalacofilho @JoergAtGithub @todbot 

Hopefully the above answer clear the doubt on this issue.

Youw, 2023-06-22:
that is consistent with the "general statement" above
nice

mcuee, 2023-06-24:
BTW, I believe the following kernel code explains the behavior of Output Report handling under Linux.

For HID device without numbered report, it is okay to skip the report ID 0 for Output Report (not following HIDAPI documentation) under Linux hidraw backend. But it is better to follow HIDAPI documentation to include Report ID 0 in the buffer in order to be cross-platform code.

https://github.com/torvalds/linux/blob/master/drivers/hid/usbhid/hid-core.c
```
static int usbhid_output_report(struct hid_device *hid, __u8 *buf, size_t count)
{
	struct usbhid_device *usbhid = hid->driver_data;
	struct usb_device *dev = hid_to_usb_dev(hid);
	int actual_length, skipped_report_id = 0, ret;

	if (!usbhid->urbout)
		return -ENOSYS;

	if (buf[0] == 0x0) {
		/* Don't send the Report ID */
		buf++;
		count--;
		skipped_report_id = 1;
	}

	ret = usb_interrupt_msg(dev, usbhid->urbout->pipe,
				buf, count, &actual_length,
				USB_CTRL_SET_TIMEOUT);
	/* return the number of bytes transferred */
	if (ret == 0) {
		ret = actual_length;
		/* count also the report id */
		if (skipped_report_id)
			ret++;
	}

	return ret;
}
```

Same for I2C HID.
https://github.com/torvalds/linux/blob/master/drivers/hid/i2c-hid/i2c-hid-core.c

```
static int i2c_hid_get_raw_report(struct hid_device *hid,
				  u8 report_type, u8 report_id,
				  u8 *buf, size_t count)
{
	struct i2c_client *client = hid->driver_data;
	struct i2c_hid *ihid = i2c_get_clientdata(client);
	int ret_count;

	if (report_type == HID_OUTPUT_REPORT)
		return -EINVAL;

	/*
	 * In case of unnumbered reports the response from the device will
	 * not have the report ID that the upper layers expect, so we need
	 * to stash it the buffer ourselves and adjust the data size.
	 */
	if (!report_id) {
		buf[0] = 0;
		buf++;
		count--;
	}

	ret_count = i2c_hid_get_report(ihid,
			report_type == HID_FEATURE_REPORT ? 0x03 : 0x01,
			report_id, buf, count);

	if (ret_count > 0 && !report_id)
		ret_count++;

	return ret_count;
}
```

mcuee, 2023-06-24:
But I can not find similar things for bluetooth, maybe the underline Bluetooth subsystem is handling that.
https://github.com/torvalds/linux/blob/master/net/bluetooth/hidp/core.c