libusb/hidapi #592: hid_get_feature_report() needs to be call two times to get a report

sdegrande, 2023-06-22:
I have an issue getting a feature report from a HMD device: a first call to hid_get_feature_report() reports an error (On Windows, it is a ERROR_GEN_FAILURE - A device attached to the system is not functioning) while a second call returns the expected buffer.

I have the same issue on Linux (Fedora 38).

Even if I close and re-open the device between the 2 requests. For instance, a first use of 'hidapitester' returns -1 bytes, while a second use returns the 64 expected bytes.

And that behavior is cyclic: a third call fails, a forth call succeeds, and so one.

I tried with the hid client test from Microsoft (hclient), which uses an other API (HidD_GetFeature), and I have the exact same behavior.

Could any of you have an explanation ?

mcuee, 2023-06-22:
You may need to provide more details about the device (eg: the output of hidtest of your device) and the your test results.

Do you know the detail communication protocol of the device? It can be the device is broken. It can be that you need to send a set feature request before sending the get feature request.

mcuee, 2023-06-25:
Move this to discussion as it is unlikely hidapi issue.