libusb/hidapi #595: call hid_close stuck on android

pengxianheng-nreal, 2023-06-29:
as the title show, I use hidapi to communicate with my usb hid device on android, sometimes, I call hid_close will stuck, It stuck at 
hidapi_thread_join(&dev->thread_state); and when stucking, The "read_thread" thread is running thebelow loop：
	while (!dev->transfer_loop_finished)
		libusb_handle_events_completed(usb_context, &dev->transfer_loop_finished);
and not jump out。
  anyone cause the same problem? thanks a lot.

mcuee, 2023-06-30:
@pengxianheng-nreal 

How do you use hidapi under Android? What is the Android device you use for testing?

Take note there are limitations on how to use hidapi under un-rooted Android devices.
* https://github.com/libusb/hidapi/issues/178#issuecomment-931252311

pengxianheng-nreal, 2023-06-30:
@mcuee  The way I use hidapi on android is the same with https://github.com/libusb/hidapi/issues/178#issuecomment-931252311.  use java to get fd, and pass to native hidapi,[hid_libusb_wrap_sys_device](https://github.com/libusb/hidapi/blob/0ec60c03cbe87cdbfb9fb199c7536fdcbc0a94b8/libusb/hidapi_libusb.h#L48);

Youw, 2023-07-03:
Can you verify that you're not trying to call `hid_close` concurently with any other HIDAPI function?
Can I assume you're using latest `master`/latest release of HIDAPI?

mcuee, 2023-08-14:
@pengxianheng-nreal 

Any updates? Thanks.

pengxianheng-nreal, 2024-05-14:
@mcuee  I use https://github.com/libusb/hidapi/pull/457/commits to fix my problem. thanks.  my code is older than master/latest

pengxianheng-nreal, 2024-05-14:
but now,I cause the new cause . even if I use https://github.com/libusb/hidapi/pull/457/commits.  also encouter stuck when call hid_close on android.  and before stuck ,  I use hid_write alaways return -1. and i get the errno is 9(Bad file descriptor). and after a few seconds usb device has been plugged out.  and then I call hid_close stuck . I found it stuck at 
	hidapi_thread_join(&dev->thread_state);  in libusb/hid.c .  I don't know why. @mcuee

pengxianheng-nreal, 2024-05-14:
> Can you verify that you're not trying to call `hid_close` concurently with any other HIDAPI function? Can I assume you're using latest `master`/latest release of HIDAPI?

 I call 'hid_write' concurently with 'hid_close' in different thread,  is this right calling logic?

Youw, 2024-05-14:
HIDAPI is not thread-safe.
You definitely shouldn't call both hid_write and hid_close concurently. That is undefined behavior.

Youw, 2025-01-23:
Closing as per my last comment.

closed 2025-01-23 by Youw