libusb/hidapi #603: OSX Build instructions

Lewiscowles1986, 2023-07-29:
Hello 👋 

Thanks for writing this amazing software. It's really cool and will help me to talk to USB devices.

I Had a particular frustration when getting this to be used by a python library on Mac OSX

It kept saying the dynlib could not be found. Normally, I can do simple things, such as `brew install libhid`, and it works, but it seems like [the formulae is unmaintained](https://formulae.brew.sh/formula/libhid).

At no point do I expect, or am I asking you to maintain that utility code. It's kind enough of you to have written this software.

But I do now have instructions from my own debugging and guessing at some of the instructions in [the "formulae" homebrew uses to build libhid](https://github.com/Homebrew/homebrew-core/blob/f0c6e070add5312cb25639bcade628c21a6de2af/Formula/libhid.rb)

The instructions without implying homebrew or ruby (although I have both installed):

```
mkdir -p ~/projects/study
cd ~/projects/study
git clone https://github.com/libusb/hidapi.git
cd hidapi
autoreconf --force --install --verbose
 ./configure --disable-debug --disable-dependency-tracking --disable-swig
make build
sudo make install
```

It might be pure luck that led to me having python and associated libraries, and other system level utilities (likely supplied by homebrew); so your mileage with the above may vary. The machine is not new.

To test the script, I did not get the test documented in homebrew to work. I did manage to get your `mac` folder to build as laid out in the `BUILD.MD`. If you don't want to also run that, I have the following python script, which worked for me. Python comes bundled with Macs at the moment.

```
import ctypes
libHidBindings = ctypes.cdll.LoadLibrary('libhidapi.dylib')
```

I Hope this helps someone, or spurs someone else to learn enough about homebrew to fix, or to contribute documentation on building `libhidapi.dylib` from this great repo.

mcuee, 2023-07-29:
You may want to use cython-hidapi which is an established binding for hidapi.
https://github.com/libusb/hidapi/wiki#various-related-projects-and-repositories

mcuee, 2023-07-29:
There is also another binding (not tested).
https://github.com/apmorton/pyhidapi

mcuee, 2023-07-29:
And HIDAPI has nothing to do with libhid. The correct hidapi homebrew forumla is this one.
https://formulae.brew.sh/formula/hidapi
```
brew install hidapi
```

closed 2023-07-29 by mcuee

Lewiscowles1986, 2023-07-29:
All I can say is that it works, and thank you for providing this information, which will surely help others 👍