libusb/hidapi #609: hidraw hid_enumerate issue with vendor / product IDs

braewoods, 2023-08-14:
The current code for `hid_enumerate` does not properly handle zero value product and vendor IDs. At present the code will check both product and vendor IDs for matches if either of them is non-zero, due to this conditional: `vendor_id != 0 || product_id != 0`. However the public header documentation implies they should be considered separately and not as a group. This is to allow you to get all devices matching the vendor id, product id, or both, or all devices period. However `hidraw` as written will only work return devices matching both or all devices (if both are zero).

Youw, 2023-08-14:
I don't believe that's true.
[There](https://github.com/libusb/hidapi/blob/5fd7a79b37ee4981a699558e280ba3102c463770/linux/hid.c#L893-L896) lines are making sure of that.

Is your observation are based on visual code analysys or a result of your testing?

braewoods, 2023-08-14:
Nevermind, I got the impression because of the outer conditional.

closed 2023-08-14 by braewoods