libusb/hidapi #616: hid_red() returning 0

modhusudhan, 2023-08-23:
I am working on Win10 and Win11.
I am able to see that hid_write() is successful to the HID device, Device is also sending data back to the PC.
But when I try to read the data using hid_read(), the return is always 0.
hid_read() is inturn calling ReadFile()

Send and Receive data is 64 in length.

Using Qt 5.15, MinGW compiler with CMake.

Could anyone please help me

Youw, 2023-08-23:
What version of HIDAPI do you use?
What kind of data does the Device is sending back?
Do you use the device in blocking or non-blocking mode?
Are you retrying in a loop?

Not enough info even try to answer you.

modhusudhan, 2023-08-23:
What version of HIDAPI do you use? -->**0.14.0**
What kind of data does the Device is sending back?
**It is a custom data from the device of 64 bytes. Product information to start with.**
Do you use the device in blocking or non-blocking mode?
**Tried in both the modes, behavior is the same**
Are you retrying in a loop?
**Yes**

modhusudhan, 2023-08-24:
Tried testing with hidapitester

C:\Users\Modhu\Downloads\hidapitester-windows-x86_64>hidapitester --vidpid 483 --usagePage 0x00FF --open --send-output 0xa3,0x91,0x20,0xdf --read-input
Opening device, vid/pid:0x0483/0x0000, usagePage/usage: FF/0
Device opened
Writing output report of 64-bytes...wrote 64 bytes:
 A3 91 20 DF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Reading 64-byte input report 0, 250 msec timeout...read 0 bytes:
Closing device

Can anyone guide me where to look further?

Youw, 2023-08-24:
For starters I'd say you need to make sure your device is HID compliant and that the device really is sending an input report.

Unless you find an issue in HIDAPI, this is not a place to discuss device-specific development.
This is a HIDAPI issue tracker, not a support forum.

closed 2023-08-24 by Youw

mcuee, 2023-08-27:
@modhusudhan 

If you want to get technical support, you can try libusb-devel mailing list. You need to subscribe in order to post a question.
You can mention this link in your post.
https://sourceforge.net/projects/libusb/lists/libusb-devel