libusb/hidapi #617: The problem with the build for android

Sokolargm, 2023-08-27:
So, I'm trying to build .so of this via Android Studio NDK, but when i run this
`LOCAL_PATH:= $(call my-dir)

HIDAPI_ROOT_REL:= ../..
HIDAPI_ROOT_ABS:= $(LOCAL_PATH)/../..

include $(CLEAR_VARS)

LOCAL_SRC_FILES := \
  $(HIDAPI_ROOT_REL)/libusb/hid.c

LOCAL_C_INCLUDES += \
  $(HIDAPI_ROOT_ABS)/hidapi \
  $(HIDAPI_ROOT_ABS)/android

LOCAL_SHARED_LIBRARIES := libusb1.0

LOCAL_MODULE := libhidapi

include $(BUILD_SHARED_LIBRARY)`

I am getting this error
`
D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c>ndk-build.cmd
Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-19.
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:586: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:599: *** Android NDK: Note that old versions of ndk-build silently ignored this error case. If your project worked on those versions, the missing libraries were not needed and you can remove those dependencies from the module to fix your build. Alternatively, set APP_ALLOW_MISSING_DEPS=true to allow missing dependencies.    .  Stop.
`
how can I add a usb 1.0 module there?

Sokolargm, 2023-08-27:
Ok I added APP_ALLOW_MISSING_DEPS=true and now I get 
`D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c>ndk-build.cmd
Android NDK: APP_PLATFORM not set. Defaulting to minimum supported version android-19.
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:586: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:586: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:586: Android NDK: Module hidapi depends on undefined modules: usb1.0
D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:586: Android NDK: Module hidapi depends on undefined modules: usb1.0
[arm64-v8a] Compile        : hidapi <= hid.c
[arm64-v8a] SharedLibrary  : libhidapi.so
ld: error: undefined symbol: libusb_init
>>> referenced by hid.c:489 (jni/../../libusb/hid.c:489)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_init)
>>> referenced by hid.c:489 (jni/../../libusb/hid.c:489)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:489 (jni/../../libusb/hid.c:489)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced 1 more times

ld: error: undefined symbol: libusb_exit
>>> referenced by hid.c:504 (jni/../../libusb/hid.c:504)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_exit)

ld: error: undefined symbol: libusb_get_device_list
>>> referenced by hid.c:801 (jni/../../libusb/hid.c:801)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1275 (jni/../../libusb/hid.c:1275)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)

ld: error: undefined symbol: libusb_free_config_descriptor
>>> referenced by hid.c:891 (jni/../../libusb/hid.c:891)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1314 (jni/../../libusb/hid.c:1314)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced by hid.c:1392 (jni/../../libusb/hid.c:1392)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_libusb_wrap_sys_device)

ld: error: undefined symbol: libusb_get_device_descriptor
>>> referenced by hid.c:809 (jni/../../libusb/hid.c:809)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:844 (jni/../../libusb/hid.c:844)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1281 (jni/../../libusb/hid.c:1281)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced 2 more times

ld: error: undefined symbol: libusb_get_active_config_descriptor
>>> referenced by hid.c:821 (jni/../../libusb/hid.c:821)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1285 (jni/../../libusb/hid.c:1285)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced by hid.c:1352 (jni/../../libusb/hid.c:1352)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_libusb_wrap_sys_device)

ld: error: undefined symbol: libusb_get_config_descriptor
>>> referenced by hid.c:823 (jni/../../libusb/hid.c:823)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1287 (jni/../../libusb/hid.c:1287)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced by hid.c:1354 (jni/../../libusb/hid.c:1354)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_libusb_wrap_sys_device)

ld: error: undefined symbol: libusb_close
>>> referenced by hid.c:885 (jni/../../libusb/hid.c:885)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_enumerate)
>>> referenced by hid.c:1309 (jni/../../libusb/hid.c:1309)
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_open_path)
>>> referenced by hid.c:1394 (jni/../../libusb/hid.c:1394)
[... cut ...]
>>>               D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/objs/hidapi/__/__/libusb/hid.o:(hid_close)

ld: error: too many errors emitted, stopping now (use -error-limit=0 to see all errors)
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [D:/HidBuild/android-ndk-r25c-windows/android-ndk-r25c/build/../build/core/build-binary.mk:657: D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c/obj/local/arm64-v8a/libhidapi.so] Error 1

D:\HidBuild\android-ndk-r25c-windows\android-ndk-r25c>`

closed 2023-08-27 by Sokolargm

criminact, 2024-10-16:
@Sokolargm What solved the issue?